thomluther/ha-anker-solix #20: 'SensorDeviceClass' for 'CO2 Savings' and 'Saved Costs' sensors missing

DFS-90, 2024-03-13:
### System Health details

## System Information

version | core-2024.3.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.34.0
Stage | running
Available Repositories | 1418
Downloaded Repositories | 100

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 19.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.10.0), Samba share (12.3.1), Spotify Connect (0.13.0), AirSonos (4.2.0), Advanced SSH & Web Terminal (17.2.0), Home Assistant Google Drive Backup (0.112.1), Samba Backup (5.2.0), File editor (5.8.0), Let's Encrypt (5.0.15), Portainer (2.19.4-3), Matter Server (5.4.1), Filebrowser (2.23.0_12), DynDNS Client (0.2.1), Lego (0.7.6), Mosquitto broker (6.4.0), ESPHome (2024.2.2), Node-RED (17.0.9), Speedtest (1.5.0), MariaDB (2.6.1), Deye-Dummycloud (1.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 16
-- | --
resources | 60
views | 49
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3. März 2024 um 14:49
-- | --
current_recorder_run | 13. März 2024 um 12:47
estimated_db_size | 1101.12 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After installing the plugin, HomeAssistant throws the error shown in the next box.
Thanks for helping me out and best regards,
David

### Reproduction steps

1. Install plugin
2. Restart HomeAssistant

### Debug logs

```text
Logger: homeassistant.components.mqtt.mixins
Quelle: components/mqtt/mixins.py:270
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 12:47:47 (2 Vorkommnisse)
Zuletzt protokolliert: 12:47:47

Error 'expected SensorDeviceClass or one of 'date', 'enum', 'timestamp', 'apparent_power', 'aqi', 'atmospheric_pressure', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'data_rate', 'data_size', 'distance', 'duration', 'energy', 'energy_storage', 'frequency', 'gas', 'humidity', 'illuminance', 'irradiance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'ph', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'sound_pressure', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'volatile_organic_compounds_parts', 'voltage', 'volume', 'volume_storage', 'volume_flow_rate', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/sensor/solarbank_e1600/statistics_co2_savings/config', message: '{'name': 'Solarbank E1600 Statistics CO2 Savings', 'state_topic': 'solix', 'value_template': '{{ value_json.statistics[1].total }}', 'device_class': 'mass', 'unit_of_measurement': 'kg', 'unique_id': 'solarbank_e1600_statistics_co2_savings', 'platform': 'mqtt'}'

Error 'expected SensorDeviceClass or one of 'date', 'enum', 'timestamp', 'apparent_power', 'aqi', 'atmospheric_pressure', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'data_rate', 'data_size', 'distance', 'duration', 'energy', 'energy_storage', 'frequency', 'gas', 'humidity', 'illuminance', 'irradiance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'ph', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'sound_pressure', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'volatile_organic_compounds_parts', 'voltage', 'volume', 'volume_storage', 'volume_flow_rate', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/sensor/solarbank_e1600/statistics_saved_costs/config', message: '{'name': 'Solarbank E1600 Charging Statistics Saved Costs', 'state_topic': 'solix', 'value_template': '{{ value_json.statistics[2].total }}', 'device_class': 'currency', 'unit_of_measurement': '{{ value_json.statistics[2].unit }}', 'unique_id': 'solarbank_e1600_statistics_saved_costs', 'platform': 'mqtt'}'
```

thomluther, 2024-03-13:
Hi @DFS-90 
This integration does not use MQTT, so I wonder what issue you have with the MQTT integration that you attached the logs for?

I guess you mix this integration up with the Anker Solix Add-on that also exists and provides the sensors via MQTT, This is totally different, not not related at all with this project.
Pls confirm

DFS-90, 2024-03-13:
Sorry for bothering you - you're absolutely right. I mixed this indeed with some old MQTT entities from the Anker Solix Add-on.
Thanks for your help! 👍

closed 2024-03-13 by DFS-90