thomluther/ha-anker-solix #26: Add temperature entity

Zordandi, 2024-03-31:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

During cold temperatures the Anker Solix sometimes behaves a little bit strange.
It might make debugging easier if the temperature is avaiable.

### Describe the solution you'd like

Add a temperature entity to 

### Describe alternatives you've considered

Look into the Anker App in Live view.

### Additional context

Sensor Description already available, but not used:
    AnkerSolixSensorDescription(
        key="temperature",
        translation_key="temperature",

thomluther, 2024-03-31:
The cloud API does not provide temperature value, therefore it cannot be added.
What you see in the App device details is only provided via Bluetooth or MQTT cloud Server, which is not accessible via the API.
As long as no temperature available in API, it cannot be added as entity.
You need to build your own temperature sensor alternately. There are various temperature sensors that you can attach to the solarbank and integrate to HA

thomluther, 2024-03-31:
Once temperature becomes available in Api, it will be added to the library and then is automatically available as entity since all is prepared in the integration.

thomluther, 2024-04-01:
Reopened to keep tracking requirement in case temperature will ever be supported by cloud Api

ampersandhd, 2024-11-26:
How to get the temperature to HomeAssistant via Bluetooth or MQTT?

thomluther, 2024-11-27:
I have no idea

jmozmoz, 2025-04-24:
> The cloud API does not provide temperature value, therefore it cannot be added. What you see in the App device details is only provided via Bluetooth or MQTT cloud Server, which is not accessible [...]

Just for information: The temperature is shown in the app even if the mobile phone is not connected to the same Wifi network as the solar bank and far away from it, so also no Bluetooth connection exists.

ampersandhd, 2025-04-25:
@jmozmoz Temperature in the App is only shown for the Owner, not for a Member. The Owner Access is Blocked by HomeAssistant and the API Integration. So i can Not Access the temperature in Both, App or HomeAssistant.

Panoramiac, 2025-04-27:
Does the App not use the same Api as Homeassistant? Would be nice to see the temperature. Today my Solarbank 2 was loading 400 Watts to the net instead it loading the accu. Root cause could be heat of the system.

thomluther, 2025-04-27:
If anybody will find the Api endpoint and a response that contains the temperature, you can let me know and I will add it.
So far, there is not datapoint known that show it.

jmozmoz, 2025-04-28:
> If anybody will find the Api endpoint and a response that contains the temperature, you can let me know and I will add it. So far, there is no data point known that shows it.

I can confirm, that in the Anker app the temperature is only shown if using the owner login. How would one search for such an API point (using the owner or the guest accounts)?

thomluther, 2025-04-28:
Always start searching with the owner account to avoid expected Api response errors due to permission problems.
If anything was found and documented, you can try the same queries as member account to see if they work as well.

closed 2025-12-15 by thomluther