thomluther/ha-anker-solix #31: Solix F3800 has no useful sesnors

mcowger, 2024-04-02:
### System Health details

## System Information

version | core-2024.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | aarch64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4743
Installed Version | 1.34.0
Stage | running
Available Repositories | 1403
Downloaded Repositories | 22

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 48

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 28.0 GB
disk_used | 11.5 GB
healthy | true
supported | true
board | green
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.15.0), Advanced SSH & Web Terminal (17.2.0), Network UPS Tools (0.13.0), ESPHome (2024.3.1), Mosquitto broker (6.4.0), Anylist (1.7.1)

</details>

[... 50 lines cut ...]
2024-04-02 14:48:02.014 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api.anker.com/power_service/v1/site/get_site_list
2024-04-02 14:48:02.014 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'US', 'Timezone': 'GMT-07:00', 'x-auth-token': '8d###masked###cfd1###masked###943d###masked###79', 'gtoken': '20###masked###c206###masked###86'}
2024-04-02 14:48:02.014 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-04-02 14:48:02.360 DEBUG (MainThread) [custom_components.anker_solix] Request Response: {'code': 0, 'msg': 'success!', 'data': {'site_list': []}, 'trace_id': 'ae0b1ce130345684ad8f015a956309d3'}
2024-04-02 14:48:02.360 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator matt is updating devices 
2024-04-02 14:48:02.361 DEBUG (MainThread) [custom_components.anker_solix] Updating Sites Details
2024-04-02 14:48:02.361 DEBUG (MainThread) [custom_components.anker_solix] Getting unread messages indicator
2024-04-02 14:48:02.362 DEBUG (MainThread) [custom_components.anker_solix] Request Url: GET https://ankerpower-api.anker.com/power_service/v1/get_message_unread
2024-04-02 14:48:02.362 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'US', 'Timezone': 'GMT-07:00', 'x-auth-token': '8d###masked###cfd1###masked###943d###masked###79', 'gtoken': '20###masked###c206###masked###86'}
2024-04-02 14:48:02.363 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-04-02 14:48:02.451 DEBUG (MainThread) [custom_components.anker_solix] Request Response: {'code': 0, 'msg': 'success!', 'data': {'has_unread_msg': True}, 'trace_id': '9947b8e75f105e24fc81e3a6537f066d'}
2024-04-02 14:48:02.452 DEBUG (MainThread) [custom_components.anker_solix] Updating Device Details
2024-04-02 14:48:02.452 DEBUG (MainThread) [custom_components.anker_solix] Getting bind devices
2024-04-02 14:48:02.453 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api.anker.com/power_service/v1/app/get_relate_and_bind_devices
2024-04-02 14:48:02.454 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'US', 'Timezone': 'GMT-07:00', 'x-auth-token': '8d###masked###cfd1###masked###943d###masked###79', 'gtoken': '20###masked###c206###masked###86'}
2024-04-02 14:48:02.454 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-04-02 14:48:02.612 DEBUG (MainThread) [custom_components.anker_solix] Request Response: {'code': 0, 'msg': 'success!', 'data': {'data': [{'device_sn': 'AZV7440E02100189', 'product_code': 'A1790', 'bt_ble_id': 'F02C4CCA-7DAC-8D2B-B9EF-7DC39C53', 'bt_ble_mac': 'E8EECCD24436', 'device_name': 'SOLIX F3800', 'alias_name': 'SOLIX F3800', 'img_url': 'https://public-aiot-ohi-prod.s3.dualstack.us-east-2.amazonaws.com/anker-power/public/product/2023/11/22/iot-admin/bXAgVYcpnksWkU4g/picl_A1790_normal.png', 'link_time': 1712094180, 'wifi_online': True, 'wifi_name': '', 'relate_type': ['ble', 'wifi'], 'charge': False, 'bws_surplus': 0, 'device_sw_version': 'v1.7.6', 'has_manual': True, 'hes_data': None}]}, 'trace_id': '1e9241ce2dd64638aa334e6b99679a85'}
2024-04-02 14:48:02.613 DEBUG (MainThread) [custom_components.anker_solix] Getting OTA settings
2024-04-02 14:48:02.615 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api.anker.com/power_service/v1/app/get_auto_upgrade
2024-04-02 14:48:02.616 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'US', 'Timezone': 'GMT-07:00', 'x-auth-token': '8d###masked###cfd1###masked###943d###masked###79', 'gtoken': '20###masked###c206###masked###86'}
2024-04-02 14:48:02.616 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-04-02 14:48:02.708 DEBUG (MainThread) [custom_components.anker_solix] Request Response: {'code': 0, 'msg': 'success!', 'data': {'main_switch': True, 'device_list': [{'device_sn': 'AZV7440E02100189', 'device_name': 'SOLIX F3800', 'auto_upgrade': True, 'alias_name': 'SOLIX F3800', 'icon': 'https://public-aiot-ohi-prod.s3.dualstack.us-east-2.amazonaws.com/anker-power/public/product/2023/11/22/iot-admin/bXAgVYcpnksWkU4g/picl_A1790_normal.png'}]}, 'trace_id': 'afd2d85e66556bc7c058d65df32cd2fb'}
2024-04-02 14:48:02.709 DEBUG (MainThread) [custom_components.anker_solix] Coordinator matt data: {'AZV7440E02100189': {'device_sn': 'AZV7440E02100189', 'device_pn': 'A1790', 'type': 'pps', 'bt_ble_mac': 'E8EECCD24436', 'name': 'SOLIX F3800', 'alias': 'SOLIX F3800', 'wifi_online': True, 'charge': False, 'bws_surplus': '0', 'sw_version': 'v1.7.6', 'auto_upgrade': True}}
2024-04-02 14:48:02.709 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_matt data in 0.703 seconds (success: True)
```

thomluther, 2024-04-03:
Hello
The cloud Api provides only data for devices that are configured into a Power system (= Api site).
You have no sites listed for your account in the cloud, means your PPS is a standalone device only, not power data available in the cloud.
What you see in the App is data provided either via BT or via a MQTT server that gets the data when you watch them in the App.
None of them is accessible via cloud Api, therefore cannot be supported in the integration.

thomluther, 2024-04-03:
The sensors that you see is the ONLY data that the cloud Api has about your device.
The meaning of BWS surplus is unknown to me as well, never saw any value there in the cloud.

mcowger, 2024-04-03:
Understood.

Might be worth a warning in the README that stand alone systems don’t work
before folks do the work to get signed up.

-- Matt

thomluther, 2024-04-04:
I have already information there which devices are supported, it's only the solarbank and the Anker inverters. But I will add a warning to make the standalone limitation clear

scyto, 2024-07-05:
> I have already information there which devices are supported, it's only the solarbank and the Anker inverters. But I will add a warning to make the standalone limitation clear

Given in the US they are called 'Portable Powerstations' and within that 'Flex Series,' 'Camping Series' and 'eXtreme Series' and there is nothing called Solarbank on the website anymore https://www.anker.com/anker-solix you might want to be less judgmental of people reading your notes wrong and mis-understanding.

I read you notes many times over and had to find this thread before i understood there were some models that provide nothing of value over the API.

tl;dr there is nothing purchasable on the market in the US that this integration would seem to support - i guess solar bank is something that was discontinued?  or maybe rebranded?

kitokirisaki, 2024-07-05:
> tl;dr there is nothing purchasable on the market in the US that this integration would seem to support - i guess solar bank is something that was discontinued?  or maybe rebranded?

Guess the US does not have the solar (panel/bank/inverter) portfolio like it is in the EU.

thomluther, 2024-07-05:
Correct they have different portfolios for the devices. And Solarbank is just sold in the EU, there might be less of market in other countries, even in EU it is used just on few countries.
But Anker started to sell power panels like X1 in EU as well, so there will be more Anker devices connecting to the cloud.
And various flexible units called PPU (Portable Power unit) are also sold.
Anyway, at the end it does not depend on the device itself if they can be supported by the Api. Only if you can create a power system in the App and add your device into a system it potentially delivers regular consumption data to the Api cloud that could be supported. It depends on every device.
I also just learned for example, that it is not possible to create a power system only with Solar Panels and Anker inverters. So they cannot be supported by the Api either when not bundled with a Solarbank in a Power system...

Regarding the HA integration Readme documentation, I listed both of the Solarbank and Inverter as supported and made also a note that Standalone devices cannot be supported. So don't know what else to be documented there...

thomluther, 2024-07-05:
By the way, also with the X1 there seem to be issues when using PPS attached to it, see[ Api library issue](https://github.com/thomluther/anker-solix-api/issues/117).
Up to now, I haven't seen any system export that provided any consumption values for any type of PPS. Not sure how they are supported via the app, maybe PPS do not report regular usage data to the cloud, although the structures seem to support them, but the Api responses never show any data for the PPS.

thomluther, 2024-07-08:
Well, only solarbank e1600 systems are supported at this point in time. Nothing else and other devices don't seem to report even data when the main device it added to a power system, like a Power Panel that has attached F3800 PPS, see #117

So today only following models are supported, I will add them to the readme during next update.
A17C0: Solarbank E1600 Gen 1
A5143: MI80 inverter