thomluther/ha-anker-solix #35: Issues with currency symbol, Invalid currency code : â¬

FWeinb, 2024-04-09:
### System Health details

## System Information

version | core-2024.4.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.0-18-amd64
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 12
views | 23
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 3, 2024 at 20:08
-- | --
current_recorder_run | April 5, 2024 at 22:26
estimated_db_size | 129.24 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When configuring the price currency this component offers €,$,£,¥,₹,원 and will configure the entity to use that unit. That unit will be used as-is in the `Intl.NumberFormat` as the currency code causing the following issue: 
```
format_number.ts:73 RangeError: Invalid currency code : â¬
    at new NumberFormat (<anonymous>)
    at c (format_number.ts:66:14)
    at m (compute_state_display.ts:93:16)
    at f (compute_state_display.ts:52:10)
```
This API is expecting a [well-formed 3-letter ISO 4217 currency codes](https://tc39.es/ecma402/#sec-currency-codes) I would expect this component to convert between these symbols and the ISO 4217 currency code. 

### Reproduction steps

1. Go to a Power Site Type 2 device
2. Open the browsers development tools 
3. Observe the Stacktrace in the console

### Debug logs

```text
Not applicable as this is only a frontend issue
```

thomluther, 2024-04-09:
Hi, 
Can you desribe in more detail how you see this issue?
The App officially only supports € and $, not a currency code. Its simply a char that must be sent.
I tried also with other single currency chars as supported by the integration and did not see an issue with it.
Even the App displayed them (like £,¥), but of course a change in the App only allows $ or €

FWeinb, 2024-04-09:
When I view the cost savings entity on the a dashboard (example was the Power Site Type 2 device page) this error is thrown:
[image]

 This happens because the "€" symbol inside the `currency` property for the `Intl.NumberFormat` is used. I think to communicate the unit towards HA this component needs to translate "€" into "EUR" and vice-versa towards the Anker API.

thomluther, 2024-04-09:
Hm, sounds weird, because the number and the unit are completely different things in HA. However, the entity is classified as currency as I remember, and maybe therefore HA expect this to be iso format currency unit. If that is the case, I'm going to change the classification of the entity to accept any chars as unit.

FWeinb, 2024-04-09:
Thanks for all you work on this component! Glad you could identify a solution to this.

thomluther, 2024-04-09:
I cannot recreate this issue with Firefox, so it seems to be browser related on how monetary values are managed.
For compatability, I'm going to remove the SensorDeviceClass.MONETARY declaration from this sensor so that it does not have a device class. That should allow any char as unit to be handled correctly.

closed 2024-04-09 by thomluther