thomluther/ha-anker-solix #38: Everything becomes Unavailable

FWeinb, 2024-04-17:
### System Health details

## System Information

version | core-2024.4.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.0-18-amd64
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 13
views | 23
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 9, 2024 at 19:51
-- | --
current_recorder_run | April 9, 2024 at 21:51
estimated_db_size | 134.83 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Seeing a lot of 504 Gateway Time-out's in the log.

### Reproduction steps

1. Go to Power Site Type 2 device and see that everything is offline 

### Debug logs

```text
2024-04-17 12:18:26.458 ERROR (MainThread) [custom_components.anker_solix] Error fetching anker_solix_XXX data: Error fetching information: There was an error while requesting power_service/v1/site/get_site_list: 504, message='Gateway Time-out', url=URL('https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list')
2024-04-17 12:21:34.456 ERROR (MainThread) [custom_components.anker_solix] Request Error: 504, message='Gateway Time-out', url=URL('https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list')
2024-04-17 12:21:34.457 ERROR (MainThread) [custom_components.anker_solix] Response Text: <html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>openresty</center>
<p><em>Powered by <a href="https://apisix.apache.org/">APISIX</a>.</em></p></body>
</html>
```
```

thomluther, 2024-04-17:
Yeah, Anker cloud availability issues. Nothing that can be fixed in the integration....

thomluther, 2024-04-17:
lots of people seeing this recently and I do as well.
Also the App may reveal this issue when you look at energy graphs that do not show up

closed 2024-04-17 by thomluther

LukasGrebe, 2024-04-23:
the app works, but when I try the integration I also see this error
```

2024-04-23 16:24:06.633 ERROR (MainThread) [custom_components.anker_solix] Request Error: 502, message='Bad Gateway', url=URL('https://ankerpower-api-eu.anker.com/power_service/v1/get_message_unread')
2024-04-23 16:24:06.634 ERROR (MainThread) [custom_components.anker_solix] Response Text: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>openresty</center>
<p><em>Powered by <a href="https://apisix.apache.org/">APISIX</a>.</em></p></body>
</html>
```

I thought it might be an error on my end but looks like this may just be bad timing with setting up the integration?

thomluther, 2024-04-23:
Yes, Any gateway error means the cloud server is not reachable.
You need to retry later on when the App is showing data on the home page.

montagnrw, 2024-04-23:
The Anker Cloud Service currently has massive problems (offline / no connection)
[image]
Let's hope Anker gets this under control or they are preparing an update with more features :) (local api --> !!! hope !!!)