thomluther/ha-anker-solix #42: Update 1.2.0 not compatible with HA 2024.04.1

FWeinb, 2024-04-28:
### System Health details

## System Information

version | core-2024.4.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.0-18-amd64
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4817
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 13
views | 24
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 21, 2024 at 10:04
-- | --
current_recorder_run | April 28, 2024 at 17:17
estimated_db_size | 151.54 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

```
Unable to install package aiohttp>=3.9.4: ERROR: Cannot install aiohttp>=3.9.4 because these package versions have conflicting dependencies. ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

### Reproduction steps

1. Install 1.2.0 on HA 2024.04.01
2. Restart
3. See error log

### Debug logs

```text
Not needed
```

thomluther, 2024-04-29:
Yeah, good point, thanks for reporting this issue. aiohttp was just updated recently due to a security issue and I was 'force' to increase the aiohttp version to aiohttp>=3.9.4
I wasn't aware, that this has backwards compatability issues, so will change the min HA release to 2024.04.4 as suggested

thomluther, 2024-04-29:
@FWeinb 
I choose another route to decrease aiohttp>=3.9.3 to let it be compatible with previous HA releases.
Can you try re-install 1.2.0 version in your < 2024.04.03 version (which bumped aiohttp to 3.9.4)?
It should then work hopefully, without the need to increase min HA version...

FWeinb, 2024-04-30:
@thomluther that sounds like the better approach.  Already updated to 2024.04.04 so can't test it anymore.

thomluther, 2024-05-01:
No problem. I think with the backward aiohttp tolerance there shouldn't be a problem with previous HA releases.

closed 2024-05-01 by thomluther