thomluther/ha-anker-solix #43: Anker Solix e1600 is disconnected but values will loged on last value

montagnrw, 2024-04-28:
### System Health details

## System Information

version | core-2024.4.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.25-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4970
Installed Version | 1.34.0
Stage | running
Available Repositories | 1399
Downloaded Repositories | 29

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 30.8 GB
disk_used | 13.1 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | InfluxDB (5.0.0), File editor (5.8.0), Grafana (9.2.2), Node-RED (17.0.12), Mosquitto broker (6.4.0), MQTT Explorer (browser-1.0.3), evcc (0.125.0), SQLite Web (4.1.2), Advanced SSH & Web Terminal (17.2.0), Studio Code Server (5.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 21
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 18. April 2024 um 20:38
-- | --
current_recorder_run | 28. April 2024 um 17:44
estimated_db_size | 1365.79 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I restart my router, one e1600 goes to sleep (for whatever reason) and not reconnected with my WiFi (i only noticed later).
Now the last value was continuously reported back. I think the Anker Cloud is set up like this.
After I pressed the right button on the E1600, it immediately reconnected to the WiFi and the right value was transferred.
[image]

Maybe you can link the value transfer with the validity of the cloud connection.
I don't think it's directly an integration error

### Reproduction steps

1.  turn off WiFi
2. wait until e1600 goes off
3. turn on WiFi
4.  wakeup e1600 to connect with WiFi

### Debug logs

```text
Not needed i think
```

thomluther, 2024-04-29:
Hi
I noticed this behavior as well for the Api and that it reports stale values even if the devices is no longer connected to the cloud.
However, this is Api behavior and the Api is not official by any means, so it could also be changed anytime without notice.
At the end I have no clue which values are stale, and which not. And making everything unavailable upon cloud disconnect might be a bad idea either...
Therefore I'm reluctant to 'fix' Api issues in the integration or build work arounds for it, because anything you change could make things worse on the other end.

The Cloud connection sensor should be used in any automation to judge whether the Api values can be trusted or not. So it stays all under the control of the HA users and there is no dependency on how the Integration behaves. The integration is just a pass through of the Api to HA and pretty much reports what you would also see on the Anker App Home Page.

The App device details information is a different story since the App uses an MQTT server (different Api Interface) to get them (not the Cloud Api as usable by the integration)

If you want alerting for cloud connection or wifi connection state, I recommend to setup HA automation to send notifications when they change from online/connected state for a certain time. This will prevent to get alerts on integration or HA restarts...

thomluther, 2024-04-29:
BTW, this stale data problem is also the reason why you cannot trust the energy stats collected and reported by the Cloud. I have no idea what they consider when the device lost connection while the cloud is accumulating energy data from stale power values...

closed 2024-04-29 by thomluther