thomluther/ha-anker-solix #47: Vers 1.2.0 did not start, Conflict Dependency

Wesie, 2024-05-02:
### System Health details

## System Information

version | core-2024.3.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20+rpt-rpi-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.34.0
Stage | setup
Available Repositories | 1400
Downloaded Repositories | 6
Disabled | rate_limit

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 3
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22. April 2024, 04:14
-- | --
current_recorder_run | 2. Mai 2024, 13:54
estimated_db_size | 182.44 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After upgrade from V1.1.2 to V1.20 the Anker INtegration could not load anymore

### Reproduction steps

1. HA is running in a Docker ( PI4) 
2. Upgrade from V1.1.2 to 1.20 via HACS >Reload (select the Version 1.2.0
3. Restart HA
4. Entinitäten are not available.
5. Reload once again
6. Restart HA
7. Entinitäten also not available.
8. DL Log and found the attached error message.
9. ---------------
10. Downgrade back to V1.1.2 > Anker Integration is working again.
...

### Debug logs

```text
2024-05-02 13:40:48.050 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package aiohttp>=3.9.4: ERROR: Cannot install aiohttp>=3.9.4 because these package versions have conflicting dependencies.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
2024-05-02 13:40:55.002 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package aiohttp>=3.9.4: ERROR: Cannot install aiohttp>=3.9.4 because these package versions have conflicting dependencies.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
2024-05-02 13:41:01.969 ERROR (SyncWorker_2) [homeassistant.util.package] Unable to install package aiohttp>=3.9.4: ERROR: Cannot install aiohttp>=3.9.4 because these package versions have conflicting dependencies.
ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
2024-05-02 13:41:01.970 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'anker_solix': Requirements for anker_solix not found: ['aiohttp>=3.9.4'].
```

thomluther, 2024-05-02:
Released 1.2.1 to fix the aiohttp requirement in github.
Please refresh the HACS information for the Anker repo and update to 1.2.1, that should be installable now

closed 2024-05-02 by thomluther

Wesie, 2024-05-03:
Thx for your fast support. The integration is running again, 2 entinitäten are disabled from the integration. 
Issue is solved with V1.2.1