thomluther/ha-anker-solix #48: output power value limit for 2 banks too low

Rilgamon, 2024-05-04:
### System Health details

## System Information

version | core-2024.5.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.25-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.34.0
Stage | running
Available Repositories | 1400
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 2. Mai 2025 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | false
remote_connected | false
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-18.ui.nabu.casa
certificate_status | ready
instance_id | 9e1c075250f249778cd5482277018671
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.4
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 30.8 GB
disk_used | 17.2 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.4.0), File editor (5.8.0), Terminal & SSH (9.13.0), SQLite Web (4.1.2), InfluxDB (5.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 2
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24. April 2024 um 07:12
-- | --
current_recorder_run | 3. Mai 2024 um 17:59
estimated_db_size | 1175.25 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Not sure if this is an error. With 2 solarbanks the output power value you set on one bank is split between both banks. 
So setting 800 will reduce each bank to 400 while the max value for it is 800. To be able to have one bank provide the allowed 600 W (in germany) the limit should be at least 1200.

`Error while executing automation automation.dynamische_einspeisung_kontrolle: Value 1600.0 for number.sb1_system_einspeisevorgabe is outside valid range 0 - 800`

### Reproduction steps

Try to set a value above 800 for output power

### Debug logs

```text
Error while executing automation automation.dynamische_einspeisung_kontrolle: Value 1600.0 for number.sb1_system_einspeisevorgabe is outside valid range 0 - 800
```

thomluther, 2024-05-04:
Hi, the power limit of 800 W is given by the Api, even for dual bank setups as you can see in the [example 2](https://github.com/thomluther/hacs-anker-solix/blob/main/custom_components/anker_solix/examples/example2/device_load_WO42TA2RIXLX2WLZ.json):
```
"min_load":100,
"max_load":800,
```
Actually you always pass the total discharge to the system via the Api and the integration (not the individual Solarbank output as it may appear), and that appliance output is limited from 100 to 800 W. Even when you pass 50 W, it will be shown in the schedule but 100W will be applied. So you cannot discharge more than 800W from the whole appliance at any time to my knowledge, even if each solarbank could do 800 and provide 1600 in total. Or does your App allow to output 1600W to the house?

What you probably want to adjust is the share per solarbank, so lets say you want output of 800 W but all delivered from 1 Solarbank => 100 % share, the other then will have 0 % share. While this is possible in the App, it is not supported in the Api library
I have no idea how to apply that via the Api and no way to test it. Someone from the community must come up with tests using the Api library and document how the schedule object has to look like for adjusting the share in a dual bank setup.
Once this structure/parameters are known and validated, it must also be verified that it doesn't harm single bank setups, since HA Integration users can have any setup.
Then we can think of how it can be implemented into the Api helper methods by using either/or a percentage or additional Watt parameter for the specific solarbank. The helper method can calculate the resulting output for another solarbank in the setup and adjust this accordingly in the structure.
The Api will also have to be enhanced to get the actual share (percentage or Watt or both) for the specific solarbank from the shared schedule object of the site. Those extracted values from the current time slot are required for the HA Integration as changeable entities for the Solarbank. But you need to keep in mind, that changing one Solarbank in a dual bank system will always change the other bank too, since the schedule is shared between both.

Rilgamon, 2024-05-04:
I had a feeling that its not a bug but a feature ;)
No to 1600 in the app. I just ran into this while scripting the logic for adjusting the output power. My plan is not to have more than 800. Even 600 is enough. But currently it can happen that only one batterie is discharged and the max amount would be 400 only. Even though the system is capable of providing 600 to max out the inverter.
Luckily this is a rare case since both my banks seem to charge equally fast most of the time. Not worth putting too much energy into it.
Thank you for looking into this.

thomluther, 2024-05-04:
Exactely for such cases I think they provided to adjust the share of the 2 banks in the app, so you can control which bank is providing which power to the given appliance output.
If someone from the community is able to test and document how to adjust the share, I'm happy to help implementing it into the Api and integration

jonashein, 2024-05-05:
Hi both, I did some testing and provided some (hopefully useful) insights [here](https://github.com/thomluther/anker-solix-api/issues/97#issue-2279694526). 

Regarding the maximum power setting, I can actually set up to 1600W in the Anker app, see the screenshots below. I cannot test whether the Solarbanks will actually output 1600W as my inverter is only supports about 425W per input. Though I can confirm that the Solarbanks will output a total of 850W, so I guess they'll also output 1600W if the inverter can draw that power.

<details>
<summary>Default Mode</summary>

[image]

</details>

<details>
<summary>Advanced Mode</summary>

[image]

</details>

Also @thomluther let me know if you need help implementing the custom load feature. So far I only had the time to set up your python API, but I'm planning to look into your HA integration as well if needed.

thomluther, 2024-05-24:
@Rilgamon, @Rilgamon 
Please see announcement for 1.3.0 Beta #61 and give it a try

closed 2024-06-18 by thomluther