thomluther/ha-anker-solix #51: single inverter - api always responds with 'wifi_online': False

LukasGrebe, 2024-05-07:
### System Health details

## System Information

version | core-2024.4.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4991
Installed Version | 1.34.0
Stage | running
Available Repositories | 1402
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.4
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 116.5 GB
disk_used | 22.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | vzlogger to MQTT Gateway (0.9.5), Mosquitto broker (6.4.0), Frigate (Full Access) (0.13.2), Studio Code Server (5.15.0), Advanced SSH & Web Terminal (17.2.0), eBUSd (23.2.5), ESPHome (2024.4.2), ha-sip (3.7)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 3
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23. April 2024 um 14:23
-- | --
current_recorder_run | 7. Mai 2024 um 17:19
estimated_db_size | 498.80 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

None of the data available in the app is shown.

for some reason the api response always containers `'wifi_online': False` this may be a timing issue, but most recently I tried this right after I was getting what appears to be live data in the anker app.

after optimizing the wifi a few days ago, the app had never again reported "offline/sleep mode" for the inverter when the panels were in broad daylight. but they seem to when accessing through the integration.

maybe its also related to the fact that I can not create a power system?
i've setup the integration but failed at setting up separate access data as I can't create one in the app, I think this may be because I only have a single device and I need to select at least two to setup a power system(?) - the option to create a power system is grayed out in my app.
[image]
[image]

### Reproduction steps

1. setup integration
2. observe initial and continuous "offline" reporting for the device's wifi state.

### Debug logs

```text
2024-05-07 17:29:44.160 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator lfndly is updating sites 
2024-05-07 17:29:44.160 DEBUG (MainThread) [custom_components.anker_solix] Updating Sites data
2024-05-07 17:29:44.161 DEBUG (MainThread) [custom_components.anker_solix] Getting site list
2024-05-07 17:29:44.162 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list
2024-05-07 17:29:44.163 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': 'fa###masked###688b###masked###7b33###masked###b8', 'gtoken': 'dd###masked###1dab###masked###7c'}
2024-05-07 17:29:44.163 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-05-07 17:29:44.259 DEBUG (MainThread) [custom_components.anker_solix] lfndly request post https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list response received
2024-05-07 17:29:44.259 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Tue, 07 May 2024 15:29:44 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '97', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': '980a84475186574b11aeddfcf0eeb513', 'Traceparent': '00-980a84475186574b11aeddfcf0eeb513-23391d7ebd7fa89c-00', 'Server': 'APISIX')>
2024-05-07 17:29:44.265 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'site_list': []}, 'trace_id': '980a84475186574b11aeddfcf0eeb513'}
2024-05-07 17:29:44.266 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator lfndly request statistics: 8 last hour, 2 last minute
2024-05-07 17:29:44.266 DEBUG (MainThread) [custom_components.anker_solix] Coordinator lfndly data: {'E07100020513': {'device_sn': 'E0***masked***3', 'device_pn': 'A**masked**3', 'type': 'inverter', 'bt_ble_mac': '8**masked**E', 'name': 'MI80 Microinverter(BLE)', 'alias': 'MI80 Microinverter(BLE)', 'wifi_online': False, 'wifi_name': 'W***masked***k', 'charge': False, 'bws_surplus': '0', 'sw_version': 'v0.0.1'}}
2024-05-07 17:29:44.266 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_lfndly data in 0.107 seconds (success: True)
```

thomluther, 2024-05-07:
hi
The Cloud API needs a power system containing your devices. Without power system, you will not see any values in the Api or the integration because the device does not deliver the values to the cloud.
Standalone devices deliver values only to the MQTT server when you look into the device details or via Bluetooth to the App.

closed 2024-05-07 by thomluther

LukasGrebe, 2024-05-07:
Thanks for the feedback!

I sent a feature request for local MQTT access

timothy3001, 2024-08-05:
Did someone find any solution to this?

Huskynarr, 2025-02-18:
> hi The Cloud API needs a power system containing your devices. Without power system, you will not see any values in the Api or the integration because the device does not deliver the values to the [...]

The values are sent to the cloud server. The data is also visible in the mobile app without an active delivery, including the history. 
As in my case, it would not make sense to purchase an extra power system. As I never deliver to the outside, but consume everything directly.