thomluther/ha-anker-solix #60: Does your Plugin Supprt the new Solix 2 Pro ?

xsasx, 2024-05-23:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I want to know if your plugin support the new Solix 2 Pro ?

### Describe the solution you'd like

I want to know if your plugin support the new Solix 2 Pro ?

### Describe alternatives you've considered

I want to know if your plugin support the new Solix 2 Pro ?

### Additional context

I want to know if your plugin support the new Solix 2 Pro ?

thomluther, 2024-05-24:
Nope.
And I have nothing in plan to enhance the integration for it, since I will neither have such a device, nor time to develop and test required enhancements.

scho0ck, 2024-05-24:
I've preordered a Solix 2 Pro do you think it would be hard to implement? I'd guess the devices share the same api but then we'll only see that once devices start shipping.

thomluther, 2024-05-24:
well, you need python skills and start with the Api libary first:
https://github.com/thomluther/anker-solix-api
Once you have a system in your account, you can use the export_system tool to get an extract of all data.
The new solarbank may need/use additional endpoints to get more data, which are not used in the export tool yet.
This data must be built into the api library dictionaries for site details and device details. Those dictionaries are used as base for the HA integration.

To enhance the integration, I recommend to setup a development environment as described [here](https://developers.home-assistant.io/docs/development_environment/). Otherwise you will have little fun when enhancing integrations, since each change requires restart of the HA instance and you need debugging capabilities. That is nothing you want to do in your productive HA.
The dev container in Visual Studio code also provide great help and simplification for this development work.
But since Anker allows only a single owner account for the system, you cannot use it in your productive HA instance while using it in the dev container. You may have to switch each other more frequently between owner and shared account configuration, since shared account permissions also need to be validated when doing enhancements.
Not every query works for shared accounts, but that is something you need to catch in the api library already by doing manual testing.

So its LOTS of work and testing, and only possible when you can access a real device in your account, since NOTHING is documented, and expect that reported api fields may have inconsistent values between various query types. That is something you need to figure out during testing and implement work arounds in the api library, which builds the consolidated device information dictionaries.

thomluther, 2024-05-24:
What will become a challenge is the new schedule and preset changes. I expect it will use completely different schedule structures since weekdays are now definable per schedule and the endpoint to change the schedule may be also different.
Solarbank 2 also does not seem to have same settings per interval like the old one, so the structure and changable fields per slot may be less. This would simplify the implementation of a new helper method to modify the solarbank 2 schedules (and manual settings if no smartreader is used).

scho0ck, 2024-05-28:
It seems Anker has no interest in third party integrations as posted on their forum which means any effort being made could break forever at any given time. I've canceled my order as it is my requirement to have the device in HA including energy dashboard etc. hopefully in the future anker decides to release something.

thealkly, 2024-06-12:
I just tested the new Solix 2 Pro with the current API version. It seems to retrieve some or all values, but there are issues with timing. As usual, you get significantly more sensor values than with the current app version.

For example, the SOC is transmitted correctly, but after a few minutes, a value of 0 is received, and then a few minutes later, the correct SOC is received again. This happens with many values like battery power, battery energy, solar power, and grid feed-in power. Daily solar kWh seems to work, as well as daily discharging in kWh.

If I find some spare time in the next few weeks, I will try to check the Python API and conduct some investigations. For now, I am very grateful that some data is already available in Home Assistant. With a simple template sensor, I could filter out the 0 values as a short workaround.

kitokirisaki, 2024-06-19:
I switched from a dual Solix E1600 to the Solix 2 Pro today. Worked well with removing the configuration and changing to the new one. But like @thealkly already mentioned, it works perfectly well until the moments where the integration just receives nothing anymore.

For me, it looks like it's a firmware bug of the Solix 2 Pro, since the API does not have a timeout or something like that. It just shows a regular response, just with empty / null values.

I'm still not sure, but also don't know what data I should provide.
Maybe @thomluther can tell me for what he is looking in detail.

Here are some parts of the debug log of HA:

```
2024-06-19 16:24:30.192 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_has* data in 0.464 seconds (success: True)
2024-06-19 16:25:29.728 DEBUG (MainThread) [custom_components.anker_solix] Api Coordinator has* is updating sites 
2024-06-19 16:25:29.729 DEBUG (MainThread) [custom_components.anker_solix] Updating Sites data
2024-06-19 16:25:29.729 DEBUG (MainThread) [custom_components.anker_solix] Getting site list
2024-06-19 16:25:29.729 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list
2024-06-19 16:25:29.729 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '<AUTH_TOKEN>', 'gtoken': '<G_TOKEN>'}
2024-06-19 16:25:29.729 DEBUG (MainThread) [custom_components.anker_solix] Request Body: None
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] has* request post https://ankerpower-api-eu.anker.com/power_service/v1/site/get_site_list response received
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 19 Jun 2024 14:25:29 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '347', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': '6dcead252505fcaa14bd91c03e7f2d83', 'Traceparent': '00-6dcead252505fcaa14bd91c03e7f2d83-41cadc46a2cee314-00', 'Server': 'APISIX')>
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'site_list': [{'site_id': '<SIDE_ID>', 'site_name': 'Home', 'site_img': '', 'device_type_list': [3], 'ms_type': 2, 'power_site_type': 5, 'is_allow_delete': True, 'support_device_models': ['A17C1', 'A17X7', 'A17X8'], 'current_site_device_models': ['A17C1']}]}, 'trace_id': '6dcead252505fcaa14bd91c03e7f2d83'}
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Getting scene info for site
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/site/get_scen_info
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '<AUTH_TOKEN>', 'gtoken': '<G_TOKEN>'}
2024-06-19 16:25:29.839 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'site_id': '<SIDE_ID>'}
2024-06-19 16:25:30.235 DEBUG (MainThread) [custom_components.anker_solix] has* request post https://ankerpower-api-eu.anker.com/power_service/v1/site/get_scen_info response received
2024-06-19 16:25:30.235 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 19 Jun 2024 14:25:30 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '1790', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': 'a396ecfd8e4aa303f197338edb60dfe9', 'Traceparent': '00-a396ecfd8e4aa303f197338edb60dfe9-207dea7870a000fc-00', 'Server': 'APISIX')>
2024-06-19 16:25:30.235 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'home_info': {'home_name': 'Home', 'home_img': '', 'charging_power': '0.00', 'power_unit': 'W'}, 'solar_list': [], 'pps_info': {'pps_list': [], 'total_charging_power': '0.00', 'power_unit': 'W', 'total_battery_power': '0.00', 'updated_time': '', 'pps_status': 0}, 'statistics': [{'type': '1', 'total': '0.57', 'unit': 'kWh'}, {'type': '2', 'total': '0.57', 'unit': 'kg'}, {'type': '3', 'total': '0.20', 'unit': '€'}], 'topology_type': '1', 'solarbank_info': {'solarbank_list': [{'device_pn': 'A17C1', 'device_sn': '<SERIAL_NUMBER>', 'device_name': 'Solarbank 2 E1600 Pro', 'device_img': 'https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png', 'battery_power': '67', 'bind_site_status': '', 'charging_power': '249', 'power_unit': 'W', 'charging_status': '0', 'status': '1', 'wireless_type': '1', 'main_version': '', 'photovoltaic_power': '137', 'output_power': '249', 'create_time': 1718799203, 'set_load_power': '', 'sub_package_num': 0, 'output_cutoff_data': 10, 'is_display': True}], 'total_charging_power': '0', 'power_unit': 'W', 'charging_status': '0', 'total_battery_power': '0.67', 'updated_time': '1970-01-01 00:00:00', 'total_photovoltaic_power': '137', 'total_output_power': '249.00', 'display_set_power': False, 'battery_discharge_power': '112', 'ac_power': '20', 'to_home_load': '229', ' [... cut ...] 6', 'battery_energy': '0', 'charging_power': '0', 'power_unit': 'W', 'charging_status': '0', 'charging_status_desc': 'detection', 'status': '1', 'status_desc': 'online', 'wireless_type': '1', 'input_power': '0', 'output_power': '0', 'set_output_power': '250', 'power_cutoff': 10, 'alias': 'Solarbank 2 E1600 Pro', 'set_system_output_power': '250'}}
2024-06-19 16:28:30.179 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_has* data in 0.450 seconds (success: True)
```

Disane87, 2024-06-20:
I got my SolarBank 2 pro today and can assist if some testers are needed :)

thomluther, 2024-06-22:
The last query in the debug data is showing that the API really returns 0 values for most fields.
They are interpreted as is, since it must be considered these are the real values.
Seems to be a cloud bug that it is sometimes providing 0 values in queries

thomluther, 2024-06-22:
@kitokirisaki , any idea what the field 'ac_power': '20' could mean in your debug data while the output power to home is 100 or higher?

kitokirisaki, 2024-06-22:
> @kitokirisaki , any idea what the field 'ac_power': '20' could mean in your debug data while the output power to home is 100 or higher?
> 

No, sadly I don't know what it could mean. But I will keep an eye on it maybe I will figure it out :)

Disane87, 2024-06-24:
I used the `solarbank_monitoring.py` of the python lib and it really seems that these values are directly 0 from the api. IMHO these values are only 0 when the charge status is `detection`:

> Sites: 1, Devices: 1,           Data-Updated: 1970-01-01 00:00:00
Device         : NoName                  Alias         : Unterstand
Serialnumber   : [redacted]        Admin         : NO
Site ID        : [redacted]
Wifi SSID      :
Wifi state     : Unknown                 Signal        :  --- %
SW Version     : Unknown                 Auto-Upgrade  : Unknown
Cloud Status   : online                  Status code   : 1
Charge Status  : detection               Status code   : 0
State Of Charge:    0 %                  Min SOC       :   10 %
Battery Energy :    0 Wh                 Capacity      : 1600 Wh
Solar Power    :    0 W                  Output Power  :    0 W
Charge Power   :    0 W                  Device Preset :  200 W

thomluther, 2024-06-24:
The status description 'Detection' is just my translation in the library for a status code 0.
This is 0 like all other values, so it cannot be used as differenciator.
When value updates are skipped, they will be stale for each query interval until good data is received again. This is not a good approach I think.
I would rather set them to None to make it visible in the entity history when false data is received.
Eventually for Solarbank 2 the query interval must be reduced, since that probably updates the cloud more often.
Did you test the Monitor tool with 10s refresh? Do the data update that frequently?
When valid data is received, you see a data-updated timestamp as well. This is when the solarbank sent the data to the cloud.
Question is how much of delay does that have to the query time as maximum when Solarbank is working?
Solarbank1 was sending every 60 seconds, so how often does Solarbank 2 send data?
In Standby mode it does not send updates that frequently.

If that is for example every 10 seconds, the query interval could be reduced to 10 sec, but this will not eliminate responses with false data. However there could be more frequent queries that return valid data.
So if there are typically 3-4 false responses and 1-2 correct responses within a minute as average, the false could simply be skipped to have at least one reliable data update within a minute.

thomluther, 2024-06-24:
You need to understand the frequency and intervals of invalid responses in relation to the frequency the data is queries.
When the API queries only once per minute, it could query at a time where the cloud is just refreshing the data from the device and this not send valid data. If the device is sending in shorter intervals, there are higher changes that a 60 sec query hits such an update period.
So various update intervals must be validated to:
- understand interval for device sending data to the cloud
- how long are potentially the periods where the cloud is responding false data

Then the query interval can maybe be adjusted around those metrics to reduce number of false responses.

nicolinuxfr, 2024-06-24:
I also have a Solix 2 Pro installed and I enjoy this integration, even if I also have noticed the bug with 0 values sent regularly. Fortunately, Home Assistant is showing a somewhat correct line in the graphs :

[image]

I am no developper but if I can help expand the integration, it will be my pleasure. 

The two items I'm missing the most are the electricity returned to the grid which is always 0 W (though I have installed Anker's Smart Meter) and for my need, I would love to have a switch in order to (de)activate the smart/AI mode and the custom mode. I want to keep the battery at maximum capacity during the night, since I'm paying the electricity less than during the day and I would rather use the kWh stored inside in the morning when the sun is not high enough. Unfortunately, Anker won't let you program a mix of smart and fixed values, so I was counting on Home Assistant to do so.

thomluther, 2024-06-25:
Another open question is how the Solix 2 will use the status codes. Do they have the same coding as for Solix 1?
What you currently see for the status code description is translated only by the library based on what the Solix 1 was doing when reporting a certain code.
We need a translation table for the Solix 2 status codes as well. Ignore 0 when false values are returned, but watch status code and actual description for other conditions:
- charge (Solix 1 has only code 3 for any charging mode, but library translates it to more detailed charging mode like charge priority or charge bypass, depending on other values at reported point in time.
- discharge only
- bypass only
- standby
- discharge-bypass (new mode only supported by Solix 2), does that have a new status code?
- other codes not translated yet
- codes that are translated incorrectly for Solix 2

The solarbank monitor will show both, the status code and its translation. In HA you can also see the history of both, but the code is only an attribute and you need to create a template helper entity first for status code history tracking and comparison to the description:
{{state_attr('sensor.sb_e1600_betriebszustand','charging_status')}}

thomluther, 2024-06-26:
@nicolinuxfr , @TobTkyDev , @thealkly , @scho0ck , @xsasx , @kitokirisaki , @Disane87 
I released a [new version of the Api library](https://github.com/thomluther/anker-solix-api).
Please use it to extract your system configuration and explore existing or new endpoints to query or change data.
If you can assist development, have a look to the [open issues ](https://github.com/thomluther/anker-solix-api/issues) where help is needed. I have no Solarbank 2, so cannot explore anything through the Api.

Enhancements of the Api library are the pre-req for doing enhancements to the HA integration which uses the libarary and its helper methods

thomluther, 2024-06-26:
Hi @thealkly 
Thanks for the examples so far. So it seems the 'is_display_data' field can be reliably used to decide whether most the whole data in the response is invalid or not. The big question is what can be the typical or max duration for false responses?
Depending on that we could simply skip those responses (if it will be few typically in shorter query intervals), or render the HA entities to unavailable.

Other things to note:
- The charge status code description is most likely different or does not match anymore to Solarbank 2 behavior for each of the codes. A new validated code translation table will be required.
- The Smart Meter has no wireless type value (not a big problem). Its status code of 1 probably means the same as for any device: Connected to the cloud.
- "photovoltaic_to_grid_power": "0", => Most likely the grid export
- "grid_to_home_power": "669", => Grid import as you confirmed
- "grid_status": 0, => What could that be? Probably similar to the charge_status for Solarbanks. Also here we would need a translation table for each possible code that can be seen and what is means
- "has_history_data": true => What could that be? Can be ignored maybe. The history data as seen in the app can be queried via the energy statistics. Maybe it is just an indicator to the App to display the extra tab for grid energy diagrams.
- "ac_power": "0", => Could be the output of the internval inverter, but that should be identical with the "to_home_load": "0" field, So why this extra new field?
- "sub_package_num": 1, => This is the only field I've seen so far for the extra battery packs. Are there other fields somewhere showing their individual SOC at least, since that could be different to main battery.

Disane87, 2024-06-27:
Here is my `api_sites.json` whoch looks like it doesn't got any data because the battery is full accordingn to the Annker app and there is pretty much sun on the panels.

```json
{
  "0d6f482d-fb6a-edbc-fedf-9ec66eacddef": {
    "type": "system",
    "site_info": {
      "site_id": "0d6f482d-fb6a-edbc-fedf-9ec66eacddef",
      "site_name": "Rennstr",
      "site_img": "",
      "device_type_list": [
        3,
        6
      ],
      "ms_type": 2,
      "power_site_type": 5,
      "is_allow_delete": true,
      "support_device_models": [
        "A17C1",
        "A17X7",
        "A17X8"
      ],
      "current_site_device_models": [
        "A17C1",
        "A17X7"
      ]
    },
    "site_admin": false,
    "home_info": {
      "home_name": "Home",
      "home_img": "",
      "charging_power": "0.00",
      "power_unit": "W"
    },
    "solar_list": [],
    "pps_info": {
      "pps_list": [],
      "total_charging_power": "0.00",
      "power_unit": "W",
      "total_battery_power": "0.00",
      "updated_time": "",
      "pps_status": 0
    },
    "statistics": [
      {
        "type": "1",
        "total": "20.85",
        "unit": "kWh"
      },
      {
        "type": "2",
        "total": "20.79",
        "unit": "kg"
      },
      {
        "type": "3",
        "total": "6.05",
        "unit": "\u20ac"
      }
    ],
    "topology_type": "1",
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C1",
          "device_sn": "YIGFIMJ6Q4UE8OKD",
          "device_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png",
          "battery_power": "0",
          "bind_site_status": "",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "0",
          "status": "1",
          "wireless_type": "1",
          "main_version": "",
          "photovoltaic_power": "0",
          "output_power": "0",
          "create_time": 1718984888,
          "set_load_power": "0",
          "sub_package_num": 0,
          "output_cutoff_data": 10,
          "is_display": false,
          "alias_name": "Unterstand",
          "current_home_load": "0"
        }
      ],
      "total_charging_power": "0",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.00",
      "updated_time": "1970-01-01 00:00:00",
      "total_photovoltaic_power": "0",
      "total_output_power": "0.00",
      "display_set_power": false,
      "battery_discharge_power": "0",
      "ac_power": "0",
      "to_home_load": "0",
      "is_display_data": false,
      "solar_power_1": "0",
      "solar_power_2": "0",
      "solar_power_3": "0",
      "solar_power_4": "0"
    },
    "retain_load": "0W",
    "scene_mode": 0,
    "home_load_power": "0",
    "updated_time": "01-01-0001 00:00:00",
    "power_site_type": 5,
    "site_id": "0d6f482d-fb6a-edbc-fedf-9ec66eacddef",
    "powerpanel_list": [],
    "grid_info": {
      "grid_list": [
        {
          "device_pn": "A17X7",
          "device_sn": "Y7P5T5EDG2I7CT1T",
          "device_name": "Smart Meter",
          "device_img": "",
          "create_time": 1719053239,
          "status": "1",
          "bind_site_status": "",
          "wireless_type": ""
        }
      ],
      "photovoltaic_to_grid_power": "0",
      "grid_to_home_power": "0",
      "grid_status": 0,
      "has_history_data": true
    },
    "is_downgrade": false,
    "error_code": 0,
    "smart_plug_info": null,
    "feature_switch": null,
    "site_details": {
      "has_unread_msg": true
    }
  }
}
```

thomluther, 2024-06-27:
Hello all
It does not help to provide examples with  `"is_display_data": false`, we learned that such data is NOT USABLE and does not give any glue how the data is relating to the App or the devices/systems.
Please report and discuss any Solarbank 2 related stuff in the [Api library](https://github.com/thomluther/anker-solix-api).
A full set of export data is needed if you want to provide examples via the system export tool. This will also randomize your sensitive data, but make sure that  `"is_display_data": true` in the scene_info or api_sites output.

Please use the tools in the repo to extract your system configuration and explore existing or new endpoints to query or change data.
If you can assist development, have a look to the [open issues ](https://github.com/thomluther/anker-solix-api/issues) and report examples there or participate in the related discussions.
If you have a new topic to consider, open a new issue there

Disane87, 2024-06-27:
Sorry that was my fault. Didn’t read your post good enough. Sorry!

xsasx, 2024-07-01:
So I installed the Solix 2 at the weekend - and the Plugin works more or less which you know :) very frustrating is that the plugin very often disconnect and shows me on all entities 0 also for example battery " 0! " so i cant use the sensors for solar meassureing and stuff like that ( i have tested it with my admin account also with the 2nd account which has only read permissions.

You have all have these same behaviors ?

I will try to export you the data which maybe i can help with.

user86000, 2024-07-01:
> So I installed the Solix 2 at the weekend - and the Plugin works more or less which you know :) very frustrating is that the plugin very often disconnect and shows me on all entities 0 also for [...]

The official anker app also disconnects often (smart meter and solix2), wlan and internet is stable

xsasx, 2024-07-03:
I would like to export also my Stuff from Solix 2 Pro and Smart Meter for your help - could anybody advice me how to do that under windows ?

thomluther, 2024-07-03:
> I would like to export also my Stuff from Solix 2 Pro and Smart Meter for your help - could anybody advice me how to do that under windows ?

You can install python 3.12 runtime for windows. It comes with IDLE that can be used the lightweight runtime and development environment.
You can install following required packages manually
Required Python modules:
pip install cryptography
pip install aiohttp
pip install aiofiles

Then you should be good to go to run the tools from the Api or build your own client.py according the example for testing your own Api requests.

c0zm0, 2024-07-06:
> The official anker app also disconnects often (smart meter and solix2), wlan and internet is stable

According to my observations, the cause of the 0 values is a change in the way the data is transferred to the Anker cloud:
When I have opened the app, the plugin also receives current values. After a few minutes without using the app, there are only 0 values. If the app is opened again, the plugin also receives values again.
It looks to me like the Solix 2 stores values locally (for the app statistics), but only actively transfers them to the cloud when the app is running (presumably to save power?). 
This is also easy to observe when you start the app. It takes approx. 1 second for values to be displayed. 
I don't think there is much point in manipulating the API, you would have to "fool" the Slolix 2 into thinking that the app is permanently running.

DIMtriad, 2024-07-07:
_When I have opened the app, the plugin also receives current values. After a few minutes without using the app, there are only 0 values. If the app is opened again, the plugin also receives values again._

I can confirm that. Same here 👍

thomluther, 2024-07-07:
@DIMtriad , @c0zm0 , The change data transmission behaviour for SB2 is understood meanwhile. Therefore there is nothing the Api library or HA Integration can do to improove it. See [Api library issue 114](https://github.com/thomluther/anker-solix-api/issues/114).
You can only open Tickets with Anker to claim that you cannot longer use the App with your second account since the value issue is the same there. Shared accounts do not trigger frequent data updates either as of today.
If you are lucky, Anker has to go back to a 60 sec update interval to support shared account reporting.

c0zm0, 2024-07-07:
@thomluther : Thanks for the feedback, I just sent a support mail to Anker. Let's hope, they listen.

SBBS1981992, 2024-07-07:
Hello, I read through this and thought I was the only one with this problem. I always get zero values ​​too, so there's nothing left to do but wait? Thanks

thomluther, 2024-07-08:
Correct, not much possible in the api libray. It must flag invalid data when received in the response.
HA integration can only show them as unvailable entities for those reponses, or skip entitiy updates until a good reponse is received. However, this means HA will report outdated values that could be 5 minutes old or even more and user has no way to see if data is valid or stale

xsasx, 2024-07-08:
> @DIMtriad , @c0zm0 , The changed data transmission behavior for SB2 is understood meanwhile. Therefore there is nothing the Api library or HA Integration can do to improve it. See [Api library issue 114](https://github.com/thomluther/anker-solix-api/issues/114). You can only open Tickets with Anker to claim that you cannot longer use the App with your second account since the value issue is the same there. Shared accounts do not trigger frequent data updates either as of today. If you are lucky, Anker has to go back to a 60 sec update interval to support shared account reporting.

but i got also with my main account in HA the 5 min problem - is that normal or i understand it wrong ?

thomluther, 2024-07-08:
> but i got also with my main account in HA the 5 min problem - is that normal or i understand it wrong ?

The main account must be open in the APP. Only the mobile App triggers the frequent data update.
It does not matter what account you using in HA.

xsasx, 2024-07-08:
ah ok understand - than the only clue to get valid data is a VM with Android which run 24/7 or a mobile phone which is 24/7 open in the app. so maybe a VM would be better. Better would be your HAC solution would also trigger the updates :)

thomluther, 2024-07-08:
> Better would be your HAC solution would also trigger the updates :)

You are welcome to contribute and find out how that would work :-)

andsk8, 2024-07-20:
It would be very nice if this code can be upgraded to support the newest model Solarbank 2 and the smart meter, which is the new setup everybody will start using now that they have shipped the first batch. I can be a beta tester if needed. Thanks!

@thomluther thanks for the contribution you have already made!

thomluther, 2024-07-20:
> It would be very nice if this code can be upgraded to support the newest model Solarbank 2 and the smart meter, which is the new setup everybody will start using now that they have shipped the first [...]

Hi @andsk8 , still need testers for the Api library, especially [what type of SB2 schedule formats can be applied or not](https://github.com/thomluther/anker-solix-api/issues/108#issuecomment-2231955182)

kitokirisaki, 2024-07-24:
Hey together, 

i did the firmeware update from yesterday.

It kind of confirms my thought about a bug in the firmeware. Because since then I do not receive any wrong data anymore. Even with the second Account.

TobTkyDev, 2024-07-24:
Can confirm the same. Follows the sample time of HA settings.

Disane87, 2024-07-24:
Can confirm too, looks like the update heals it up. You can see clearly it seems to work reliable now. 

[image]

thomluther, 2024-07-24:
Interesting...maybe Anker changed the update intervall back to 1 minute.
I just published the 2.0.0 pre-release. Would be good if you can all test it against systems before and after the update.
Would be good to know whether the new config option to skip invalid data responses would also help to eliminate the 'gaps'.
And if there is no significant change in the responses by the new firmware, that may break anything.
I assume the is_display_data field in the solarbank_info structure of the scene_info is no longer showing false, which is now used by the library and HA integration to skip the entity updates....
Maybe someone can confirm this with shared account in the solarbank monitor while app is closed. It should no longer indicate Invalid data on any response hopefully, even when using short intervals.

Disane87, 2024-07-24:
Additionally I have to say I did the update a few minutes ago but the values seems to be fine since this night at 5am. So they have done some work on the API I guess and the update is only for some local fixes. As you can see in my HA history the values are fine since this morning even without firmware update

kitokirisaki, 2024-07-24:
Hey @thomluther I do use the HA integration with a shared account and I have the date without any problem from yesterday after I did the firmware update.

I did update to the version 2.0.0 and so far, it looks good :)

It did get all the extension batteries and values from each of the PV's 👍

thomluther, 2024-07-24:
So Anker does not send data more frequently, they just keep the stale date and respond it until the device sends new ones after 5 minutes. Its the same approach as I implemented as work around in the Integration

TobTkyDev, 2024-07-25:
Just updated to V2.0.0 and noticed that the sample time changed to approx. 5min now. Before I run V1.3.2 and the sample time was approx. 30s as it was set in the configuration. So, I guess Anker is making more frequent sampling in the cloud, but something has changed between V1.3.2 and V2.0.0. In the configuration I always kept 30s.

thomluther, 2024-07-25:
The devices never sent data in less than 60 seconds intervals. SB sends data only in 5 min intervals per default, Only when it is triggered by an open Anker App watching the home screen of the system, you have more frequent data updates in the cloud.
It does not make sense at all to decrease the integration polling to less than 60 seconds, since the data frequency will not change. It just causes lots more Api traffic around the clock

TobTkyDev, 2024-07-25:
Ok, sorry, then probably this was the reason when I saw 30s updates, because I had the App open on my mobile. Never mind.

kitokirisaki, 2024-07-25:
Is there a special header the app sends for that?

Somehow they need to know its the app and not any other client so that they make the updates more frequently

thomluther, 2024-07-25:
> Is there a special header the app sends for that?
> 
> Somehow they need to know its the app and not any other client so that they make the updates more frequently

If someone if able to debug / reverse engineer the App communication you may find out, but the payload is typically EDHC encrypted and you have no chance to see the content.
My assumption is that the App triggers frequent updates via the MQTT cloud, like it does for the Device Details real time data. But that could be wrong.
If it uses an endpoint on the Api cloud, there might be a way to trigger it also by the library. If nobody is able to figure this out, its the way it is.

thomluther, 2024-07-25:
> Ok, sorry, then probably this was the reason when I saw 30s updates, because I had the App open on my mobile. Never mind.

Yeah, and then probably the updates are in about 3s interval (screen refresh rate of the App home screen).
But I do not allow less than 30 sec Api query intervals for various reasons. But of course, on triggered data updates every 3 seconds, each Api query will provide new data...

Speedliner, 2024-07-30:
@thomluther I installed version 2.0.0 of the plug-in and wondered whether there will also be switches in the future that will allow you to tell the Solarbank 2 Pro whether it should be charging or how much energy it should be feeding into the household grid, for example. So far, the options are largely limited to displaying the current values and changing the mode, could this change in the future?

thomluther, 2024-07-30:
@Speedliner , I don't think there is a way to adjust how much the SB2 is charing.
The only thing that can be adjusted is the usage mode (which is supported by the 2.0.0 release), and for the manual mode you can create/adjust time schedules with how much power to export to the house for defined intervals.
This is something that will be supported on a future release in a couple of weeks, similar to how the SB1 output power can be adjusted (via direkt entity which adjust the current time slot in the schedule, as well as services to adjust other schedule intervals as needed)

hypery2k, 2024-08-03:
I had the issue with no update in the background also. Had to leave the app open to get updates. Also had no history in the Anker App.

The support told me to delete the device and re-add it. That helped. Everything now works.

HTH

Speedliner, 2024-08-03:
> @Speedliner , I don't think there is a way to adjust how much the SB2 is charging. The only thing that can be adjusted is the usage mode (which is supported by the 2.0.0 release), and for the manual mode you can create/adjust time schedules with how much power to export to the house for defined intervals. This is something that will be supported on a future release in a couple of weeks, similar to how the SB1 output power can be adjusted (via direkt entity which adjust the current time slot in the schedule, as well as services to adjust other schedule intervals as needed).
> 
> Beside that, I don't know what other options can be changed via the Api. You can explore if there is more available with a [new endpoint that still has to be explored by the community.](https://github.com/thomluther/anker-solix-api/issues/107)

Perhaps I have expressed myself badly - what I am actually looking for is the "System Output Preset" setting, which could at least tell Solarbank 1 how much it should currently be feeding in. I hope that I can also set this for Solarbank 2 so that I can map a zero feed-in.

thomluther, 2024-08-03:
Yeah, there seem to be a couple of people having such issues. I assume there is maybe an important configuration attribute (like timezone setting) that is synchronized and set ONLY during initial setup of the Solarbank.
Since another use had a time lag of 2 hours (CET Summertime to GMT time), I assume that such time discrepancies between the Solarbank and the cloud may cause all this weird behavior. An if the timezone is just configured during initial setup (which may be wrong from time to time), a subset of users could be affected by it and have to do all setup steps from scratch...

thomluther, 2024-08-03:
> Perhaps I have expressed myself badly - what I am actually looking for is the "System Output Preset" setting, which could at least tell Solarbank 1 how much it should currently be feeding in. I hope that I can also set this for Solarbank 2 so that I can map a zero feed-in.

This will come in a future release when the new SB2 schedule modifications are fully supported for individual changes as well as for services modifying the schedule...

Speedliner, 2024-08-03:
Good news, glad to hear. many thanks!

thomluther, 2024-08-10:
Output preset and schedule updates supported with pre release 2.0.2

closed 2024-08-10 by thomluther