thomluther/ha-anker-solix #94: Unable to login due to failed requests /compatible/get_ota_update and /device/get_device_home_load

GRNspz, 2024-07-24:
### System Health details

## System Information

version | core-2024.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4869
Installed Version | 1.34.0
Stage | running
Available Repositories | 1384
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 13. Juli 2025 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-20.ui.nabu.casa
certificate_status | ready
instance_id | f402486bbba142c5832c02848b088d06
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 28.0 GB
disk_used | 9.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | green
[... 178 lines cut ...]
2024-07-24 10:57:00.515 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'power_cutoff_data': [{'id': 1, 'is_selected': 1, 'output_cutoff_data': 10, 'lowpower_input_data': 5, 'input_cutoff_data': 10}, {'id': 2, 'is_selected': 0, 'output_cutoff_data': 5, 'lowpower_input_data': 4, 'input_cutoff_data': 5}]}, 'trace_id': '00722a935dc621b09ba29bad8dbb85b1'}
2024-07-24 10:57:00.515 DEBUG (MainThread) [custom_components.anker_solix] Getting OTA update info for device
2024-07-24 10:57:00.517 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/app/compatible/get_ota_update
2024-07-24 10:57:00.517 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '42###masked###d51a###masked###e095###masked###2f', 'gtoken': '35###masked###5ed1###masked###8b'}
2024-07-24 10:57:00.518 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'device_sn': 'APCGQ80E22400516', 'insert_sn': ''}
2024-07-24 10:57:00.952 DEBUG (MainThread) [custom_components.anker_solix] sas*** request post https://ankerpower-api-eu.anker.com/power_service/v1/app/compatible/get_ota_update response received
2024-07-24 10:57:00.952 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 24 Jul 2024 08:57:00 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '84', 'Connection': 'keep-alive', 'Bizcode': '-1', 'Trace': '492bfcb13d666009f8a8fca818e9b1d3', 'Traceparent': '00-492bfcb13d666009f8a8fca818e9b1d3-bf83ce03e5101076-00', 'Server': 'APISIX')>
2024-07-24 10:57:00.953 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': -1, 'msg': 'Failed to request.', 'trace_id': '492bfcb13d666009f8a8fca818e9b1d3'}
2024-07-24 10:57:00.953 DEBUG (MainThread) [custom_components.anker_solix] Getting inverter settings for device
2024-07-24 10:57:00.954 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/app/compatible/get_compatible_solar_info
2024-07-24 10:57:00.955 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '42###masked###d51a###masked###e095###masked###2f', 'gtoken': '35###masked###5ed1###masked###8b'}
2024-07-24 10:57:00.955 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'solarbank_sn': 'APCGQ80E22400516'}
2024-07-24 10:57:01.290 DEBUG (MainThread) [custom_components.anker_solix] sas*** request post https://ankerpower-api-eu.anker.com/power_service/v1/app/compatible/get_compatible_solar_info response received
2024-07-24 10:57:01.291 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 24 Jul 2024 08:57:01 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '220', 'Connection': 'keep-alive', 'Bizcode': '0', 'Trace': 'e4cebf670117930ebe28a69a60fd401c', 'Traceparent': '00-e4cebf670117930ebe28a69a60fd401c-895dcfd565ae59b9-00', 'Server': 'APISIX')>
2024-07-24 10:57:01.291 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 0, 'msg': 'success!', 'data': {'brand_id': '8bd6d6f5-2485-4995-80ee-97e4eca9eff9', 'solar_brand': 'Others', 'solar_model': 'Others', 'solar_sn': '', 'solar_model_name': 'Others'}, 'trace_id': 'e4cebf670117930ebe28a69a60fd401c'}
2024-07-24 10:57:01.292 DEBUG (MainThread) [custom_components.anker_solix] Getting schedule details for device
2024-07-24 10:57:01.293 DEBUG (MainThread) [custom_components.anker_solix] Request Url: POST https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_home_load
2024-07-24 10:57:01.293 DEBUG (MainThread) [custom_components.anker_solix] Request Headers: {'Content-Type': 'application/json', 'Model-Type': 'DESKTOP', 'App-Name': 'anker_power', 'Os-Type': 'android', 'Country': 'DE', 'Timezone': 'GMT+02:00', 'x-auth-token': '42###masked###d51a###masked###e095###masked###2f', 'gtoken': '35###masked###5ed1###masked###8b'}
2024-07-24 10:57:01.294 DEBUG (MainThread) [custom_components.anker_solix] Request Body: {'site_id': 'f408434b-5dec-465a-819b-d18728de64f2', 'device_sn': 'APCGQ80E22400516'}
2024-07-24 10:57:01.642 DEBUG (MainThread) [custom_components.anker_solix] sas*** request post https://ankerpower-api-eu.anker.com/power_service/v1/app/device/get_device_home_load response received
2024-07-24 10:57:01.642 DEBUG (MainThread) [custom_components.anker_solix] Response Headers: <CIMultiDictProxy('Date': 'Wed, 24 Jul 2024 08:57:01 GMT', 'Content-Type': 'application/json; charset=utf-8', 'Content-Length': '87', 'Connection': 'keep-alive', 'Bizcode': '10004', 'Trace': '7327c109aeb1d1e343ddaa571e7845c7', 'Traceparent': '00-7327c109aeb1d1e343ddaa571e7845c7-8ea5c956b285649d-00', 'Server': 'APISIX')>
2024-07-24 10:57:01.642 DEBUG (MainThread) [custom_components.anker_solix] Response Data: {'code': 10004, 'msg': 'Failed to request.', 'trace_id': '7327c109aeb1d1e343ddaa571e7845c7'}
2024-07-24 10:57:01.643 ERROR (MainThread) [custom_components.anker_solix] (10004) Anker Api Error: Failed to request.
2024-07-24 10:57:01.643 ERROR (MainThread) [custom_components.anker_solix] Response Text: {"code":10004,"msg":"Failed to request.","trace_id":"7327c109aeb1d1e343ddaa571e7845c7"}
2024-07-24 10:57:01.643 DEBUG (MainThread) [custom_components.anker_solix] Finished fetching anker_solix_sas*** data in 3.722 seconds (success: False)

thomluther, 2024-07-24:
Can you confirm the version of the integration?
Is this the new 2.0.0 pre-release already? Or the old release that does not support Solarbank 2 yet?

GRNspz, 2024-07-24:
It is version 1.3.2. Nevermind - I thought it would be the most recent version but I should have activated "show beta versions" before downloading. Gonna try again with 2.0.0 and give an update in a second.

GRNspz, 2024-07-24:
Works like a charm! Sorry to bother you, and thank you so much for your invaluable work! Have a great day!