thomluther/ha-anker-solix #96: Potential to add support for F3800?

shadshack, 2024-07-25:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I added the integration to HA and authenticated, and it recognizes the F3800 device on my account, but the only sensors it has is firmware version and wifi connection.

### Describe the solution you'd like

It would be useful if this could pull info about the battery state of charge and power input/output wattages.

### Describe alternatives you've considered

No other Anker Solix integrations that I'm aware of.

### Additional context

.

thomluther, 2024-07-25:
@shadshack 
standalone devices do not provide data via cloud Api, see #31
Also the F3800 does not provide data even if attached to a Power Panel and is therefore configured into a system.
See [here ](https://github.com/thomluther/anker-solix-api/issues/117)

If you find a way how you get more data out of the system via the api library, you can document the queries here and I can see how to add support for it

thomluther, 2024-09-14:
> Any update?

@sierralx Are you able to explore additional Api charging_* endpoints for your account as mentioned [here](https://github.com/thomluther/anker-solix-api/issues/117)?
If nobody finds valid PPS consumption data in the cloud, there is nothing that can be added to the integration.

thomluther, 2025-09-29:
If anybody here is still interested in F3800 data, you may look into [MQTT data for your device and start to decode those data](https://github.com/thomluther/ha-anker-solix?tab=readme-ov-file#mqtt-data-from-devices).