thomluther/ha-anker-solix #97: Solix Smart Meter

ThorbenK13, 2024-07-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

For the HA Energy Board the House Consumption is needed. In my Case the Smart Meter measures it. But i can´t find it as a Device or Sensor in HA.

### Describe the solution you'd like

A Integration for the Smart Meter as a Sensor

### Describe alternatives you've considered

Installing a Shelly Meter parallel...but it´s alsready tight in my Fusebox.

### Additional context

Let me know if you have any Ideas ;)

thomluther, 2024-07-28:
Did you check the 2.0.0 announcement #93 ?

thomluther, 2024-07-30:
Implemented with 2.0.1