thomluther/ha-anker-solix #100: Energy dashboard help

asemev, 2024-07-30:
### System Health details

## System Information

version | core-2024.7.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1459
Downloaded Repositories | 38
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 219.3 GB
disk_used | 73.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.18.0), File editor (5.8.0), Frigate (Full Access) Beta (0.14.0-rc1), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.1), NGINX Home Assistant SSL proxy (3.9.0), Samba share (12.3.1), Spotify Connect (0.13.0), WireGuard (0.10.2), AdGuard Home (5.1.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 27
views | 7
mode | storage

</details>

<details><summary>Miele</summary>

component_version | 2024.5.0
-- | --
reach_miele_cloud | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20 July 2024 at 14:11
-- | --
current_recorder_run | 30 July 2024 at 13:18
estimated_db_size | 920.84 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hi,

Firstly, apologies for using the bug report for requesting help. i could not find a dedicated option to request help.

I'm struggling setting up the Home Assistant Energy dashbaord using the Anker Solix Solarbank 2 Pro system to view the energy data correctly.

This is what i see. For some reason there is a huge spike from 00:00 to 01:00 hours.

[image]

These are the entities i used to setup the energy dashboard:

[image]

What am i doing wrong?

Thank you!

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Not sure this is needed?
```

thomluther, 2024-07-30:
@asemev 
Those energy sensors are daily totals and reset frequently, but there is no guarantee that the cloud also responds erroneous values from time to time or completely offset peeks.
Looking at the documentation it seems you are right to change their state class to total_increasing so they sum up adequately in energy dashboard...
https://developers.home-assistant.io/blog/2021/09/20/state_class_total/
I will change that with the next release

**However:**
The integration entities do **NOT** have the goal to be used directly in the energy dashboard. And I would never recommend to use direct integration or devices entities in the dashboard for the simple reason that you loose their (long term) history if you have to replace those entities for any reason (breaking changes, integration change, device change etc).
Furthermore those energy data from the cloud could be wrong, have peek offset etc and will screw up your energy statistics completely by incorrectly interpreted meter cycle resets!

The proper usage of the Solarbank values in the energy dashboard is discussed in detail here in discussion #16
Please have a look there on how to integrate it properly with helpers and template entities.
The safer way is to use integral helpers for your power values to get total increasing energy entities (which never reset) for use in the energy dashboard.

thomluther, 2024-07-30:
Changed with 2.0.1

closed 2024-07-30 by thomluther

asemev, 2024-07-30:
Thank you very much for your guidance Thomas!

asemev, 2024-07-30:
May i ask a big favor? I want to follow your instructions but i dont know how.

So when you say "I recommend to use only template or helper entities from real devices or integration entities in the energy dashboard", how can i do this (using a template or helper)?

Could you please give an example?

Many thanks again.

thomluther, 2024-07-30:
> May i ask a big favor? I want to follow your instructions but i dont know how.
>
> So when you say "I recommend to use only template or helper entities from real devices or integration entities in the [...]

[This is my recommended approach](https://github.com/thomluther/hacs-anker-solix/discussions/16#discussioncomment-9968884).
But there are a lot of examples how people created those helpers and templates