thomluther/ha-anker-solix #102: Refresh time with Anker Solix 2 Pro

Tetraplegie, 2024-08-02:
### System Health details

## System Information

version | core-2024.7.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1394
Downloaded Repositories | 10
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 219.4 GB
disk_used | 6.1 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (6.3.1), Advanced SSH & Web Terminal (18.0.0), File editor (5.8.0), Ring-MQTT with Video Streaming (5.6.7), Mosquitto broker (6.4.1), Cloudflared (5.1.15)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 1
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23. Juli 2024 um 22:00
-- | --
current_recorder_run | 2. August 2024 um 00:00
estimated_db_size | 86.08 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello, I have added the Anker Solix 2 to my Home Assistant a few days ago. Now to my problem. The entities are not updating. I have the same values displayed for several hours.

### Reproduction steps

no Refersh 

### Debug logs

```text
??
```

thomluther, 2024-08-02:
Hi, you may have to do a total reset of your solarbank if it does not updates the values in the cloud.
[Have a look at this discussion with another user...](https://github.com/thomluther/hacs-anker-solix/discussions/93#discussioncomment-10179730)

Tetraplegie, 2024-08-02:
> Hi, you may have to do a total reset of your solarbank if it does not update the values in the cloud. [Have a look at this discussion with another user...](https://github.com/thomluther/hacs-anker-solix/discussions/93#discussioncomment-10179730) That is nothing the integration can change. It only displays the values returned from the cloud server... These values you can also query directly with the the [service Get System Info](https://github.com/thomluther/hacs-anker-solix/blob/main/INFO.md#get-system-info-service). This response will be used to update most of the entities.

In the Anker app, everything works without problems; the issue only exists in HA. A reset did not help.

thomluther, 2024-08-02:
That does not mean anything if it works in the app since that has more capabilities than just the cloud api.
Are you seeing total statistics?
What do you receive with the [service Get System Info](https://github.com/thomluther/hacs-anker-solix/blob/main/INFO.md#get-system-info-service)?
Are the data valid coming from the cloud?

The integration can just display what it gets and as long as there are no errors, there is nothing that can be changed in the integration to resolve your problem...

Tetraplegie, 2024-08-02:
> That does not mean anything if it works in the app since that has more capabilities than just the cloud api. Are you seeing total statistics? What do you receive with the [service Get System [...]

Statistics are not displayed for me in HA.

thomluther, 2024-08-02:
Probably because you did exclude them in the config options (default).
You should really the [INFO ](https://github.com/thomluther/hacs-anker-solix/blob/main/INFO.md#data-refresh-configuration-options )first... It is all described there....
I really cannot do anything about this when you don't get the values from the cloud or don't have any errors in the log

closed 2024-08-10 by thomluther