thomluther/ha-anker-solix #107: Solarbank 2 E1600 Pro 'home load preset' minimum

andythomas, 2024-08-12:
### System Health details

## System Information

version | core-2024.8.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 30.8 GB
disk_used | 21.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Samba share (12.3.2), ESPHome (2024.7.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 3
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 3, 2024 at 06:26
-- | --
current_recorder_run | August 12, 2024 at 18:48
estimated_db_size | 1009.17 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The Solarbank 2 E1600 Pro original app allows to set the output power of the battery as low as, e.g., 50W. The HA integration only allows 100W as the minimum. 

### Reproduction steps

1. Developer tools
2. Actions
3. Anker SOlix: Update Solarbank Schedule
4. Home load preset
- 100 W works fine
- 90 W or even less does not.

### Debug logs

```text
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:307
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 18:48:50 (4 occurrences)
Last logged: 19:50:22

[139912489348848] Error handling message: not a valid value for dictionary value @ data['appliance_load']. Got None (invalid_format) Andy from 192.168.63.54 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36)
[139912474413664] Error handling message: not a valid value for dictionary value @ data['appliance_load']. Got None (invalid_format) Andy from 192.168.63.54 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/127.0.0.0 Safari/537.36)
```

thomluther, 2024-08-13:
Can you please confirm the integratino version you use?
The latest 2.0.2 pre-release should not longer have this limit for SB2 and previous versions did not support the home load pre-set at all for SB2

andythomas, 2024-08-13:
[image]

I downloaded the Git zip and manually moved it to HA if that matters, i.e. I am not using HACS.

andythomas, 2024-08-13:
If it helps, I tried to manually change the minimum entries in `custom_components/anker_solix/services.yaml` but to no avail. Otherwise, already the HA GUI indicates via a red bar underneath the entry field that the value is not allowed.

thomluther, 2024-08-13:
I think I found the issue. It should be indeed in the services.yaml.
You can validate it with the system output preset entity. If you can change this directly to 0-100 W and this change is applied to current schedule time slot without error, then the limit problem is only with the services definitions in services.yaml.

The problem is that the service field limit definition cannot be made dynamic, but has to use the outer limits that should be possible for the field. And those fields have to support SB1 and SB2 limits, which may be different.
When the service is applied, the value will also be checked against the dynamic limits of the entity itself and eventually be reduced with those services to stay within appropriate limits.

To fix it, you can manually update the `services.yaml` in 2 places where the `appliance_load` field is defined:
```
    appliance_load:
      required: false
      example: 100
      default: 100
      selector:
        number:
          min: 100
          max: 1600
          step: 10
          unit_of_measurement: "W"
```
Change the min: from 100 to 0 for both services
**Afterwards you have to restart HA, otherwise the change is not picked up for the integration.**

andythomas, 2024-08-13:
> You can validate it with the system output preset entity.

I did this the following way:
[image]

but the set state did _not_ change the value looking in the integration. That fits to my observation where I did

> manually update the services.yaml in 2 places where the appliance_load field is defined:

yesterday, including the restart of HA, but could not make it work. It work accept the number, i.e. no more red bar, but the value did not change either.

thomluther, 2024-08-13:
I'm not sure if this method does trigger really a change in the number entity.
You can go to your solarbank device in the integration and then change the slider to any value in the allowed range.
As you see in the attributes of the entity, you can change it between 0W and 800W in steps of 10

andythomas, 2024-08-13:
Now, I believe there are 2 independent things.

1. The developer tools action has indeed min max values and I can easily adjust these to my liking.
2. The home preset sometimes changes the DC output and sometimes does not. This does not depend on the value.

Two non-working examples to visualize. 

[image]

[image]

I read somewhere in your instructions that you advise against the slider, since it update too often, basically every time and slides a little bit further. Is there another way of trying to change the value? If it sometimes registers and sometimes does not, I could also have been unlucky with the developer 'set change' method.

andythomas, 2024-08-13:
I apologize, it seems that I am too impatient: Please see to updated values, it just needed 4 min to register. 

[image]
[image]

thomluther, 2024-08-13:
Changing an entity value in the developer tools is not the right way to adjust them. That is the reason why other dependent entities may not be updated properly, and that the required action to apply the change is not executed. You just change the active object values in the HA state machine, that's it.
The slider is the recommended UI form, since it applies the value only when released. While the input number field applies the value each time when you click the up or down arrow, which will stress the Api to much when multiple steps are changed. Of course you can also enter the new value manually in that case, but risk is high, that users click the arrows multiple times for multiple step changes and each step would trigger a change to the entity and therefore Api calls.

thomluther, 2024-08-13:
> I apologize, it seems that I am too impatient: Please see to updated values, it just needed 4 min to register.
> <img alt="Anker5" width="285" src="https://private-user-images.githubusercontent.com/995840/357387308-59866ae2-ddb9-41ad-aeaf-3611c65e3ef2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NDc4NjUsIm5iZiI6MTcyMzU0NzU2NSwicGF0aCI6Ii85OTU4NDAvMzU3Mzg3MzA4LTU5ODY2YWUyLWRkYjktNDFhZC1hZWFmLTM2MTFjNjVlM2VmMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxM1QxMTEyNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05N2FjYzNiMDllZjgxZmQwYjBkN2I4YzM4MWYxNDhkMDM3YmZhNjQ4MjQzMjc3Y2Y2ODY5ZTMxZGI0ZDYxNmI3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EoYfmAfjn8dOai2Vma6a8sTMumil83VHIqSqRC21z04"> <img alt="Anker6" width="296" src="https://private-user-images.githubusercontent.com/995840/357387335-1aba894e-da62-438d-8561-70fab5d205c2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NDc4NjUsIm5iZiI6MTcyMzU0NzU2NSwicGF0aCI6Ii85OTU4NDAvMzU3Mzg3MzM1LTFhYmE4OTRlLWRhNjItNDM4ZC04NTYxLTcwZmFiNWQyMDVjMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxM1QxMTEyNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNzQzNmU0ZjM1NWQ5OTU5ZTRhYTIzZTZiMDMzMzNmODA1MGFhZTE0OTU3M2FiYmVjMWE2MGM4YTI1ZTBlMDQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.cy_3qiU3JniuYmxL_UZZeJ4iQ_rpgc6XZSKGSQ3ZYug">

When you change the preset via the UI or service (not developper tools), the change of the other entities should also be immediately applied. 
The Home preset sensor should be changed directly with the preset change, the DC output power will change only when the change was applied by the system AND the solarbank provided new values to the cloud (which happens only every 5 minutes for SB2) and a new Api refresh was done (every 60 seconds per default)
So yeah, a delay of up to 6 minutes are possible for the final confirmed sensor values from the SB2.
Actually I'm not sure if that is also valid for the Home Preset sensor, and whether that changes back to previous value until SB2 sent the new data to the cloud. The sensors extracted from the Api scene info will only update once the SB2 sent updated data (delay up to 6 minutes).
The sensors extracted from the Solarbank schedule should remain actual value once change was applied via HA.

andythomas, 2024-08-13:
Thank you for the detailed instructions and explanations. Updates every approx. 5min is more than good enough for (my) every day use. Please close at your own discretion.

This is a very nice integration!

thomluther, 2024-08-13:
Thanks
Please confirm that the change via service is working after you applied the suggested changes in the services.yaml.
Then I will change that for the final release.
Thx

andythomas, 2024-08-13:
> Please confirm that the change via service is working after you applied the suggested changes in the services.yaml.
Then I will change that for the final release.

Even after the two changes in `services.yaml` 

```
    appliance_load:
      required: false
      example: 100
      default: 100
      selector:
        number:
          min: 0
          max: 1600
          step: 10
          unit_of_measurement: "W"
```

it does not work and responds

`Failed to perform the action anker_solix.update_solarbank_schedule. not a valid value for dictionary value @ data['appliance_load']. Got None`

andythomas, 2024-08-13:
It works like a charm now. Thank you for all the help.

thomluther, 2024-08-16:
This is fixed in the main branch, will be available with final 2.1.0 release

thomluther, 2024-08-16:
Re-opening for reference until final release 2.1.0

thomluther, 2024-08-21:
Fix released with 2.1.0

closed 2024-08-21 by thomluther