thomluther/ha-anker-solix #110: Ability to change the used account for present integrations

nlincke, 2024-08-15:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Not that i know of.

### Describe the solution you'd like

In case a integration is already present, it would be great if i could replace the used account via the Integration settings ....

### Describe alternatives you've considered

Deleting the integration and create a new one with the new account. I guess this could mean that the used entity IDs will be new one (at least if i create a new connection before i delete the present one).

### Additional context

Nothing to add. 
Perhaps it could be an option within Configuration (Konfiguration):
[image]

thomluther, 2024-08-16:
Hi, there is no special way in HA to handle that.
In the backend, it would be the same as deleting the existing configuration and create a new one with the new account
When systems and devices are shared, the old entities will be re-used since the unique ID is composed out of the device SNs and Site ID, which will not change when switching accounts

thomluther, 2024-08-16:
Actually HA 2024.04 implemented a reconfiguration step for integrations:
https://developers.home-assistant.io/blog/2024/03/21/config-entry-reconfigure-step/

Need to evaluate if that may be an option to change the account for a configuration entry.
While it would simplify the user experience, it would have to do the same in the backend (unload the actual configuration and associated devices and entities and reload it with the new account if that is possible)

thomluther, 2024-08-21:
Account reconfiguration capability added with release 2.1.0

closed 2024-08-21 by thomluther