thomluther/ha-anker-solix #112: Custom schedules

hypery2k, 2024-08-19:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

First, thanks for this plugin. Really awesome work.

My question is: What do you think about adding own preset schedules based on sensor values?

### Describe the solution you'd like

My idea is: Adding a "new schedule" calling home-assistant und adjust the output to home based on the values from smart plugs, basically like with Anker ones, but not bound the this vendor. 

I would also volunteering for making a PR if you like the idea.

### Describe alternatives you've considered

Anker Plugs :D

### Additional context

none

thomluther, 2024-08-19:
If I understand your request right, this is not what the integration is supposed to do, because that is core HA capability.
You can create any automation that does what you want, triggered by any sensor state and using the integration services to create whatever schedules you need.

hypery2k, 2024-08-19:
thanks for the fast response, I didn‘t wanted to replicate HA functionality. I just wasn‘t aware that schedule modification is already available. I will try that

closed 2024-08-19 by hypery2k