thomluther/ha-anker-solix #115: C1000X support

kaiwoknaj, 2024-08-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I have added my Anker c1000 but i cant see the SOC, it it super limited and Im only able to see firmware and wifi status
[image]
[image]

### Describe the solution you'd like

See SOC and charging status

### Describe alternatives you've considered

—-

### Additional context

[image]
[image]

thomluther, 2024-08-29:
Hi @kaiwoknaj , unfortunately no Power Station is supported actually via the cloud Api.
All data that you can see in the App device details is only provided to the MQTT cloud servers and probably only as long as you access them via the app.
There is no interface to the MQTT cloud available.

The Api cloud only supports devices that can be added to a Power System which you need to create in the app.
And even then it can be that the device consumption data are not provided via the Cloud Api and only merged by the App from the MQTT servers.
See 
- Potential to add support for F3800? #96 
- Solix F3800 has no useful sesnors #31 
From Api library repo:
- [Add F3800/BP3800 Equipment when connected to Home Power Panel - Config Example #117](https://github.com/thomluther/anker-solix-api/issues/117)
- [Question: Is it possible to use this API by prometheus and scrape data to Grafana (For example for Anker Solix C800X) #123](https://github.com/thomluther/anker-solix-api/issues/123)

MrHollowPS, 2024-10-06:
This would be interesting for me as well, i got the C1000X on purpose to use as a home battery backup with your integration and now i noticed it doesn't work 🤦

thomluther, 2024-10-07:
> This would be interesting for me as well, i got the C1000X on purpose to use as a home battery backup with your integration and now i noticed it doesn't work 🤦

Hi @MrHollowPS , The Readme clearly shows which devices are supported by the Library and HA Integration.
A portable power station was never supported. This will need support from the community to find a way where their consumption data is available in the cloud and document that accordingly. But to my knowledge, any device not configured into a power system will NOT send consumption data by any means to the cloud regularly. Only when the App is accessing the device real time charts, the devices are triggered to send real time data to the MQTT server while the charts have the be updated. So no regular reporting via the cloud.

MrHollowPS, 2024-10-13:
Anyway i can look into this? I was thinking of reading the data from the bluetooth but i couldn't fins any uuid for these values.

sircambridge, 2025-04-26:
i would very much like to have the ability to turn on/off the AC output on my c1000. sadly it sounds like there is no official api solution

thomluther, 2025-04-26:
Correct. You would have to sniff traffic of the App when switching your device. If it does not communicate with the Api servers, nothing is done via the cloud Api.
If it does, you would need to know which Api endpoints are used at least, then you could explore them with the Api library or the HA integration.
The payload of the Api is encrypted with ECDH assymetric key exchange between client and server, so no chance to sniff and decrypt it since you have no access to private key of the App client.

khartahk, 2025-11-12:
Just pulled and ran the following:

```sh
docker compose build
docker compose stop
docker compose start -d
```

And here is the log file: [logs.txt](https://github.com/user-attachments/files/23510552/logs.txt)

And now the Expansion pack is showing 😄 

[image]

000al000, 2025-11-14:
Try to run, but got error

 ImportError: cannot import name 'create_mqtt_device' from 'api.mqtt_factory' (/usr/local/lib/python3.12/site-packages/api/mqtt_factory.py)
anker-solix-mqtt    | Traceback (most recent call last):
anker-solix-mqtt    |   File "/app/ha_mqtt_publisher.py", line 31, in <module>
anker-solix-mqtt    |     from api.mqtt_factory import create_mqtt_device

ohadlevy, 2025-11-14:
@000al000 how are you running it? directly from git or via the container?

thomluther, 2025-11-14:
So the question is now, which field actually identifies the number of installed expansion packs? 
This is required to calculate the capacity and remaining Wh of the device.

ohadlevy, 2025-11-14:
> So the question is now, which field actually identifies the number of installed expansion packs? This is required to calculate the capacity and remaining Wh of the device.

see https://github.com/ohadlevy/ha-anker-mqtt/blob/master/ha_mqtt_publisher.py#L1327

MrHollowX, 2025-11-14:
@ohadlevy thanks a lot for this 

@thomluther I see that the PR is merged, however i cannot see anything useful. Am I missing something?

[image]

EDIT: Nevermind, i noticed the milestone is due by the end of the year :(

thomluther, 2025-11-15:
> [@thomluther](https://github.com/thomluther) I see that the PR is merged, however i cannot see anything useful. Am I missing something?

Which PR is merged?
The ha-anker-mqtt is completely different concept than the HA integration and cannot be re-used by the integration. 
It follows the concept of separat VM with MQTT broadcast using HA autodiscovery topics to have the MQTT entities created automatically. 
The HA integration does not make any use of it, it pulls its own data from the cloud (Api and MQTT) and directly integrates with HA. It does not need the overhead of additional VM and MQTT proxy, and can directly integrate with HA core functionality. This is not possible with an external VM or MQTT proxy.

The only thing both share is the Api libary code, but this will still change significantly for the MQTT classes to make it scalable for the broad Solix device landscape. I want to get rid of the dedicated C1000 classes since that concept does not scale. It was just a quick implementation to test the MQTT concept and controls via the Anker MQTT server, but its nothing the can easily be re-used in a modular way for other device types.
Therefore any tools and integration code that is built on the dedicated C1000 class will break in future once the final MQTT device handling is implemented in the library.

thomluther, 2025-11-15:
@MrHollowX , @khartahk  what is still missing for C1000 are some MQTT commands and fields which must be discovered by device owners. If you want to contribute, please follow [this discussion](https://github.com/thomluther/anker-solix-api/discussions/221#discussioncomment-14948605).

thomluther, 2025-12-15:
Partial MQTT data have been implemented with 3.4.0 for F3800(P) models.
If you miss data, you need to analyze and describe them first for your device model.
see [MQTT connection and integration](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#mqtt-connection-and-integration)

Commands are planned with 3.5.0

thomluther, 2025-12-15:
What is wrong, if you have real capacity of 1056 x 2 and the integration shows 2112 Wh?
For the state of charge, you need to analyze the values in the mqtt messages.
As I said, I cannot fix that since I don't know what is provided by the device and which fields show the correct values that match with the App.

thomluther, 2025-12-15:
Or at the minimum, you need to provide a system export zip file including the MQTT messages, which you take at a time where your batteries show different SOC.
You need to provide the SOC values at the time you took the export, otherwise they cannot be identified correctly in the exported MQTT messages.

khartahk, 2025-12-15:
I have the extension battery attached and for me it show properly. I'll try to check in a couple of hours and send the messages here. Can you just point me to where I can find them please?

khartahk, 2025-12-15:
Here's the dump. I have 1 expansion pack as mentioned before and:
- the expansion battery was charging
- was arround 10-15% at the time when this was dumped
- it was charging at arround 600W
- ultrafast chraging was enabled (I think the expansion can't charge at 1200 as the main one can when ultrafast is enabled)
- the main battery was at 100%.

Nothing else was plugged in and it was connected to WIFI.

[A1761_mqtt_dump_2025-12-15__22_31_10.txt](https://github.com/user-attachments/files/24175869/A1761_mqtt_dump_2025-12-15__22_31_10.txt)

khartahk, 2025-12-16:
I'll remove the expansion pack today and rerun the script to see what's different. Will attach the changed file when I have it.

thomluther, 2025-12-16:
Thanks for the data.
It looks like the deterministic code for the expansions is wrong, which is used if no expansion number is extracted only.
I fixed that.
For your device, it looks like c5 is the correct expansion field, so the deterministic code is not needed at all for the device.

Yet, it would be good if you can determine what c0 reflects, that may see values 0, 1 or 2 so far.
Is that maybe an overall charging state like inactive, charge, discharge, similar to what is used for the USB ports?
2 could be charging, since you had grid to battery power at that time. You can determine that only in live mode of the messages, while battery is used under various conditions.

thomluther, 2025-12-18:
> <img alt="Image" width="327" height="145" src="https://private-user-images.githubusercontent.com/2655332/528126247-5db1d088-8b08-4ec0-8378-1044e5ae6f77.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjYwNzI5MjIsIm5iZiI6MTc2NjA3MjYyMiwicGF0aCI6Ii8yNjU1MzMyLzUyODEyNjI0Ny01ZGIxZDA4OC04YjA4LTRlYzAtODM3OC0xMDQ0ZTVhZTZmNzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTIxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEyMThUMTU0MzQyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzU0YWZmYzlmZDdjZjMwN2QzMjJmNmVjYTRlMTUwYjhlMDIxYjQxZjVjYjljNDM2MTgwZjkxMDVhY2NhMThiZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.nI-9trpSkZonPdKWfgYkZoK4_VqFUtSGM17QuCdiJyc">
> 
> I think both or at least the combined percentage should be there instead.

I have no idea what is combined by HA for the device, if there are multiple entities with device class battery. But that's the way it is and it is just a HA display for the device, so ignore if not correct.

thomluther, 2025-12-24:
My next work item will be to implement a generic command executer in the normal monitor tool.
This will allow user selections for commands, parameters and values based on described and supported controls for the device.

closed 2026-01-29 by thomluther

thomluther, 2026-02-10:
> [log.txt](https://github.com/user-attachments/files/25210030/log.txt)
> 
> [@thomluther](https://github.com/thomluther) i've added the log, i've redacted some info like keys, email address and certificates

Hi
These logs don't help, they do not show anything without debug mode enabled.
Please send me a system export, that is better to get anonymized information from your account which I can use to investigate. You should probably open a new issue, there are instructions for the system export.

thomluther, 2026-02-10:
There are no devices listed for your account. 
You use "RO" country ID. Based on issue #410 , RO has been re-assigned to the other cloud server.
Actually RO accounts cannot be assigned to both cloud servers, either the EU one or the COM server should be used when you create your account. And the county you selected at creation is relevant on which server your data are.

So with the actual integration you should be connected to the COM server. You can see which server is used in your account device details.
If you want to try the other server, you need to reconfigure your hub to any EU country.
If that will work, you have to leave it as is. There is nothing else I can do to fix it. Either I break the RO assignment for you, or for the other user who requested the change to the COM server.
One of you must have used a different country at time of account creation, otherwise it would not be explainable, why RO accounts should land on different cloud servers...

MrHollowX, 2026-02-11:
Thanks @thomluther. I don't care about which country is selected as long as it works.
I've switched to Germany and it works fine, thanks for clarifying. I tried using USA previously as I though i had the wrong continent selected when creating my account. 

Anyway it works so we can leave it as is, unless someone else from Romania comes with the same issue as I do then I suggest this is switched back to the correct server.