thomluther/ha-anker-solix #134: Can't add to custome repository with the given link

djovic78, 2024-10-09:
### System Health details

## System Information

version | core-2024.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4958
Installed Version | 2.0.1
Stage | running
Available Repositories | 1501
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 19. Oktober 2024 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | false
remote_connected | false
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-16.ui.nabu.casa
certificate_status | ready
instance_id | 75e8b7d79d5a4b9b9da257596f339875
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.10.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 57.8 GB
disk_used | 12.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (19.0.0), Let's Encrypt (5.2.1), NGINX Home Assistant SSL proxy (3.11.0), File editor (5.8.0), ESPHome (2024.9.2), Matter Server (6.6.0), openWakeWord (1.10.0), Rclone Backup (3.3.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 10
-- | --
resources | 0
views | 24
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 30. September 2024 um 10:16
-- | --
current_recorder_run | 9. Oktober 2024 um 11:52
estimated_db_size | 285.59 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Integration version

n/a can't add it

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I try to follow the HACS installation steps described in the README. When I add the custom repository with the given link (https://github.com/thomluther/ha-anker-solix) and clicking "Add" it shows up in the list. There is no save button. This was changed in a recent release. So I close the Form. Going back into the form it still shows all custom repos including ha-anker-solix. But I can't find the integration. When reloading HACS integration and going back to the custom repositories the entry is gone (only the anker-solix, the others are still there and working)

### Reproduction steps

1. In the side panel click HACS
2. Top right corner click 3 dots
3. A Popup form shows --> add the "https://github.com/thomluther/ha-anker-solix" to the Repository field
4. As Type chose Integration
5. Click "Add"
6. Pop-Up form will show the custom repo in the list as "Anker Solix - thomluther/ha-anker-solix (integration)"
7. There is no "Save" button, only "Cancel" and "Add"
8. Clicking "Add" again will give the error: "Repository 'thomluther/ha-anker-solix' exists in the store."
9. Click on either "X" or "Cancel"
10. Open 3-dots menu again: Ther repo is still listed
11. Can't find the anker integration
12. Reload HACS
13. Go back to HACS
14. Go back into the 3-dots menu
15. The repo is gone

### Debug logs

```text
don't have logs as installation fails.
```

### Diagnostics dump

[config_entry-hacs-d991eb867366277806eb61fc669cb215.json](https://github.com/user-attachments/files/17306582/config_entry-hacs-d991eb867366277806eb61fc669cb215.json)

thomluther, 2024-10-09:
Hi @djovic78 , with your step 11 Can't find the anker integration:
Where do you search for the integration?
After adding the repo to HACS, you should find the Anker Solix Integration in the HACS integration list (not the HA integration list).
Make sure you have no other HACS filters active that may prevent the integration to show up.

I'm also not sure why your HACS repo entry is gone after reloading HACS. Seems it may not be added correctly?

I have also a Pull request open to get the repo added to default HACS integration list, but I have not idea when this will be accepted. There is a huge backlock for default integration list updates in HACS...

djovic78, 2024-10-09:
Hi @thomluther .
Thank you for your reply. I did search in the HACS repo. No filters either. It is also not the first custom repository. The others did work. 
However, I think it might actually be a HACS issue. I'm not sure, though. 
But I've enable debug logging for HACS and stubled upon this:

djovic78, 2024-10-09:
HACS debug log uploaded

[home-assistant_hacs_2024-10-09T10-39-28.543Z.log](https://github.com/user-attachments/files/17310618/home-assistant_hacs_2024-10-09T10-39-28.543Z.log)

thomluther, 2024-10-09:
I cannot see anything specific to this repo that may cause issues in your setup. You have other hacs integrations that cause errors.
The last entries I see for this integration are:
```
2024-10-09 12:28:51.050 DEBUG (MainThread) [custom_components.hacs] <Integration thomluther/ha-anker-solix> Unregister stale custom repository
...
2024-10-09 12:29:00.243 DEBUG (MainThread) [custom_components.hacs] <Integration thomluther/ha-anker-solix> Checking repository.
2024-10-09 12:29:00.243 DEBUG (MainThread) [custom_components.hacs] <Integration thomluther/ha-anker-solix> Checking repository.
2024-10-09 12:29:00.890 DEBUG (MainThread) [custom_components.hacs] <Integration thomluther/ha-anker-solix> Running checks against 2.1.2
2024-10-09 12:29:02.513 INFO (MainThread) [custom_components.hacs] <Integration thomluther/ha-anker-solix> Registration completed
```

To me this looks like the repo is registered successfull in HACS, and you should be able to find the integration...

djovic78, 2024-10-11:
Hm. I'm very sure I did everything correct. But it definitely doesn't work for me.
Manually installing works, but I guess I'll have to update manually each time.

Here's a short video about the workflow: [https://youtu.be/VkDsz3oVIhI](https://youtu.be/VkDsz3oVIhI)

thomluther, 2024-10-11:
Yeah, looks strange, maybe HACS does not register it correctly although it did not show an error for the repo. But when it is gone after HACS reload, this looks like it was not registered correctly.
I could not recreate this on my system. When I remove the custom repo, I don't see the integration anymore in the HACS list and once I add it again, it directly appears again in the list. All without reloading HACS.
Maybe your aother errors have some dependencies on how hacs is behaving...

djovic78, 2024-10-16:
Hey @thomluther 
Please close this issue. I seriously didn't find out what the root cause for my problem is. But after a I added successfully a different custom repo, today I thought let's try ha-anker-solix again. And it worked as expected and it seems I was the only one having it anyways.
Must have been related to my system. Thank you for your assistance and most importantly for the amazing integration.

Cheers

thomluther, 2024-10-16:
Closing as requested (invalid)

closed 2024-10-16 by thomluther