thomluther/ha-anker-solix #143: Daily production entity

SiSc1987, 2024-10-23:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No problem, its a feature request for a new entity.

### Describe the solution you'd like

Hi,

is it possible to include an entity showing the daily production of all of my panels?
The Anker app is showing that value, but I don't know if the API is providing this information or not. I can not find any already available entity showing the value for "todays production".

Best Regards
Simon

### Describe alternatives you've considered

Doing something manually with helpers in home assistant. But I'm not an expert in it so far.

### Additional context

no further remarks.

thomluther, 2024-10-23:
Hi @SiSc1987 
You need to check your configuration entry options. The energy categories are excluded by default since they require couple of extra Api queries.
Once you remove them from the exclude list, you should get entities for the various daily energies (with attribute of previous period = yesterday)

SiSc1987, 2024-10-23:
Hi @thomluther,

got it! Thanks a lot. I was not aware, that there are entities inactive. This request can be seen as closed.

closed 2024-10-23 by SiSc1987