thomluther/ha-anker-solix #150: Add support for individual smart plug device settings (switch, timer, schedule)

thomluther, 2024-10-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The Anker Smart plugs support various settings via the App,
Switching the plug on/off
A timer for delayed switching
A schedule for the switch state

### Describe the solution you'd like

Integration entities supporting the switch control similar to what is supported by the App.

### Describe alternatives you've considered

None

### Additional context

It is not clear, whether the plug settings are supported via cloud Api.
The Api endpoints must be explored by the community to find proper queries and parameters to get the device settings and change them accordingly.
See 
- [Add support for smart plugs](https://github.com/thomluther/anker-solix-api/issues/124)
- [Explore usage of new device_attrs endpoints](https://github.com/thomluther/anker-solix-api/issues/107)

Thunder1805, 2024-11-03:
Fehler beim Einrichten: Retries exceeded: (100053) Anker Api Error: Please try again in 24 hours.

thomluther, 2024-11-03:
@Thunder1805, what does your comment mean in this feature request?
If you are trying to use anything via Api, you should use another account than in the App, otherwise they kick out eachother and after 10 times your IP is banned for 24 hours, which is what this error means.

Thunder1805, 2024-11-04:
hmm but why after installing 2.2. so many ping request happen to the Anker server that I get blocked...

sethgregory, 2024-11-23:
+1 for this request.  Have successfully connected the integration and it sees my 240W charging station but receives none of the metrics and has no ability to control power output or on/off.

thomluther, 2024-11-25:
Hi @sethgregory 
Not sure if this issue is right for you. This issue is solely for Anker Smart plug device support.
The 240W charging device is not supported at all and probably cannot be supported via cloud Api since it cannot be configured into a power system in the mobile App. This is mentioned in various places.
So the only information you will ever get from that device is what you see with the current integration, unless you find additional fields in the cloud via Api exploration that could be implemented. Then you have to document the Api requests, parameters and response examples to pick this up in the api libaray and Integration.

ovflowd, 2025-03-16:
I'd also appreciate if we could control the smart plugs. Im actually baffled there is no support for that. Id be more than happy to donate to the project in order to get this done :)

ovflowd, 2025-03-16:
@thomluther is this something you have started exploring already? Is this something complex? Something I could contribute with?

thomluther, 2025-03-17:
> [@thomluther](https://github.com/thomluther) is this something you have started exploring already? Is this something complex? Something I could contribute with?

I cannot work on this. Someone would need to explore the Api calls as system owner and test what is provided via the Api cloud and whether there is any data available or even changes possible.
I haven't figured out yet what the *_device_attrs endpoints are used for...
See https://github.com/thomluther/anker-solix-api/issues/107 and https://github.com/thomluther/anker-solix-api/issues/124

thomluther, 2025-03-17:
Please also have a look to the newly discovered endpoints https://github.com/thomluther/anker-solix-api/blob/2d2c2808323174cab7a42cde05f0c68e7e2319ed/api/apitypes.py#L291:
```
    'mini_power/v1/app/charging/get_charging_mode_list',
    'mini_power/v1/app/charging/update_charging_mode',
    'mini_power/v1/app/charging/add_charging_mode',
    'mini_power/v1/app/charging/delete_charging_mode',
    'mini_power/v1/app/setting/set_charging_mode_status',
    'mini_power/v1/app/egg/get_easter_egg_trigger_list',
    'mini_power/v1/app/egg/add_easter_egg_trigger_record',
    'mini_power/v1/app/egg/report_easter_egg_trigger_status',
    'mini_power/v1/app/setting/get_device_setting',
    'mini_power/v1/app/power/get_day_power_data',
```

ovflowd, 2025-03-17:
> I cannot work on this. Someone would need to explore the Api calls as system owner and test what is provided via the Api cloud and whether there is any data available or even changes possible.

Is there any documentation on how to debug said method calls? Should I use an Android Emulator or something? Or what exactly are the instructions here?

I'm more than happy to run said API calls and do all these things, but some instructions are needed 🙈

thomluther, 2025-03-17:
Each Api client will try its own authentication is nothing is in the authcache folder.
If you want to share the auth tokens between an active HA client and your Python client, you can copy your account file from the HA integration authcache folder in the api directory.
While this shared token file remains valid, both can use the Api in parallel. but if any of them needs to re-authenticated, it will invalidate the others token. So manual re-copy of the file may be required afterwards, or not using the same Anker account at the same time.

But be aware, that for many queries you will need the owner account since otherwise you may get request errors due to permission issues.

ovflowd, 2025-03-23:
That sounds like it is going to take some good time; I, unfortunately, am a bit short on time. I wouldn't mind sharing a "family account" to my system with you, and you could play around if you wish :)

ovflowd, 2025-03-23:
(otherwise you'd need to wait a tad bit until I get time to debug all of these)

thomluther, 2025-03-24:
Well, I'm not in a hurry. I don't need this capability or anything of the SB2 since I have just a SB1.
A family account is not good enough, since many queries need admin account as mentioned. Especially if you have to test and explore the endpoints. Family accounts have no access to any device data or details, just the system overview. Like in the App.
And I don't want to play around with admin accounts on other peoples devices for multiple reasons.

thomluther, 2025-04-10:
Update: The get_device_attrs endpoint may deliver some info from plugs...I only found following attributes list item that provides a value response for solarbank AC:
- rssi

There are more endpoints that may be usable for the plugs:
-     'mini_power/v1/app/charging/get_charging_mode_list',
-     'mini_power/v1/app/charging/update_charging_mode',
-     'mini_power/v1/app/charging/add_charging_mode',
-     'mini_power/v1/app/charging/delete_charging_mode',
-     'mini_power/v1/app/setting/set_charging_mode_status',
-     'mini_power/v1/app/egg/get_easter_egg_trigger_list',
-     'mini_power/v1/app/egg/add_easter_egg_trigger_record',
-     'mini_power/v1/app/egg/report_easter_egg_trigger_status',
-     'mini_power/v1/app/setting/get_device_setting',
-     'mini_power/v1/app/power/get_day_power_data',

You can try them with the new [2.6.0 Action for Api requests](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#api-request-action) right in HA:

ovflowd, 2025-04-11:
> Well, I'm not in a hurry. I don't need this capability or anything of the SB2 since I have just a SB1. A family account is not good enough, since many queries need admin account as mentioned. [...]

That's alright, I will attempt to investigate next week or this weekend!

> Update: The get_device_attrs endpoint may deliver some info from plugs...I only found following attributes list item that provides a value response for solarbank AC:

Ill check that out. Besides of the endpoints below, any other you want me to check? What is the full endpoint for `get_device_attrs` ?

thomluther, 2025-04-12:
I think most is described with the [Api request action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#api-request-action)

ovflowd, 2025-04-19:
I got a new phone (just for this) as I want to use HTTP Toolkit to sniff the requests, but, eh, having some good amount of issues here.

ovflowd, 2025-04-19:
> * mini_power/v1/app/charging/get_charging_mode_list
> * mini_power/v1/app/charging/update_charging_mode
> * mini_power/v1/app/charging/add_charging_mode
> * mini_power/v1/app/charging/delete_charging_mode [...]

None of these endpoints got hit whilst seeing or updating or deleting switch, timer, schedule updates. At least not to Anker servers. Apparently (my feeling) is that it might communicate locally with the devices.

ovflowd, 2025-04-19:
I see that no network requests are done when creating or updating or deleting "Timings" -- my Anker Solix is on sleep mode (since it is evening) right now, and not sure if that makes a difference.

thomluther, 2025-04-20:
I think the mini power endpoints are for the mini charger devices. Those devices cannot be added to a System...

ovflowd, 2025-06-08:
Hi @thomluther! Apologies, didn't mean to ghost this thread, but unfortunately a few days after we last chatted the system completely disappeared from my App (I assume due to the solarbank not being active anymore) and I wasn't able to get the smart plugs individually added to the App. Fast-forward I'm selling my Anker Solix system (Pretty much because my Landlord vetted its usage)

Best of wishes on getting this feature implemented!

thomluther, 2025-06-26:
Some new finding on this: 
I recent Scene Info there seems to be more data for smart plugs:
```json
        {
          "device_pn": "A17X8",
          "device_sn": "Z7TAQWPTMV3ER5TK",
          "device_name": "Smart Plug",
          "device_img": "",
          "create_time": 1747848134,
          "current_power": "0",
          "tag": "Dining Room",
          "tag_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/06/17/iot-admin/Te3LpcS7jXZhS924/icl_A17X8_location_Dining%20Room.png",
          "total_power": "",
          "bind_site_status": "",
          "status": "1",
          "wireless_type": "",
          "err_code": 0,
          "priority": 0,
          "auto_switch": false,
          "running_time": null
        },
```
priority, auto_switch and running time are new.
Can someone explain what they mean and define such settings in the plug and provide an owner system exports afterwards?
I assume the schedule and other things may now be listed at least, even if there is not a change capability yet.

derMara123, 2025-06-29:
Hi, hope this is what you need, as requested in the PV forum.

[derMara_2025-06-29_2156.zip](https://github.com/user-attachments/files/20969896/derMara_2025-06-29_2156.zip)

derMara123, 2025-06-30:
Sorry, I think I misunderstood something and yesterdays file was without active schedule and delayed action for a plug.
Did the export again today with Smart plug "TV" having an active schedule for switching as well as a delayed action running.

I have not really checked the plug options in the Anker app until now.

See Attachment.

[derMara_2025-06-30_2054.zip](https://github.com/user-attachments/files/20985997/derMara_2025-06-30_2054.zip)

thomluther, 2025-06-30:
> Did the export again today with Smart plug "TV" having an active schedule for switching as well as a delayed action running.

The info did not change:
```json
        {
          "device_pn": "A17X8",
          "device_sn": "0DQO39LCKA55CCXR",
          "device_name": "Anker Plug TV",
          "device_img": "",
          "create_time": 1748072566,
          "current_power": "0",
          "tag": "Living Room",
          "tag_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/06/17/iot-admin/gdu727accXe3HGI5/icl_A17X8_location_livingroom.png",
          "total_power": "",
          "bind_site_status": "",
          "status": "1",
          "wireless_type": "",
          "err_code": 0,
          "priority": 0,
          "auto_switch": false,
          "running_time": null
        },
```
Interestingly, there is also a new feature switch for smart plugs which shows smart_plug = False although you have plugs in the system:
```json
          "feature_switch": {
            "0w_feed": true,
            "0w_feed_v2": true,
            "exceed_power": false,
            "heating": false,
            "manual_backup_mode": true,
            "micro_inverter_output_exceed": false,
            "micro_inverter_power_exceed": false,
            "offgrid_with_micro_inverter_alert": false,
            "peak_valley_mode": true,
            "shelly_meter": true,
            "smart_plug": false
          },
 ```
This lets me assume following:
- The new smart_plug feature switch indicates whether a smart plug was added to the new smart mode as automatic plug controlled device. Therefore it is unlikely to get this feature to True in a SB2 system?
- The 3 new smart plug device values are also just used once the plug is automatically controlled by Smart mode. Following is assumed (to be proven with such an SB3 system)
  - Priority is the order how multiple plugs should be enabled if there is surplus expected (or existing)
  - auto_switch indicates, whether this plug is controlled by smart mode or not
  - running_time may indicate a duration, or be an object with multiple values

Once meaning and structure of these new values are validated by SB3 owners, I can add them to the integration as sensors (no control entities)

thomluther, 2026-01-04:
A remote desktop session won't help, its nothing what you can do in a short time frame.
You just need a PC with Python runtime and follow the installation instructions in the [Api library](https://github.com/thomluther/anker-solix-api) linked above.
Then you can use the [mqtt_monitor](https://github.com/thomluther/anker-solix-api?tab=readme-ov-file#mqtt_monitorpy) tool out of the box with your main account and monitor your devices.

thomluther, 2026-01-04:
I also expect the timer or schedule commands to be complex byte structures that must be fully understood by testing various settings and analyzing the published commands from the App.
This may be also required to get any active schedule or timer from the status messages.
The plug switch is probably the easiest command.

thomluther, 2026-01-09:
> [smart_plug_buro_switch_off_on_2026-01-08__14_46_04.txt](https://github.com/user-attachments/files/24531883/smart_plug_buro_switch_off_on_2026-01-08__14_46_04.txt) [smart_plug_buro_timing_30sec_switch_2026-01-08__14_49_40.txt](https://github.com/user-attachments/files/24531882/smart_plug_buro_timing_30sec_switch_2026-01-08__14_49_40.txt)
> 
> Here are 2 dumps. The one called "off_on" is one single switch off - first. and afterwards on. The second dumps was triggered with 30seconds to switch of, and afterwards 30seconds to on again.

I don't get that from the logs when you did what, so therefore you need to do live monitoring for the commands, which message refers to which action in the app.

Here is the timer command:
```shell
14:51:18 INFO: ---------------------- Smartplug / A17X8 / 007e / Header -----------------------
ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
28 00   : 2 Byte total message length (40) in Bytes (Little Endian format)
03 00 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
00 7e   : 2 Byte message type pattern (varies per device model and message type)
        : 1 Byte optional message increment (  0)
-> 5c <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
-- Fields --|- Value (Hex/Decode Options)---------------------------------------
Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
a1   01  --    22
└->   1 unk              b'"'
a2   02  01    01
└->   2 ui                  1               1
a3   04  04    1e:00:00
└->   4 bin   b'\x1e\x00\x00'
a4   02  01    01
└->   2 ui                  1               1
a5   04  04    00:00:00
└->   4 bin   b'\x00\x00\x00'
fe   05  03    66:c4:5f:69
└->   5 var      -15258;26975      1767883878 16907370198287517174726656.000  102;196;95;105
--------------------------------------------------------------------------------
```

a1 is a standard pattern and fe is the msg_timestamp.
a2 and a4 are probably the settings, one for off delay, and one for on delay.
a3 and a5 are probably the delays with first 2 bytes, but what is the 3rd byte?
1e:00 little endian is 001e hex => 30. That means the timer in seconds. The question is now was that the off or on timer.
And how does the other timer relate to it since it was also set with 0 seconds...

thomluther, 2026-01-09:
It seems this status is contained in 0405, so no need to query it extra.
However, the structure puzzles me

Command with one 30 sec time, not sure what is toggled. I assume the first setting is the toggle_to setting, the second delay and setting is for toggle back? If that is the same as toggle_to, it simply does not toggle back?
:
```shell
14:51:18 INFO: ---------------------- Smartplug / A17X8 / 007e / Header -----------------------
ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
28 00   : 2 Byte total message length (40) in Bytes (Little Endian format)
03 00 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
00 7e   : 2 Byte message type pattern (varies per device model and message type)
        : 1 Byte optional message increment (  0)
-> 5c <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
-- Fields --|- Value (Hex/Decode Options)---------------------------------------
Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
a1   01  --    22
└->   1 unk              b'"'
a2   02  01    01
└->   2 ui                  1               1
a3   04  04    1e:00:00
└->   4 bin   b'\x1e\x00\x00'
a4   02  01    01
└->   2 ui                  1               1
a5   04  04    00:00:00
└->   4 bin   b'\x00\x00\x00'
fe   05  03    66:c4:5f:69
└->   5 var      -15258;26975      1767883878 16907370198287517174726656.000  102;196;95;105
--------------------------------------------------------------------------------
```

Status 1 second later:
```shell
14:51:19 INFO: ---------------------- Smartplug / A17X8 / 087f / Header -----------------------
ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
19 00   : 2 Byte total message length (25) in Bytes (Little Endian format)
03 01 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
08 7f   : 2 Byte message type pattern (varies per device model and message type)
00      : 1 Byte optional message increment (  0)
-> b3 <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
-- Fields --|- Value (Hex/Decode Options)---------------------------------------
Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
a1   01  --    34
└->   1 unk              b'4'
a2   09  04    1e:00:00:01:03:01:00:00
└->   9 bin   b'\x1e\x00\x00\x01\x03\x01\x00\x00'
--------------------------------------------------------------------------------
```

Another status from 0405 2 seconds later, see ad field which is the same as a2 field in 087f message
```shell
14:51:20 INFO: ---------------------- Smartplug / A17X8 / 0405 / Header -----------------------
ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
66 00   : 2 Byte total message length (102) in Bytes (Little Endian format)
03 01 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
04 05   : 2 Byte message type pattern (varies per device model and message type)
        : 1 Byte optional message increment (  0)
-> 86 <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
-- Fields --|- Value (Hex/Decode Options)---------------------------------------
Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
a1   01  --    34
└->   1 unk              b'4'
a2   11  00    41:59:54:54:38:4c:30:46:31:33:33:30:34:35:33:36
└->  17 str   b'AYTT8L0F13304536'
a3   02  01    00
└->   2 ui                  0               0
a4   02  01    00
└->   2 ui                  0               0
a5   02  01    00
└->   2 ui                  0               0
a6   05  03    00:00:02:07
└->   5 var            0;1794       117571584           0.000         0;0;2;7
a7   05  03    00:00:02:07
└->   5 var            0;1794       117571584           0.000         0;0;2;7
a8   03  02    0a:09
└->   3 sile             2314            2314                            10;9
a9   03  02    00:00
└->   3 sile                0               0                             0;0
aa   03  02    00:00
└->   3 sile                0               0                             0;0
ab   05  03    9a:6a:00:00
└->   5 var           27290;0           27290           0.000     154;106;0;0
ad   09  04    1e:00:00:01:03:02:00:00
└->   9 bin   b'\x1e\x00\x00\x01\x03\x02\x00\x00'
ae   02  01    01
└->   2 ui                  1               1
fe   05  03    78:d2:5f:69
└->   5 var      -11656;26975      1767887480 16911523021547111037534208.000  120;210;95;105
--------------------------------------------------------------------------------
```
So I assume bytes could mean:
00-01: 30 second toggle_to_delay
02: toggle_to_switch => Off?
03: 01 => ??? current switch status?
04: 03 => delay status? 03 means toggle_to_active?
05-06: 02 seconds passed since start
07: 00 => ??? 

To understand, you have to watch how they change over runtime and what the toggle_to and toggle_back settings are.

thomluther, 2026-01-18:
Is there any update on this for the timer status decoding?
I also need a full system export zip file with MQTT message while you toggle the plug and/or start a delayed toggle in order to test and debug the MQTT extraction. Otherwise I cannot implement any support for it.

JannikKumpf, 2026-01-19:
I can do it when I come back home in around 8 hours.

thomluther, 2026-01-19:
You also need to find out more explanations for the delayed toggle command:
```
13:20:26 INFO: ---------------------- Smartplug / A17X8 / 007e / Header -----------------------
ff 09   : 2 Byte Anker Solix message marker (supposed 'ff 09')
28 00   : 2 Byte total message length (40) in Bytes (Little Endian format)
03 00 0f: 3 Byte fixed message pattern (supposed `03 00/01 0f` for send/receive)
00 7e   : 2 Byte message type pattern (varies per device model and message type)
        : 1 Byte optional message increment (  0)
-> 62 <-: XOR Checksum Byte (last message byte), Crosscheck: 00 => OK
-- Fields --|- Value (Hex/Decode Options)---------------------------------------
Fld Len Typ        uIntLe/var          sIntLe         floatLe      dblLe/4int
a1   01  --    22
└->   1 unk              b'"'
a2   02  01    01
└->   2 ui                  1               1
a3   04  04    05:00:00
└->   4 bin   b'\x05\x00\x00'
a4   02  01    00
└->   2 ui                  0               0
a5   04  04    00:00:00
└->   4 bin   b'\x00\x00\x00'
fe   05  03    98:2f:6e:69
└->   5 var       12184;26990      1768828824 17996818762379738797309952.000  152;47;110;105
--------------------------------------------------------------------------------
```
'a3' shows the 5 seconds in first byte, are the other 2 bytes for minutes and hours?
'a5' has the same pattern, is that used for toggle back delay with seconds, minutes, hours?

thomluther, 2026-01-30:
> This means the Solix plugs can be switched on/off with the version 3.5.0 and only the timers are not working?

@blueskyneufang954 , this is described in the [release notes](https://github.com/thomluther/ha-anker-solix/releases/tag/3.5.0)

closed 2026-05-15 by thomluther