thomluther/ha-anker-solix #152: Support for Shelly 3EM Smart Meter

koehntopp, 2024-10-31:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I'd like to be able to switch between charging and discharging modes, with the Shely 3EM Smart Meter. API logs attached

### Describe the solution you'd like

Allow usage_mode to be available with a Shelly as smart meter

### Describe alternatives you've considered

none

### Additional context

[fraxx_2024-10-30_1756.zip](https://github.com/user-attachments/files/17577547/fraxx_2024-10-30_1756.zip)

thomluther, 2024-10-31:
This is what I get for the device itself in my latest version (The shelly picture will be available with the next integration release)
[image]

koehntopp, 2024-11-01:
Works like a charm, thank you!

closed 2024-11-03 by thomluther