thomluther/ha-anker-solix #154: too much ping Api to Anker

Thunder1805, 2024-11-03:
### System Health details

get too much ping 2.2.0 version connection cutboff
[image]

### Integration version

2.2.0 update have such trouble 

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

via app of Anker no access anymore must wait 24 hours 

### Reproduction steps

1.instsll 2.1 not working today 
2.
3.
...

### Debug logs

```text
soory no logs
```

### Diagnostics dump

sorry

kafisc1, 2024-11-03:
Same issue here. Maybe an issue with the Anker cloud.

Thunder1805, 2024-11-03:
hmmm installing version 2.2 and you can see data pull in the panel graphics high, second later zero, then high again. I have 1 Master and 3 guest account. only 1 is showing data, do cloud shouldn't be the issue

kafisc1, 2024-11-03:
I've just reinstalled the official android App and the issue occurs here also. When starting the Android App I'm getting an error message when typing in the mail adress wihtout even logging in.
[image]

thomluther, 2024-11-04:
The new integration did nothing change on the connection method.
Yesterday during the day there have been intermittant cloud issues. You have to try again once the cloud is more stable again and your 24h ban is over (or reconnect your router to the internet again to hopefully get a new IP).
And never use the same Anker account on different devices, otherwise you can quickly get to the ban because of too many token changes for your IP address.

Thunder1805, 2024-11-04:
Now I found out the problem on user side, every time I use the app, either as Master or User, there is a token. When my Handy tells me "Anker is running in the background" and I close "all applications, and then after 1-2 hours log in again Token is used. In HA I uses the Master account log in, therefore each Time on Handy App also a token is made. Then number of token was to much. Now is running fine again, HA with a USER and Handy with Master. Sorry for inconvenience.

Thunder1805, 2024-11-04:
CLOSE MY COMMENTS

thomluther, 2024-11-04:
Yeah, this is the issue. Each device using the same account will trigger a new token on the server for that account. After 10 new tokens for the IP per day, there will be the ban for the IP.
You can use the Master account in HA and the Shared account in the App.
A temporary switch is possible when you disable the Api communication in HA with the Api switch entity, then you can login to the App with the Master account and do your stuff. This will trigger a new token for the Master.
Once finished, you can enable the Api switch in HA again and the integration will get a new token again. This will again trigger a new token for the Master account. Your App will be logged off automatically, since the App token is no longer valid.
You can do that 5 times a day before you hit the IP ban...

closed 2024-11-04 by thomluther