thomluther/ha-anker-solix #159: Support for 250W Prime Charger (A2345)

Spacelord09, 2024-11-06:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The new charger is recognised and added to HA. But there are no usable entities.
Is it possible to add some for each input?

### Describe the solution you'd like

Entities:

- For all Inputs(C1-C4 / A1+A2):
  - watt
  - voltage
  - amperage
  - connected (true/false)

- General:
  - Overall usage in % (of 250w)
  - Overall usage in watt
  - 

### Describe alternatives you've considered

NA

### Additional context

Currently this is everything HA gets.
[image]

Spacelord09, 2024-11-06:
@thomluther Why did you assign the invalid tag?

thomluther, 2024-11-06:
Hm, obviously my update did not get published. 
Basically what you ask for is not possible since only devices that can be configured into a power system will provide usage data to the cloud server. All stand alone devices like charger or portable power supplies are therefore not supportable via the cloud api and what you see is only generic information that is available in the cloud by any device.

Someone would have to develop a complete new interface to local BT or the MQTT cloud server, otherwise those devices cannot be integrated. 
I'm not going into that direction, so won't fix is that tag that matches your request.

Spacelord09, 2024-11-06:
I already suspected something like that, thx for the information

closed 2024-11-06 by Spacelord09

Spacelord09, 2024-11-07:
@thomluther Do you have more Information about ankers mqtt server? Like dns names / ports / how to auth? and more?

thomluther, 2024-11-07:
Hi @Spacelord09 , No I haven't. You may have a look to the [Api library](https://github.com/thomluther/anker-solix-api).
What you can do is first start with a [system export ](https://github.com/thomluther/anker-solix-api#export_systempy) to see what you get from the cloud server for the useful endpoints known so far.

You can also explore the [apitypes.py ](https://github.com/thomluther/anker-solix-api/blob/main/api/apitypes.py) for the known endpoints so far and test them with any parameters which you will have to figure out as well. (there is no documentation)
You may find some endpoints that may allow to query device settings and eventually modify them via the Api.

But at the end, I do not think that standalone devices will send their power data if they do not have power statistics in the app.
Even with energy statistics, there is no guarantee that power values will be send also to the cloud server. See [Add F3800/BP3800 Equipment when connected to Home Power Panel - Config Example](https://github.com/thomluther/anker-solix-api/issues/117)

The MQTT server is where all devices send their real time data. However, that obviously happens only when the app is accessing them, so this needs a trigger and I have no idea whether the duration is limited. There was already some investigation into it in this [api issue](https://github.com/thomluther/anker-solix-api/discussions/71), but I think that will be also difficult (and would also cause heavy MQTT cloud usage if all your Solix devices would report 1 sec data 24x7)