thomluther/ha-anker-solix #160: Solar Production shows 0 Watts in the morning / App says differently

mike4001, 2024-11-10:
### System Health details

## System Information

version | core-2024.11.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.31-haos-raspi
arch | aarch64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1463
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 13.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.11.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 439.4 GB
disk_used | 23.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.2), Advanced SSH & Web Terminal (19.0.0), File editor (5.8.0), Samba Backup (5.2.0), Mosquitto broker (6.4.1), AppDaemon (0.16.7), Signal Messenger (0.89.0), MQTT Explorer (browser-1.0.1), ESPHome (2024.10.3), pigpio (1.5.2), Firefox (1.3.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11. Oktober 2024 um 18:40
-- | --
current_recorder_run | 9. November 2024 um 07:12
estimated_db_size | 4309.98 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Integration version

2.2.0

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have the problem that in the morning the Integration sometimes shows 0 Watt solar production when it should not be the case. Sometimes this is only for 1-2 update intervals and then shows it again.

[image]

This time it was over 25 minutes so I deactivated API connection, logged into the Anker app and saw that solar production was normal. Then I reactivated the API connection in Home Assistant and it immediately showed Solar production.

I suppose if I waited longer it would come back sooner or later that I do not know if this is normal behaviour or a bug?

Are other having the same behaviour or is there something wrong?

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Dont Know
```

thomluther, 2024-11-10:
Hi @mike4001 
I have not SB2, so I have no experience how a real device behaves with the integration.
Without getting more debug data from the integration, I cannot tell either what data has been received from the cloud.

When you have this situation again, you can run the service to query the system info, this will show you whether actual consumption values for your system are returned from the cloud.

If the SB lost cloud connection, those values may become 0 either...

mike4001, 2024-11-11:
[mdall_2024-11-11_0859.zip](https://github.com/user-attachments/files/17707152/mdall_2024-11-11_0859.zip)

Enclosed you find the data from the service query during a phase where the integration reports 0 Watt where I should not be the case.

But I believe you of course that the integration only reports what it receives from the cloud.

Maybe this really is some kind of connection issue though I do not understand why it only appears in the morning, other than really being the temperature.

Though Wifi and Cloud sensors always report "connected"

thomluther, 2024-11-12:
Your scene info does show the data from the cloud which are used in the integration:
```
    "solarbank_info": {
      "solarbank_list": [
        {
          "device_pn": "A17C1",
          "device_name": "Solarbank 2 E1600 Pro",
          "device_img": "https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png",
          "battery_power": "10",
          "bind_site_status": "",
          "charging_power": "0",
          "power_unit": "W",
          "charging_status": "6",
          "status": "1",
          "wireless_type": "1",
          "main_version": "",
          "photovoltaic_power": "0",
          "output_power": "0",
          "create_time": 1730400859,
          "set_load_power": "",
          "sub_package_num": 2,
          "output_cutoff_data": 10,
          "is_display": true
        }
      ],
      "total_charging_power": "0",
      "power_unit": "W",
      "charging_status": "0",
      "total_battery_power": "0.10",
      "updated_time": "2024-11-11 08:58:42",
      "total_photovoltaic_power": "0",
      "total_output_power": "0.00",
      "display_set_power": false,
      "battery_discharge_power": "0",
      "ac_power": "0",
      "to_home_load": "0",
      "is_display_data": true,
      "solar_power_1": "0",
      "solar_power_2": "0",
      "solar_power_3": "0",
      "solar_power_4": "0"
    },
```

If they are 0 for PV, but you actually have PV, then something is wrong with the communication to the cloud.
Since charging status 6 means the SB is using full bypass = deep standby (No BMS invovled, all is bypassed to inverter, at least this way it was for SB1), it could mean that the SB2 is in deep standby and does not send any data to the cloud to save batter SOC, since PV is too low to heat / charge and ensure full BMS usage...

There is nothing that can be fixed in the Api for this situation, that is how the SB2 behaves in Winter.
The App might be different, since when connected it may trigger to SB2 to wake up from deep_standby to give you the real values.
But when you use the shared account in the Anker app, you should see the same values/behavior as you can see in the integration.

thomluther, 2024-11-12:
Therefore you cannot compare the Integration values with the Anker App, even if using the main account in the Integration.
But the App behavior with a shared account is the same as the integration.
Open App with main account may trigger the SB2 via the mqtt cloud server to push information into the cloud. This is also how frequent value updates are triggered to the cloud.
Therefore you see < 5 min updates in the HA integration for periods where you open the Anker App with the main account.

mike4001, 2024-11-12:
OK I see

Basically the SB2 is working as intented but sometimes does not update the correct values to the cloud. And Home Assistant can only show what it receives from the cloud. The App (Main Account) triggers an extraordinary update to the cloud.

But another question: You write the Full Bypass is Standby Mode.

According to Home Assistant my SB2 is always in Full Bypass ("Voller Bypass") during the day and in standby ("Ruhezustand") during the night. A few short periods of detection ("Erkennung").

It is in Full Bypass no matter if the battery is charing or discharging.

[image]

So I guess this is another thing which should be different?

thomluther, 2024-11-12:
I don't know. For SB1 it status 6 was very rare, and only at cold temperature when it really bypassed the BMS completely.
For SB2 the code 6 could mean something different. But you and other would need to monitor it more closely to identify what code 6 actually means for SB2 under various conditions....

thomluther, 2024-11-12:
Keep in mind, that you have a shelly. For this constellation code 6 could also mean something different than for SB2 with or without Anker Smart meter.
So the real meaning of the codes can only be determined by the community while watching which code is used in which situation.
The name is just something I invented for the codes that have been seen so far

mike4001, 2024-11-12:
Sorry, but I have to hijack this thread again for maybe another problem.

Since about 5 a.m. today the 5%/10% remaining battery charge goes unavailable.

It updates during the 5 minute interval with all the other values, is then availaby for exactly 1 minute and then goes unavailable for 4 minutes until the next cycle.

[image]

Do you need anmore data for this?

thomluther, 2024-11-12:
If you have no integration related errors in your core logs, its typically not a problem with the integration.
Could be that your cloud response indicates invalid data. You can check this in the system information service.
      "is_display_data": true,
If this field is false, the integration considers all values from the Solarbank devices as false, so basically most of your entities will become unavailable.
In the options you can configure to skip such invalid marked responses and keep the last valid value for the entities.

thomluther, 2024-11-13:
The option will be a decider what to do with the values when the field  "is_display_data": false.
Otherwise the data in the response is considered valid and it must be.
The client cannot decide whether 0 values in a response are valid or not.

mike4001, 2024-11-13:
OK, I undestand.

Thanks anyway for your help.

Maybe let this issue open for a little bit if others have similar problems. If nothing comes up we can close this as it is - as you said - cloud-related.

mike4001, 2024-12-06:
> OK I see
>
> Basically the SB2 is working as intented but sometimes does not update the correct values to the cloud. And Home Assistant can only show what it receives from the cloud. The App (Main [...]

Just wanted to add, that after a complete reboot of my Solarbank 2 it is now updatating the status codes correctly.

So this was probably some kind of a SB2 bug.