thomluther/ha-anker-solix #181: Combination of Solarbank S1 (twice) and S2 Pro - does not provide correct Data

RamsukKockoi, 2025-01-06:
### System Health details

Does not seem to be a System Problem

### Integration version

2.3.0

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The Solarpower Sensor does not show or calculate the charging Power of my both S1 to the total Solar power Sensor. These datas are missing. 
Also the Charging power is not shown correctly, it shows just the charging Power of one panel, also at high charging Power, the from S1 (at standard) delivered 50W (twice => means 100) are missing in the charging Power of the S2.

Compared with the Anker App, the charging Power of the S2 (sensor.solarbank_2_e1600_pro_akkuleistung) is halfened. And

### Reproduction steps

It is always like above described ...

### Debug logs

```text
sorry
```

### Diagnostics dump

.

RamsukKockoi, 2025-01-06:
View above

thomluther, 2025-01-06:
> All batteries are in the same System. I can not manage the S1 independent from the S2.

So it means, you cannot set any S1 schedule anymore in the App? Does that mean, they have a fixed setting of 50W each for the output? If so, would a single S1 in the system default to 100 W output fixed?

> If now the S1 (both) are loading and providing the 100Watts, Anker forgetts to calculate the Charging to the Solar Power.

It depends how the S1 power is considered in relation to what is shown in the coupled system homepage and the charts.
If all is still reported from S2 point of view, only the S1 output is considered as 'PV' power and the reporting is correct. That means anything what will be charged or produced on S1 is no longer visible for a coupled system. In other words, you loose visibility/granularity of the S1 power and energy data when added to a S2 system...
It also means, when the S1 will discharge in the evening (also with 100W fix?), this will still be shown as PV power for the S2, 
So now you will get sunshine at night in the App and the statistics :-o

Lets hope Anker will improve this, but this would be a complex change since additional fields must be added to the cloud services as well as to the App I guess. So far, all energy stats (App charts) are based on a system only, and this does not distinguish energy types on a device level. Also the S2 PV data (power and energy) do not distinguish whether it comes from a solar panel or from a cascaded S1...

RamsukKockoi, 2025-01-06:
Hi, the problem with S1 energy not showed on solarpower is cleared. If S2 ist going empty, S1 starts charging, if it does, this is counted as solarpower. Not nice, but OK.

At the moment the API seems to deliver all S1 values multiplied with 10... Even not nice. I will keep commenting here. But it does not seem bot be an intergration problem.

thomluther, 2025-01-08:
> One Question: What ist intervallmultiplication in the Configuration of the Integration. Have a look at my screenshot.

This is all described [here](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#option-considerations-for-solarbank-2-systems)

RamsukKockoi, 2025-01-09:
Sorry, did not see it.

thomluther, 2025-01-17:
Hi @RamsukKockoi 
According to the [Anker blog](https://www.ankersolix.com/de/blogs/balkonkraftwerk-mit-speicher/wie-du-den-besten-solarbank-modus-auswahlen-kannst-um-dein-zuhause-mit-strom-zu-versorgen), the SB1 schedule/output can still be adjusted while it is configured in the SB2 system, but it depends on the actual usage mode setting. 
Can you confirm, that SB1 schedule is still accessible and usable in your mobile App, e.g. when you switch to manual mode?
Can you please provide a [system export](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action) from the owner account to correlate the Api fields of such a system constellation? If you have a schedule defined for both, SB1 and SB2 in custom mode, the export should show both schedules and the HA integration should also still allow to modify both hopefully.

RamsukKockoi, 2025-01-17:
Hi,

yes I can confirm that!

In userdefined mode the settings of the S2 take the lead and sets the same setting to my both S1. Means if I set 100 Watts for output, all devices go to 100 Watt. Down to 13% the S2 seems to deliver the energy from itself. From then on it starts to take energy out of the both S1 (50Watts each). The values may not be exact, until now I did not have enough sun to test.

If the setting for S2 ist "AI", I am free to set the output for the S1 (one setting for both). What happens, is that if I set 500 Watts, both S1 support the S2 with 250 Watts each, if the S2 means that it needs support. Before that they do nothing oder charge themselves.

Without a Zero Watts switch, the S1 always deliver their first 50 Watts to the S2, to charge. Even if I have set 0. That disappears first if you use a 0-Watts switch.

One weird thing ist, that since I connected the both S1 to the S2 and made a new system, the sensor "battery Power" in german "Akkuleistung" ist alwaya just half the Value of the app.
Example: If you have 60 Watt Solarpower and it is all charging to the S2, the Batterpower (Akkuleistung) ist just 30 (or a little below). Same, if I set the S1 to zero (with switch) 50 go in and it shows 25.

Hope I could help a little.

thomluther, 2025-01-17:
According to the screenshots, it seems both still have separate schedules, but they are not longer used independently.
SB1 schedule seems to be used only when SB2 is in AI mode. When it is in manual mode, the SB2 schedule seems to be used by the SB1 as well according to description, but not sure how this is done. Maybe the SB1 then utilizes the SB2 schedule object, which is a different one than the SB1 schedule.
Therefore it will be helpful only if you also provide an export of the Api responses using the owner account in order to be able and correlate such information with the API responses and field values.
Also the problem with wrong battery power....I cannot help fixing this when nobody can correlate the Api data to the Api response fields....

RamsukKockoi, 2025-01-17:
According to the screenshots, it seems both still have separate schedules, but they are not longer used independently.
**Correct, for my Setup.**

SB1 schedule seems to be used only when SB2 is in AI mode.
Also correct for me. 

When it is in manual mode, the SB2 schedule seems to be used by the SB1 as well according to description, but not sure how this is done. Maybe the SB1 then utilizes the SB2 schedule object, which is a different one than the SB1 schedule.

Therefore it will be helpful only if you also provide an export of the Api responses using the owner account in order to be able and correlate such information with the API responses and field values.
**I am willing to do so, but I need a little help here, with a "how to"**

Also the problem with wrong battery power....I cannot help fixing this when nobody can correlate the App data to the Api response fields....
**Same, need a small "how to"**

RamsukKockoi, 2025-01-17:
I am so fu***** blind ....

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
      site_name: FluxKompensator
      site_img: ""
      device_type_list:
        - 3
        - 6
      ms_type: 1
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C0
        - A17C1
        - SHEMP3
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "18.50"
        unit: kWh
      - type: "2"
        total: "18.44"
        unit: kg
      - type: "3"
        total: "5.92"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "21"
          bind_site_status: ""
          charging_power: "131"
          power_unit: W
          charging_status: "3"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "131"
          output_power: "0"
[... 135 lines cut ...]
        solar_production_pv4: "0.00"
        solar_production: "0.25"
        battery_charge: "0.25"
        solar_to_grid: "0.00"
        battery_percentage: "1.00"
        solar_percentage: "0.00"
        other_percentage: "0"
      last_period:
        date: "2025-01-16"
        battery_discharge: "0.83"
        ac_socket: "0.00"
        battery_to_home: "0.82"
        home_usage: "14.22"
        grid_to_home: "13.32"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.03"
        solar_production_pv2: "0.07"
        solar_production_pv3: "0.08"
        solar_production_pv4: "0.04"
        solar_production: "0.22"
        battery_charge: "0.14"
        solar_to_grid: "0.00"
        battery_percentage: "0.64"
        solar_percentage: "0.36"
        other_percentage: "0"

RamsukKockoi, 2025-01-18:
Hey, did it help a little?

thomluther, 2025-01-18:
> Hey, did it help a little?

I'm puzzeled with the number of different randomized site IDs and device serials in your export zip. Much more than you have in your system...
Did you hit the Run action button multiple times?
I wonder whether the export action was kicked off more than once in a short time and was processed in parallel...
I may have to implement a delay to prevent it can be kicked off more than once within a minute or so...

So actually, your export is somehow questionable since I don't know which files belong together to the same export run.

Can you please run the export again, but pressing the Run Action button just ONCE. You get no feedback until the action is completed and turns the button green.

thomluther, 2025-01-18:
Some observations I made already:
* Anker does not count the SB1 data together with the SB2 data for the Solarbank totals in a System. Only the SB2 data is reflected in the Solarbank totals, like System SOC, Charge power, etc. Some accumulated values would not make sense either when SBs are cascaded. However, a total SOC might be useful (average of all SBs in the system)
* I'm little bit reluctant to wrap values into other totals via the Api library. This would show different values in the integration compared to what the app might show. But some Api values for the SB device and/or SB total are inconsistent anyway since the beginning. And wrapping via the library became necessary to reflect correct values per system/device.
* To correlate SB device values and system values properly in various situations, I would need shared access to such a system to be able to monitor the Api response values in various situations and make sure they are properly mapped to the library structures. Otherwise HA may display them incorrectly, like the wrong batter power
* Seems the SB1 schedule object looks slightly different when enfored by the SB2 manual mode. Eg. the Export switch is null. These unexpected fields values have to be mapped properly to the old SB1 schedule structures, otherwise they may be reflected incorrently.
* The SB1 schedule entities will remain changable even if the SB2 schedule settings are enforced. The question is what happens when the SB1 output preset entity is changed while the SB2 custom mode is active with a preset, that is synched and enforced into the SB1 schedule. So what happens when a change is applied to the SB1 schedule in that situation? Will it be ignored, result in Api error or change the SB2 output indirectly in the SB2 schedule as well?

RamsukKockoi, 2025-01-18:
Tomrrow, we should have some sun here. What can I do to help. 
Would it be helpful, if you had some Viewing Access to my System?

thomluther, 2025-01-19:
> Would it be helpful, if you had some Viewing Access to my System?

That would help so I can monitor the system values over longer periods for correlation with the App. See [here for my account](https://github.com/thomluther/ha-anker-solix/issues/175#issuecomment-2599763156) you can use.
Unfortunately it will not allow me to query the schedules. So please export your main account once more, just one click to run the export Action.

RamsukKockoi, 2025-01-19:
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
43
44
45
46
47
48
49
⌄
⌄
⌄
⌄
⌄
⌄
⌄
⌄
⌄
⌄
⌄
⌄
sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
      site_name: FluxKompensator
      site_img: ""
      device_type_list:
        - 6
        - 3
      ms_type: 1
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C1
        - SHEMP3
        - A17C0
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "19.63"
        unit: kWh
      - type: "2"
        total: "19.57"
        unit: kg
      - type: "3"

[Markus Kociok_2025-01-19_1216.zip](https://github.com/user-attachments/files/18468617/Markus.Kociok_2025-01-19_1216.zip)

RamsukKockoi, 2025-01-19:
Invitation is out.

If it would help, and you can promise to not make a toaster of it, I am willing to give you administrative credentials.
If we can handle to be sure, it is you.

thomluther, 2025-01-19:
> Invitation is out.
> 
> If it would help, and you can promise to not make a toaster of it, I am willing to give you administrative credentials. If we can handle to be sure, it is you.
> 
> Otherwise, I can export as much as you want, as soon as I read it.

Thanks, for now the shared access is sufficient. That will allow me to correlate the values. The schedule is something that I can work later on...

From what I see, Anker made a very lazy implementation for those coupled systems. The whole System totals for Solarbanks are ONLY reflecting SB2 data, which is pretty useless if you want to track various energies over time. Having Solar Power/Energy reported at night doesn't make sense and also the SB1 charge and discharge energy is not reflected anywhere to make any monitoring of the overall system efficiency. So you can basically forget now the energy stats collected by the Anker cloud since they are wrong for your system, at least PV, Charge and Discharge energies...

Maybe I'll recalculate those SB totals for the system so they make more sense from a system perspective. But that would require to have a reliable indicator, which SB2 PV entry is used by a SB1 and I have not found yet a valid indicator by the existing values that works at any point in time...
From what I see over time, looks like your SB1 are connected to PV 1 and 4? 
Does Anker allow to connect 2 SB1 into single SB2 Input? I think at least we can assume, that only one SB1 is connected to a single SB2 Input, but maybe a single SB1 can be using 2 SB2 inputs?

thomluther, 2025-01-19:
Another weird SB totals constellation which puzzles me from your export:
```
    "solar_power_1": "0",
    "solar_power_2": "91",
    "solar_power_3": "94",
    "solar_power_4": "0",
    "micro_inverter_power": "94",    => Why is this always same value as solar_power_3? And all other micro_inverter values are 0? Does not make any sense...
    "micro_inverter_power_limit": "0",
    "micro_inverter_low_power_limit": "0",
    "pei_heating_power": "0",
```
I goes this is another inconsistency how the cloud responds to Api queries...

RamsukKockoi, 2025-01-19:
Hey, it is not allowed to connect 2 SB1 into one port of SB2.

In Ankers scheme, they only show one per port ...

https://www.anker.com/eu-de/blogs/balkonkraftwerk-mit-speicher/solarbank-der-ersten-generation-und-solarbank-2-pro-zusammen-fur-mehr-energiekontrolle

Please do forget the one Microinverter, that looks in direction "West" and is only active if "Flux" ist under 500. To stay in German laws.

RamsukKockoi, 2025-01-19:
"solar_power_1": "0",
    "solar_power_2": "91",
    "solar_power_3": "94",
    "solar_power_4": "0",
    **"micro_inverter_power": "94",    => Why is this always same value as solar_power_3? And all other micro_inverter values are 0?** Does not make any sense...
    "micro_inverter_power_limit": "0",
    "micro_inverter_low_power_limit": "0",
    "pei_heating_power": "0",

I  am wondering what it means, as you. That is one of the miracles ....
Also the charging Power of 1 and 4, my SB1 alway show about half of what the App shows ....

RamsukKockoi, 2025-01-20:
Hey, I think i will not make too much sense, to start calculating the values in the integration. I am sure Anker will correct it more or less soon.

Here and there, I have my own calculations and the battery state and charging rates, for me come from the SOC ...

RamsukKockoi, 2025-01-21:
Hey, Sun ist comming Up. As now you can see the different values. The API shows more ore less half the Values of the App.

[Markus Kociok_2025-01-21_0951.zip](https://github.com/user-attachments/files/18487644/Markus.Kociok_2025-01-21_0951.zip)

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
      site_name: FluxKompensator
      site_img: ""
      device_type_list:
        - 3
        - 6
      ms_type: 1
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C1
        - SHEMP3
        - A17C0
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "20.58"
        unit: kWh
      - type: "2"
        total: "20.52"
        unit: kg
      - type: "3"
        total: "6.59"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "99"
          bind_site_status: ""
          charging_power: "53"
          power_unit: W
          charging_status: "0"
          status: "1"
          wireless_type: "1"
          main_version: ""
[... 137 lines cut ...]
        solar_production_pv4: "0.00"
        solar_production: "0.04"
        battery_charge: "0.04"
        solar_to_grid: "0.00"
        battery_percentage: "1.00"
        solar_percentage: "0.00"
        other_percentage: "0"
      last_period:
        date: "2025-01-20"
        battery_discharge: "0.00"
        ac_socket: "0.00"
        battery_to_home: "0.00"
        home_usage: "13.42"
        grid_to_home: "13.32"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.00"
        solar_production_pv2: "0.13"
        solar_production_pv3: "0.11"
        solar_production_pv4: "0.00"
        solar_production: "0.23"
        battery_charge: "0.14"
        solar_to_grid: "0.00"
        battery_percentage: "0.61"
        solar_percentage: "0.39"
        other_percentage: "0"

RamsukKockoi, 2025-01-21:
Here ist another weird behaviour!

SB2 is full now and bypasses solar energy now to the grid. Since it is doing that, I do not see Batterypower for the SB1.
See the screenshot. SB1 are charging, it is visible in the App.

[image]

thomluther, 2025-01-21:
Yes, the charging power always had to be calculated by the library since the inconsistencies in the Api responses and what is reported as SB totals and individual device power....
This calculation is now wrong for combined systems since they are cascaded, it was only implemented for parallel usage of multiple solarbanks in the system.

RamsukKockoi, 2025-01-21:
So I think we have to wait for Anker. Until then, the calculations are more or less good.
I could see discharging energy fully correctly. I will have a look at tomorrow, do you want an export, while discharging?

By the way, if you set 450 W for discharging the SB1, the SB2, if it is fully loaded wil not regulate the ammount of energy down. I will just pass the energy through, disregarding, what mit Shelly 3 pro EM says ....

thomluther, 2025-01-21:
> SB2 is full now and bypasses solar energy now to the grid. Since it is doing that, I do not see Batterypower for the SB1.
> See the screenshot. SB1 are charging, it is visible in the App.

So looking at the screenshot, only the Akkuleistung should be wrong. The other marked yellow values should all be correct since they are not calculated.

> do you want an export, while discharging?

No, I can monitor that with my shared account over time.
However, I would need owner account exports when:
- SB2 manual mode is active (and SB1 schedule is enforced by SB2 schedule). Ideally that should be be with PV when PV cannot fulfill the set output power. Will be good to see whether that is satisfied by SB2 discharge only, since SB1 should not discharge while PV is there (it cannot do both)
- SB2 AI mode is active (and SB1 schedule is used).

Can you confirm, that original saved SB1 schedule is maintained, when toggling between AI and manual mode? Or is it lost, because SB2 manual mode overwrites SB1 schedule?

I suspect another inefficiency that you may check and eventually escalate to Anker.
When SB2 battery is in the high 90 SOC, charging power must be reduced for battery protection. With 0W output setting, typically the PV must be throttled. 
However, in combined systems that is not necessary for the MPPT that are connected to the SB1, since the SB1 could still charge all PV if not limited. However, I know that MPPT limitation of SB1 output will negatively affect the generated SB1 PV.
So the SB2 PV throttling probably negatively impacts SB1 PV production and charge...

RamsukKockoi, 2025-01-22:
Hi, early in the morning, here is the first one.

System is in AI Mode. SB2 is discharging, SB1 is in scheduled mode.
The schedule is maintained when togling between the modes. I keeps the values, that were set before.

I will give charging SB2 with activated Schedule for SB1 (100Watt, 50 each) a try and see, if SB2 has a negative effect to SB1.

When SB2 goes to 90% and over, the both SB1 continue charging with the Power, that the PV modules deliver, with 0-Watt switch from Anker. It does not seem, that SB2 has an impact on charging the SB1. I can not make any statements to when it is cold, as my batteries are stored in the basement of my house, they always have 13-18 degrees.

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
      site_name: FluxKompensator
      site_img: ""
      device_type_list:
        - 3
        - 6
      ms_type: 1
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C1
        - SHEMP3
        - A17C0
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "23.71"
        unit: kWh
      - type: "2"
        total: "23.63"
        unit: kg
      - type: "3"
        total: "7.59"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "95"
          bind_site_status: ""
          charging_power: "-227"
[... 144 lines cut ...]
        battery_charge: "0.00"
        solar_to_grid: "0.00"
        battery_percentage: "0.00"
        solar_percentage: "1.00"
        other_percentage: "0"
      last_period:
        date: "2025-01-21"
        battery_discharge: "0.03"
        ac_socket: "0.00"
        battery_to_home: "0.03"
        home_usage: "11.23"
        grid_to_home: "9.04"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.36"
        solar_production_pv2: "1.16"
        solar_production_pv3: "1.20"
        solar_production_pv4: "0.42"
        solar_production: "3.13"
        battery_charge: "0.35"
        solar_to_grid: "0.64"
        battery_percentage: "0.11"
        solar_percentage: "0.89"
        other_percentage: "0"

[Markus Kociok_2025-01-22_0658.zip](https://github.com/user-attachments/files/18500522/Markus.Kociok_2025-01-22_0658.zip)

RamsukKockoi, 2025-01-22:
I have set 250 Watt in manual mode for SB2. PV ist not sufficient to deliver all 250 watts.

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "23.74"
        unit: kWh
      - type: "2"
        total: "23.67"
        unit: kg
      - type: "3"
        total: "7.60"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_name: Solarbank 2 E1600 Pro
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "93"
          bind_site_status: ""
          charging_power: "249"
          power_unit: W
          charging_status: "0"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "61"
          output_power: "249"
          create_time: 1735924816
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 0
          is_display: true
          bat_charge_power: "0"
        - device_pn: A17C0
          device_sn: AZV6Y60D31700145
          device_name: Solarbank Wohnzimmer
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png
          battery_power: "90"
          bind_site_status: ""
          charging_power: "0"
          power_unit: W
          charging_status: "3"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "46"
          output_power: "0"
          create_time: 1735925245
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 10
          is_display: true
          bat_charge_power: "44"
        - device_pn: A17C0
          device_sn: AZV6Y60D34200552
          device_name: Solarbank Wintergarten
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png
          battery_power: "94"
          bind_site_status: ""
          charging_power: "0"
          power_unit: W
          charging_status: "3"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "42"
          output_power: "0"
          create_time: 1735925314
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 10
          is_display: true
          bat_charge_power: "42"
      total_charging_power: "0"
      power_unit: W
      charging_status: "0"
      total_battery_power: "0.93"
      updated_time: "2025-01-22 09:17:29"
      total_photovoltaic_power: "61"
      total_output_power: "249.00"
      display_set_power: false
      battery_discharge_power: "188"
      ac_power: "0"
      to_home_load: "250"
      is_display_data: true
      solar_power_1: "0"
      solar_power_2: "29"
      solar_power_3: "32"
      solar_power_4: "0"
      other_input_power: "0"
      micro_inverter_power: "32"
      micro_inverter_power_limit: "0"
      micro_inverter_low_power_limit: "0"
      grid_to_battery_power: "0"
      pei_heating_power: "0"
      backup_info:
        start_time: 0
        end_time: 0
        full_time: 0
    retain_load: 250W
    scene_mode: 3
    home_load_power: "226"
    updated_time: 01-01-0001 00:00:00
    power_site_type: 5
    site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
    powerpanel_list: []
    grid_info:
      grid_list:
        - device_pn: SHEMP3
          device_sn: "9869313345840"
          device_name: Hauptstrom Keller
          device_img: ""
          create_time: 1735925025
          status: "1"
          bind_site_status: ""
          wireless_type: ""
      photovoltaic_to_grid_power: "23"
      grid_to_home_power: "0"
      grid_status: 0
      has_history_data: true
    is_downgrade: false
    error_code: 0
    smart_plug_info: null
    feature_switch:
      0w_feed: true
      heating: false
      multi_pv: true
      shelly_meter: true
      smart_plug: true
    other_loads_power: "226"
    priority_discharge_switch: 0
    display_priority_discharge_tips: 0
    priority_discharge_upgrade_devices: ""
    style_id: 0
    is_show_priority_discharge: 0

RamsukKockoi, 2025-01-22:
Negative Charge of SB2 can not be correct here.

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "23.80"
        unit: kWh
      - type: "2"
        total: "23.73"
        unit: kg
      - type: "3"
        total: "7.62"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_name: Solarbank 2 E1600 Pro
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "70"
          bind_site_status: ""
          charging_power: "799"
          power_unit: W
          charging_status: "1"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "120"
          output_power: "799"
          create_time: 1735924816
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 0
          is_display: true
          bat_charge_power: "0"
        - device_pn: A17C0
          device_sn: AZV6Y60D31700145
          device_name: Solarbank Wohnzimmer
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png
          battery_power: "91"
          bind_site_status: ""
          charging_power: "0"
          power_unit: W
          charging_status: "3"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "108"
          output_power: "0"
          create_time: 1735925245
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 10
          is_display: true
          bat_charge_power: "108"
        - device_pn: A17C0
          device_sn: AZV6Y60D34200552
          device_name: Solarbank Wintergarten
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/10/iot-admin/FycLLqjHpYf0Bdab/20230719-144818.png
          battery_power: "94"
          bind_site_status: ""
          charging_power: "0"
          power_unit: W
          charging_status: "3"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "112"
          output_power: "0"
          create_time: 1735925314
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 10
          is_display: true
          bat_charge_power: "112"
      total_charging_power: "0"
      power_unit: W
      charging_status: "0"
      total_battery_power: "0.70"
      updated_time: "2025-01-22 09:56:50"
      total_photovoltaic_power: "120"
      total_output_power: "799.00"
      display_set_power: false
      battery_discharge_power: "679"
      ac_power: "0"
      to_home_load: "800"
      is_display_data: true
      solar_power_1: "0"
      solar_power_2: "62"
      solar_power_3: "58"
      solar_power_4: "0"
      other_input_power: "0"
      micro_inverter_power: "58"
      micro_inverter_power_limit: "0"
      micro_inverter_low_power_limit: "0"
      grid_to_battery_power: "0"
      pei_heating_power: "0"
      backup_info:
        start_time: 0
        end_time: 0
        full_time: 0
    retain_load: 250W
    scene_mode: 1
    home_load_power: "1876"
    updated_time: 01-01-0001 00:00:00
    power_site_type: 5
    site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
    powerpanel_list: []
    grid_info:
      grid_list:
        - device_pn: SHEMP3
          device_sn: "9869313345840"
          device_name: Hauptstrom Keller
          device_img: ""
          create_time: 1735925025
          status: "1"
          bind_site_status: ""
          wireless_type: ""
      photovoltaic_to_grid_power: "0"
      grid_to_home_power: "1077"
      grid_status: 0
      has_history_data: true
    is_downgrade: false
    error_code: 0
    smart_plug_info: null
    feature_switch:
      0w_feed: true
      heating: false
      multi_pv: true
      shelly_meter: true
      smart_plug: true
    other_loads_power: "1876"
    priority_discharge_switch: 0
    display_priority_discharge_tips: 0
    priority_discharge_upgrade_devices: ""
    style_id: 0
    is_show_priority_discharge: 0

thomluther, 2025-01-22:
> I will give charging SB2 with activated Schedule for SB1 (100Watt, 50 each) a try and see, if SB2 has a negative effect to SB1.

Can you go below 100 W min for SB1 output setting in the schedule? Or are the original SB1 min limits remain for the schedule settings, even if connected to known SB2 MPPTs and inverter?

I think for the value calculations I have sufficient examples for now and implemented work around for proper total power calculations. The device battery power should also be fixed.

I may need more exports or admin access when I start working on support for the SB1 schedule changes in Combi systems, which is probably broken as well in the integration.
Did you test already schedule modifications with the integration Actions on SB2 or SB1? I'm not sure if the advanced power mode for SB1 schedules can still be applied, which allows to set different outputs per SB1 (not always 50% of the output).

RamsukKockoi, 2025-01-23:
Hi, I did not test schedule Actions in the Integration. But I will test. At the Moment we are going to not have Sun here.
So testing will be a little more difficult. As I always fly out of the app as soon, as I login into the integration via HAS.

For your testin I can grant some admin access to you. As I said, just do not make a toaster of it.

In Manual mode you can not change SB1, but SB2 then you can set values down to 10, see screenshots.
In AI mode you can set values down to 100 (50 for each SB1).

[image]
[image]
[image]
[image]

What exports do you need. Actually there is still some Energy in the batteries, that I can contain for testing.

RamsukKockoi, 2025-01-23:
In Manual Mode, I can change Values for the SB2 to deliver Power. They are taken by the device, as they were set.

After setting SB2 in manual Mode to 170 W:

sensor.system_fluxkompensator_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: aa6eeb96-1d06-409e-a575-1e130d9f9bf2
      site_name: FluxKompensator
      site_img: ""
      device_type_list:
        - 3
        - 6
      ms_type: 1
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C1
        - SHEMP3
        - A17C0
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "26.14"
        unit: kWh
      - type: "2"
        total: "26.06"
        unit: kg
      - type: "3"
        total: "8.37"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: APCGQ80E31400095
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "93"
          bind_site_status: ""
          charging_power: "0"
          power_unit: W
          charging_status: "1"
          status: "1"
          wireless_type: "1"
          main_version: ""
[... 453 lines cut ...]
          bind_site_status: ""
          wireless_type: ""
      photovoltaic_to_grid_power: "8"
      grid_to_home_power: "0"
      grid_status: 0
      has_history_data: true
    is_downgrade: false
    error_code: 0
    smart_plug_info: null
    feature_switch:
      0w_feed: true
      heating: false
      multi_pv: true
      shelly_meter: true
      smart_plug: true
    other_loads_power: "281"
    priority_discharge_switch: 0
    display_priority_discharge_tips: 0
    priority_discharge_upgrade_devices: ""
    style_id: 0
    is_show_priority_discharge: 0

[Markus Kociok_after having SB2 still on AI Input 50.zip](https://github.com/user-attachments/files/18517310/Markus.Kociok_after.having.SB2.still.on.AI.Input.50.zip)
[Markus Kociok_after seeting SB2 to AI via API.zip](https://github.com/user-attachments/files/18517308/Markus.Kociok_after.seeting.SB2.to.AI.via.API.zip)
[Markus Kociok_after setting SB2 via API to 170.zip](https://github.com/user-attachments/files/18517309/Markus.Kociok_after.setting.SB2.via.API.to.170.zip)

RamsukKockoi, 2025-01-23:
Hey, if you go back to the app, after the integration had owner access, there is an error message for me. Have a look at the screenshot, please.

[image]

thomluther, 2025-01-23:
Yes, that might be because the Api sent a schedule that may not be acceptable by the device at that point in time.
Also when you change output entities in HA, those changes can only be applied via a schedule to the Power system.

For the examples, you don't have to paste here the sysinfo part. This is contained in the export anyway and here they just take space and the hirarchy is lost because the HA Action prints the json in YAML format that shows hirarchy only by indents, which are completly missing here in the repo.
If you want to paste structured text, please use the code block function to maintain the indents.

Regarding admin access for schedule testing will be an issue anyway, because each time I will use it will kick off your token and vice versa. So if you use it in HA at same time, we will both quickly end up in banned IPs because HA will try to re-authenticate on each refresh cycle. I believe more than 10 authentication requests per IP per 24h will cause a 24h IP ban...

RamsukKockoi, 2025-01-23:
OK.
For my Homeassistant, I actually do not need owners Access. I can use a second account, just to read the data.

So you could test for a few hours, if needed. I should be much easier, then asking me all the time.
I would change the Password for that ...

thomluther, 2025-01-24:
> So you could test for a few hours, if needed. I should be much easier, then asking me all the time.
> I would change the Password for that ...

Yes, but that might be the challenge since I never know when I find time to do some work on it...
So it could always be conflicting.
Anyway, proper schedule support is something for a later update. 
What I found in the exports however is that I have to implement an immediate refresh of the SB1 schedule as well when SB2 schedule is being changed, Otherwise the cached SB1 schedule for instance is still showing the deactivated SB1 schedule when SB2 was switched from manual to AI mode and vice versa. (The SB2 mode is part of the SB2 schedule)

RamsukKockoi, 2025-01-26:
Hey, do you actually need anything?

thomluther, 2025-01-27:
> Hey, do you actually need anything?

Not at this point. I need to monitor the values for a couple of days and work on other fixes for the next release.

I'll let you know when I may have time and need owner access for doing work on the schedules for combined systems.
But I don't think this is high priority. In principal this should work, when changes are done only on the system that is active for schedule changes, which is:
- Manual Mode: Only SB2 schedule changes possible
- AI Mode/Smart Plug mode: Only SB1 schedule changes possible
Anything else may lead to unpredictable results or simply being ignored in the active usage mode.
I have no Idea either, how the entities that cannot be used actively at a given point in time should be presented. Eventually they should be made unavailable?

RamsukKockoi, 2025-01-28:
Do you need a beta tester? I would give it a try.

thomluther, 2025-01-28:
> Do you need a beta tester? I would give it a try.

You can test it with the Api library and the solarbank monitor. The HA integration is not ready yet

thomluther, 2025-01-28:
Here is an example. To me the calculated system totals make more sense and seem to match expected numbers based on cascaded systems. Also the total SOC is now weighted across all battery packs.

```
Solarbank Monitor (refresh 30 s, details refresh 300 s):
Sites: 2, Devices: 6
================================================================================
System         : FluxKompensator  (Site ID: #masked#)
Type ID        : 5 (Solarbank)           Device models  : #masked#
Cloud-Updated  : 2025-01-28 13:26:50     Valid Data     : YES (Requeries: 0)
SOC total      :   30 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :  352 W                  Battery Pwr Tot:  352 W
Output Pwr Tot :    0 W                  Home Load Tot  :    0 W
--------------------------------------------------------------------------------
Device [A17C1] : Solarbank 2 E1600 Pro   Alias          : Solarbank 2 E1600 Pro
Serialnumber   : #masked#                Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : charge_priority         Status code    : 3
State Of Charge:   43 %                  Min SOC        :   -- %
Battery Energy :  688 Wh                 Capacity       : 1600 Wh
Exp. Batteries :    0 Pcs                AC socket      :    0 W
Solar Power    :  186 W                  Output Power   :    0 W
Solar Ch_1     :    0 W                  Solar Ch_2     :   91 W
Solar Ch_3     :   95 W                  Solar Ch_4     :    0 W
Other Input    :  --- W                  Heating Power  :    0 W
Battery Power  :  186 W                  Device Preset  :    0 W
Home Demand    : 2290 W                  SB Home Load   :    0 W  (-)
Smart Plugs    :  --- W                  Other (Plan)   : 2290 W
Schedule  (Now): 13:26:57 UTC +0100      System Preset  :    0 W
--------------------------------------------------------------------------------
Device [A17C0] : Solarbank E1600         Alias          : Solarbank Wohnzimmer
Serialnumber   : #masked#                Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : charge_priority         Status code    : 3
State Of Charge:   21 %                  Min SOC        :   -- %
Battery Energy :  336 Wh                 Capacity       : 1600 Wh
Solar Power    :   67 W                  Output Power   :    0 W
Battery Power  :   67 W                  Device Preset  :    0 W
Schedule  (Now): 13:26:57 UTC +0100      System Preset  :    0 W
--------------------------------------------------------------------------------
Device [A17C0] : Solarbank E1600         Alias          : Solarbank Wintergarten
Serialnumber   : #masked#                Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : charge_priority         Status code    : 3
State Of Charge:   26 %                  Min SOC        :   -- %
Battery Energy :  416 Wh                 Capacity       : 1600 Wh
Solar Power    :   99 W                  Output Power   :    0 W
Battery Power  :   99 W                  Device Preset  :    0 W
Schedule  (Now): 13:26:57 UTC +0100      System Preset  :    0 W
--------------------------------------------------------------------------------
Device [SHEMP3]: Hauptstrom Keller       Alias          : Hauptstrom Keller
Serialnumber   : #masked#                Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Grid Status    : ok                      Status code    : 0
Grid Import    : 2290 W                  Grid Export    :    0 W
================================================================================
```
What I noticed is that upon changes, the devices still report their data at different intervals (SB1 every 1 minute, SB2 every 5 min?)
The Cloud timestamp however shows the last refresh appearingly, but it does not mean that all devices send their data at that time.
Due to large update time lags, the total calculations may be offset,
Eg. the SB1 may report output power 100 W, but the SB2 does not report this yet as PV input.
The totals calculate SB1 PV + SB 2 PV - SB 1 Out => Total PV.
So in such situations, the totals are also wrong, but can only be calculated from actual data at time of Api response.

RamsukKockoi, 2025-01-28:
Hi,

the SB2 does not see Solarpower from SB1, if they are set to 0 output. So if SB1 is charging with 100W and there is a 0-W Switch, SB2 will not report PV-Input from SB1.

If SB1 is set to 100W, gets 200W from PV-Input, it reports 100W charging and 100W output to home. That output ist what SB2 ist now seeing as PV-Input.

The actually different calcutlatable outputs and totals, lead to weird EnergyDashboard effects.

As the battery charging in ED reduces the home use of Solarpower. So now you have 2kWh Solarpower, but 1 went into SB1 with 0-Switch set to 0 you will not see any Solar Energy in the dashboard. 1 kWh went into SB1 and was never seen in the actual totals. And the one, that was seen in the actual totals, is "eaten" by the 1kWh SB1 has charged.
But no matter, it will appear if the Sun goes down an SB1 discharge. As now it is recognized as PV-Input in SB2.

Actually I really think about, to disconnect the SB1 from SB2 and make SB2 to a night energy system with 2 own PV-modules.

But not before you say, you could now solve all the riddles Anker gives us with that.

Btw. I am noticing, that SB2 seems to actuatlize PV-Inputs regularly every Minute, but not things as the states. F.e. "charging" to "bypass charging" and so on.

thomluther, 2025-01-28:
> the SB2 does not see Solarpower from SB1, if they are set to 0 output. So if SB1 is charging with 100W and there is a 0-W Switch, SB2 will not report PV-Input from SB1.

I know, therefore all Anker 'System totals' and Energy statistics for combined systems are wrong.

> The actually different calcutlatable outputs and totals, lead to weird EnergyDashboard effects.

The only thing calculated for a device is the battery power to have a consistent and valid value (there is nothing from the cloud).
Batt power is calculated by Solar Input - Output power of the device (so the difference must go in or out of the battery, not considering any losses). This calculation had to be done also for the totals in the system.
So on the device level, this is still correct, but system battery power cannot be longer correct when using incorrect total values from the cloud.

I changed that completely. Now totals will be accumulated from individual device levels as they are meaningfull and correct in a cascaded setup, where output of cascaded device = solar input of SB 2 device.
Therefore the new totals will be correct and once the new version is released, you should be able to use the corrected system totals for energy dashboard.
HOWEVER: The total battery power is just a 'net' power. If SB1 device is discharing 100 W and SB2 is charging 100 W, the System total battery power will be 0W. 
So if you want to track charging and discharging energy completely, I recommend track those energies on a device level and add them all to the dashboard. The dashboard will automatically accumulate them for totals per period, and you maintain charge/discharge tracking on device level which you loose when tracking on the total charging energies...

> Btw. I am noticing, that SB2 seems to actuatlize PV-Inputs regularly every Minute, but not things as the states. F.e. "charging" to "bypass charging" and so on.

Are you sure about the interval? You can only verify SB2 update frequency if you watch how often the SB2 power device entities will change (not the totals or SB1 values). 
The cloud timestamp most likely is from the last received update, which is probably from any SB1 that sends data every minute...
Regarding the states I cannot say for sure whether that is differnt now. For SB1 it should be correct with the new version which uses the correct SB1 count and totals of the system. For SB2 states I never got into details if they are correctly interpreted at all since I cannot monitor or compare it.

RamsukKockoi, 2025-01-29:
Hey, I am sure about the refreshing frequencies, both, the Power Values seem to come every Minute and the states often come later after a few minutes. For the states I am sure it mostly hits the SB2. In the app I already see "bypass discharging", the integration shows a plausible value, but the state does not change to "bypass discharge" for a period of a couple of minutes.

RamsukKockoi, 2025-01-29:
Do you  already have an Idea of the new Release ? I  had the chance zu buy extension batteries. So I am planning a new System contelkation.Do you need more Information?

thomluther, 2025-01-29:
> Do you already have an Idea of the new Release ? I had the chance zu buy extension batteries. So I am planning a new System contelkation.Do you need more Information?

Maybe end of next week. I'm in the middle of adding some stuff for AC model which needs more testing before the new release.

RamsukKockoi, 2025-02-05:
Hey there, I changed the configuration. As you may can see.
Anker did a good job by making the devices conpatible, but for me it did not work as good as I expected.

Thank you for your great work!

thomluther, 2025-02-05:
Hi @RamsukKockoi 
Yes I noticed yesterday that the combined system was not visible anymore.
You can maybe retry with the new release once available. At least in HA you should get better power values for the totals, which allows doing your own statistics for the whole system.
It will also have improvements to the SB1 changeable entities depending on the state and report proper values accordingly.
It is a bit tricky to deal with the 2 different schedules that can now be active for SB1.
I had to make the SB1 schedule entities and service 'unavailable' to avoid they try changing the SB1 schedule while the 'enforced SB1 schedule' from SB2 manual mode is active...

RamsukKockoi, 2025-02-06:
Hey,

I have to see how to realize it. As the SBo went over to my moms system.
After the update, I will try to get one battery back for a while.

thomluther, 2025-02-06:
The 2.5.0 beta is available in the main branch.
You can update via the Action install update and selecting the Anker Solix update entity. For the version you need to specify 'main' as tag for the main branch.
Note: After installation and restart, HA will continue to report a repair action since it believes the available release version is newer than the installed 'main' version. Simple ignore that until the 2.5.0 has been released.
If you have any findings with the beta, please let me know. I don't plan code changes anymore, only documentation updates prior release next week.

thomluther, 2025-02-06:
Support for combined systems added to 2.5.0
The 2.5.0 beta is available in the main branch.
You can update via the Action install update and selecting the Anker Solix update entity. For the version you need to specify 'main' as tag for the main branch.
Note: After installation and restart, HA will continue to report a repair action since it believes the available release version is newer than the installed 'main' version. Simple ignore that until the 2.5.0 has been released.
If you have any findings with the beta, please let me know. I don't plan code changes anymore, only documentation updates prior release next week.

Since the SB1 are toggeling between different schedule plans depending on the SB2 usage mode, the SB1 schedule entities will be set to unavailable when SB1 plan is enforced by SB2.
Fixes have been implemented to support unavailable entities and prevent their values can be modified while unavailable. 
Note: The UI may reflect this differently, especially for unavailable number entities. Just when you refresh the browser, the value will go back to the default display value when unavailable (which is middle of any defined value range for the entity).
I think it is weird, that an unavailable number entity is represented that way, it should be greyed out and not allow any changes while unavailable...)

thomluther, 2025-02-11:
Delivered with 2.5.0
Can be re-opened if there are further findings

closed 2025-02-11 by thomluther