thomluther/ha-anker-solix #183: "State of Charge" is no longer being provided by the anker_solix integration

BillyFKidney, 2025-01-11:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4987
Installed Version | 2.0.2
Stage | running
Available Repositories | 1501
Downloaded Repositories | 21

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 42

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 8, 2025 at 4:00 PM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | us-east-1-9.ui.nabu.casa
certificate_status | ready
instance_id |
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 228.8 GB
disk_used | 48.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Terminal & SSH (9.16.0), Samba share (12.3.2), Mosquitto broker (6.4.1), Samba Backup (5.2.0), eufy-security-ws (1.9.1), Silicon Labs Multiprotocol (2.4.5), Matter Server (7.0.0), Network UPS Tools (0.14.0), Whisper (2.4.0), openWakeWord (1.10.0), Piper (1.5.2), ESPHome Device Builder (2024.12.2), Zigbee2MQTT (2.0.0-2), MariaDB (2.7.2), RPC Shutdown (2.5), SwitchBot-Mqtt (1.0.31), Music Assistant Server (beta) (2.4.0b12), Grocy (0.23.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 16
-- | --
resources | 7
views | 22
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | December 29, 2024 at 8:59 PM
-- | --
current_recorder_run | January 10, 2025 at 6:21 PM
estimated_db_size | 2212.97 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Integration version

2.4.1

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The state of charge is no longer being reported after updating the Home Power Panel to version v1.4.4.

Anker SOLIX Firmware Versions:
[image]

State of Charge Entity:
sensor.solix_home_power_panel_state_of_charge
[image]
[image]

### Reproduction steps

1. Update Firmware
2. Entity Disappears
3. ...
4. _¿Profit?_
[image]

### Debug logs

```text
Not necessary
```

thomluther, 2025-01-11:
@BillyFKidney pls provide a system export for your integration entry to see the responses from the Api.
Without this I cannot determine what has changed...

BillyFKidney, 2025-01-11:
[image]
Sorry about that, I guess my laptop had different ideas!

[home-assistant_anker_solix_2025-01-11T14-05-22.233Z.log](https://github.com/user-attachments/files/18386101/home-assistant_anker_solix_2025-01-11T14-05-22.233Z.log)

thomluther, 2025-01-12:
I checked meanwhile with a powerpanel system where I have shared access.
It seems the cloud statistics for the SOC are broken since a while. In the past, the daily SOC charts never showed 0 values in between. Now they show lots of 0% in the past daily charts.
I don't think this is a firmware problem, but don't know for sure. You should verify that in your SOC history charts and see when the 0% data points started and open a Ticket with Anker. They have to fix that.

The SOC and Power values of the integration can just be extracted from the last valid data point of the daily statistic charts. So if they are not reliable, the integration will show garbage. Furthermore there is timestamp in the cloud Api to identifiy the time offset between the Api client and the last valid data point in the cloud. The Api library needs to find the last valid statistics time and calculate the offset based on that. I have to change that algorithm, since it never considered that SOC might be 0% before the last valid SOC entry. The algorithm will now need to look from future back in time, to find the first time stamp that is not 0% and use this for offset calculation.
Note: The offset calculation is just done during the first 5 minute interval during each refresh (typically  every 60 seconds). Once the minimum offset is determined, it will not be recalculated, but stat data only re-queried every 5 minutes to limit Api traffic, since those intraday responses are large and cannot be limited. If the cloud may have an invalid 0% entry on the last datapoint when the offset calculation is done, it may happen that the offset calculation will become larger than necessary, e.g. 10-30 minutes instead of 5 minutes delay for the reported values in the integration. Therefore it is important that Anker will fix these 0% SOC entries in the intraday charts.
You can trigger a recalculation of the offset via reload of the config entry or by toggling the Api Refresh switch off and on.

thomluther, 2025-01-13:
@BillyFKidney , I found a new statistics type of 'diesel'.
Do you know what that could be used for in a power panel system?
Is there any new chart already in the mobile app referring to 'diesel'?

Here is the structure of a single day query in the weekly format, there is also an indicator whether it `"hasDiesel": false`
I guess when this is false, the chart tab is not presented in the app, but I have no idea what diesel means and how that could be measured in electrical kWh. Does the power panel provide a separate Input connector for a fuel driven generator and this is what will be produced by the generator?
```
    "totalEnergy": "0.00",
    "totalEnergyUnit": "KWh",
    "totalImportedEnergy": "",
    "totalImportedEnergyUnit": "",
    "totalExportedEnergy": "0.00",
    "totalExportedEnergyUnit": "KWh",
    "power": null,
    "powerUnit": "",
    "chargeLevel": null,
    "energy": [
      {
        "value": "0.00",
        "negValue": "0",
        "rods": [
          {
            "from": "0.00",
            "to": "0.00",
            "sourceType": "diesel"
          }
        ]
      },
      {
        "value": "0.00",
        "negValue": "0",
        "rods": [
          {
            "from": "0.00",
            "to": "0.00",
            "sourceType": "diesel"
          }
        ]
      }
    ],
    "energyUnit": "KWh",
    "aggregates": [
      {
        "title": "Battery charging capacity",
        "value": "0.00",
        "unit": "KWh",
        "type": "hes",
        "percent": "0%",
        "imported": false
      },
      {
        "title": "Load power consumption",
        "value": "0.00",
        "unit": "KWh",
        "type": "home",
        "percent": "0%",
        "imported": false
      }
    ],
    "hasDiesel": false
  },
```

thomluther, 2025-01-17:
@BillyFKidney, can you confirm whether the 0% values in you app history are now gone?
If they are gone, you should be able to reload the configuration entry and that should fix the SOC value.
I changed the algorithm for the next release, so it will always be able to extract the last valid data points from the intraday stats (that point which shows the last valid SOC value different from 0)

Sabrown365, 2025-01-23:
@thomluther State of charge has also gone missing for me. I reloaded the configuration and it removed my two F3800's in HA

[image]

These are the values that are loaded after reconfiguring

SOLIX Home Power Panel
[image]

System Home
[image]

[nooxxx_2025-01-23_1159.zip](https://github.com/user-attachments/files/18524822/nooxxx_2025-01-23_1159.zip)

BillyFKidney, 2025-01-24:
Oddly the state of charge was reporting for about two hours today...then disappeared again.

[image]

BillyFKidney, 2025-01-24:
> [@BillyFKidney](https://github.com/BillyFKidney) , I found a new statistics type of 'diesel'. Do you know what that could be used for in a power panel system? Is there any new chart already in the [...]

I only have the main grid and solar, no idea what diesel would be.

We have 2x F3800s coupled with 2x BP3800s each.

thomluther, 2025-01-24:
> Oddly the state of charge was reporting for about two hours today...then disappeared again.

This is nothing that can be fixed in the integration.
When the cloud reports 0, that integration has to do as well. You will find those gaps typically also in the App statistics.

What will be fixed is that the library finds the last data point properly when the Api is started (or Integration reloaded). An if those was not the last point at time of start, it should adjust the offset into the future once later data points are found on next queries.

So it will also adjust automatically to the last available data points if started during such a SOC gap.
But the SOC is the only statistic that can be used to identify the time offset between HA and the cloud reporting, since SOC is never expected to be 0 (unlike other power values could be 0, except the grid consumption maybe...)

thomluther, 2025-01-24:
> [@thomluther](https://github.com/thomluther) State of charge has also gone missing for me. I reloaded the configuration and it removed my two F3800's in HA, I think this is because I changed the email/pass to a second member of the home that is not the main.

I assume there have been cloud issues when you reloaded the integration. You will have to wait and retry another time.
The 0 values for the Power Panel also indicate, that the integration was not able to find the last valid data point in the energy stats.
Maybe there have been 0 SOC gaps again in last 2 days, which the current version cannot handle when searching for the first invalid time stamp.

thomluther, 2025-02-06:
This is fixed in 2.5.0.

The 2.5.0 beta is available in the main branch.
You can update via the Action install update and selecting the Anker Solix update entity. For the version you need to specify 'main' as tag for the main branch.
Note: After installation and restart, HA will continue to report a repair action since it believes the available release version is newer than the installed 'main' version. Simple ignore that until the 2.5.0 has been released.
If you have any findings with the beta, please let me know. I don't plan code changes anymore, only documentation updates prior release next week.

thomluther, 2025-02-11:
Delivered with 2.5.0

closed 2025-02-11 by thomluther