thomluther/ha-anker-solix #187: Add item for device system battery temperature

HakunaMatata47, 2025-01-13:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

In the Anker App you see in the Device section   a temperature information.

It would be helpful to have that temperature information also in the Solix Integration.
I.e. to understand why the battery is not charged (usually temperatures below 2 degrees)
or to set a warning notification for the hotter days.

### Describe the solution you'd like

sensor.solix_site_sb_akkutemperatur   

 - just an idea

### Describe alternatives you've considered

or similar

### Additional context

I think it is missing right now as I don't find an entity with that information in the current Integration version

thomluther, 2025-01-13:
Hi,
The device temperature is not available via Cloud Api at this time, see [Add temperature entity](https://github.com/thomluther/ha-anker-solix/issues/26).
There is a new heating power field in the Api scene info response, but I have no idea how it is used for solarbanks.
You will have to monitor the Api response from time to time, especially when you see the heating on in the App.

You can use the Action to get Anker System information to fetch that data from Api.

thomluther, 2025-01-17:
@HakunaMatata47 
Any success in getting System info via the service? Does it show values other than 0 in pei_heating_power field?
Here are the new solarbank fields that are unknown and needs interpretation and correlation to App values that might be shown...
```
    "other_input_power": "0",  => Unknown? Maybe DC input for battery charging when grid to battery is AC?
    "micro_inverter_power": "0",   => AC Output power for SB2? Should be lower than other output power values, if those are DC?
    "micro_inverter_power_limit": "0",   => Inverter output limit? Used for all SB2 models?
    "micro_inverter_low_power_limit": "0",  => Unknown
    "grid_to_battery_power": "0",   => DC or AC charging power?
    "pei_heating_power": "0"   => Total heating power for the batteries?
```

HakunaMatata47, 2025-01-17:
I see those values but especially pei_heating_power is 0 for me as I have a non-heated SB1 battery model.
In the app you can see the temperature value but I don't see that in the system or device  info when probing it with test_api.py
Would be cool to get the battery temperature info if it can be identified in the API - maybe some day.

as to the unknown fields:

"total_charging_power": "32"
"total_photovoltaic_power": "32"
"micro_inverter_power": "32"

Those have the same value while I look at it and it probably means it gets 32W from the solar panel and turns that into inverter power
used i.e. to load the battery or put trough to the house

From the naming of the unknown fields I come to the same conclusion as in your comments.

thomluther, 2025-01-22:
"micro_inverter_power": "32" => all micro_inverter_* field were introduced with the AC model as well as the pei_heating_power field, Probably the inverter fields are all non-sense for any other SB2 model, e.g. micro_inverter_power power is always the same at PV channel 3 power, which does not make sense at all.
I never saw pei_heating power other than 0....maybe it will come some day, maybe just for certain models and with certain firmware levels....
Total charging power response from Api is wrong. The library recalculates and reuses that total field to reflect battery power (positive and negative)

thomluther, 2025-02-06:
A new field is in the scene info that reports a heating flag. This was added to the 2.5.0 release, but its not clear whether it is used when any heating in the system is active.
If there are only short heating periods, you also may never see it active since the cloud data is refreshed only every 5 minutes by the SB2 devices...

thomluther, 2025-02-08:
I finally got an example where both is visible, the heating flag field for the system as well as the heating power for the device:
```
================================================================================
System         : Hammeke Base  (Site ID: ###masked###)
Type ID        : 11 (Solarbank)          Device models  : A17C2,A17X7
Cloud-Updated  : 2025-02-08 10:59:01     Valid Data     : YES (Requeries: 0)
SOC total      :   21 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :  586 W                  Battery Pwr Tot:  481 W
Output Pwr Tot :    0 W                  Home Load Tot  :    0 W
Active Mode    : Smartmeter (1)          Heating        : ON
Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
--------------------------------------------------------------------------------
Device [A17C2] : Solarbank 2 E1600 AC    Alias          : Solarbank 2 E1600 AC
Serialnumber   : ###masked###            Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : detection               Status code    : 0
State Of Charge:   21 %                  Min SOC        :   -- %
Battery Energy :  672 Wh                 Capacity       : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket      :    0 W
Solar Power    :  586 W                  Output Power   :    0 W
Solar Ch_1     :   37 W                  Solar Ch_2     :  189 W
Other Input    :    0 W                  Heating Power  :  105 W
Inverter Power :  360 W                  Grid to Battery:    0 W
Inverter Limit :  800 W                  Low Limit      :    0 W
Battery charge :  481 W                 
Battery Power  :  481 W                  Device Preset  :    0 W
Home Demand    :    0 W                  SB Home Load   :    0 W  
Smart Plugs    :  --- W                  Other (Plan)   :    0 W
Schedule  (Now): 10:58:59 UTC +0100      System Preset  :    0 W
================================================================================
```
So PV1 + PV2 + external inverter power => Solar Power

thomluther, 2025-02-11:
Delivered with 2.5.0