thomluther/ha-anker-solix #190: Long time between updating power setpoint and reaction in actual power output

imLeeat, 2025-01-26:
### System Health details

After changing the power setpoint in HA it is shown in the App within a second.
But the power output changes only after a few minutes.

If I change the setpoint directly in the App the power output always after 4-5 seconds

### Integration version

2.4.1

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After changing the power setpoint in HA it is shown in the App within a second.
But the power output changes only after a few minutes.

If I change the setpoint directly in the App the power output always after 4-5 seconds

### Reproduction steps

1.changing Power setpoint in HA
2.power setpoint changes in App
3.output changes after multiple minutes
...

### Debug logs

```text
-
```

imLeeat, 2025-01-26:
Why invalid?

thomluther, 2025-01-27:
Hi, this is not an issue with the integration.
When you see the result directly in the App, it is active in the cloud.
It seems the SB2 does react much slower on schedule changes via Api than SB1.
When you change the schedule in the App, the App can also use the MQTT server to push it directly to the device, since the SB2 MQTT connection ist triggered by the App when opened as System owner.
Maybe the SB2 synchronizes it's schedule only in the 5 min interval that it connects to the cloud for data updates...

In any way, nothing that can be changed in the Integration.

imLeeat, 2025-01-27:
I understand your point.
But isn’t it possible that just an “update bit” is missing? Some bit toggle that tells the device to react immediately to the changed setpoint?

thomluther, 2025-01-27:
No, The whole schedule is sent to the Api, and when you change only the power, the whole schedule must still be send with the modified power value.

imLeeat, 2025-01-27:
yes understood, but maybe the app sends some kind of bit additionally? some kind of "update now bit"

thomluther, 2025-01-27:
Could be, but not via the cloud Api. If it does it, then via the MQTT server where I have no interface for.

imLeeat, 2025-01-27:
than how its working with the smartmeter? does the smartmeter has also MQTT connection?

imLeeat, 2025-01-27:
anyhow, dont mind. thanks for you explanations. 
i dont like the situation that it takes a couple of minutes for reaction. 
tomorrow i will get some free sockets, maybe i figure something out how to get a direct trigger.

Thanks for your time, i appreciate your responses

closed 2025-01-27 by imLeeat

imLeeat, 2025-01-27:
> [@imLeeat](https://github.com/imLeeat) I just stumbled across the method to set apply the schedule. The Api endpoint requires an additional command field which is integer. It always has been 17 and I don't know its purpose. data = { "site_id": siteId, "param_type": paramType, "cmd": command, "param_data": json.dumps(paramData), # Must be string type }
> 
> param_type must be "6" for SB2 schedules.
> 
> You may give it a try using the Api library and some [test code as shown in the repo](https://github.com/thomluther/anker-solix-api?tab=readme-ov-file#usage).
> 
> You can simply use the methods:
> 
>     * [get_device_parm](https://github.com/thomluther/anker-solix-api/blob/c0901968fb5549df45d03de913d26ff7f6268119/api/schedule.py#L129) to query the schedule, use parmType "6" for SB2 schedule in your site ID
> 
>     * [set_device_parm](https://github.com/thomluther/anker-solix-api/blob/c0901968fb5549df45d03de913d26ff7f6268119/api/schedule.py#L210) to set a modified schedule object. Here you can play/test with other cmd type integers. (No guarantee what will happen)
> 
> 
> Examples: _out(await myapi.get_device_parm(siteId=siteId, paramType="6")) _out(await myapi.set_device_parm(siteId=siteId, paramData=scheduleSB2, paramType="6", command=17, deviceSn=deviceSn))
> 
> The set_device_parm method takes an optional device SN. Although not required for the Api call, it is used to update the schedule in the Api cache for the correct device.
> 
> If you find different behavior with other command types, you can reopen the issue and we can see how to implement a proper change. The safest way to understand how the App sends the data is to sniff the traffic. But Api content is typically ECDH encrypted with different tokens for each client. So pretty much impossible for traffic sniffing...

thanks for that suggestion, but at them moment doing that requires more knowledge than I currently have.
is it possible to execute those methods with Homeassistant?

thomluther, 2025-01-27:
> thanks for that suggestion, but at them moment doing that requires more knowledge than I currently have.
> is it possible to execute those methods with Homeassistant?

No, You need to setup the Api library in a Python runtime (3.12 or 3.13).
Then you can use Python code as in the usage example to use any method of the library.

I was thinking to develop a service for HA to run Api calls directly with your account, but that may be also dangerous is people don't know what they are doing...
Maybe sometime in the future, but not in the next couple of month

imLeeat, 2025-01-27:
Today I did some measurements and found out something nice:
i installed a smart socket with electricy meter on the AC output side of SB2.
Everytime i changed the setpoint in HA, the SB2 reacted immediatley (within 2 seconds).
Even if the values "actual poweroutput" did not changed in the app and not in HA.
So the issues is not that SB2 is not changing the output power if i change the setpoint in HA.
The issue is that SB2 is only sending new updated values into the cloud after its 5 minutes cycle.

So it's a purely visual problem that I can live with very well at the moment. Sorry for the incorrect representation of the problem

imLeeat, 2025-03-10:
Hier das Feedback von Anker zur Fragestellung ob es möglich ist die Daten genauer als alle 5 Minuten über API zu bekommen:

"Wir arbeiten aktiv an einem Software-Update, um die Datenprotokollierungsfunktionen zu verbessern und genauere Statistiken bereitzustellen. Vielen Dank für Ihr Verständnis und Ihre Unterstützung."