thomluther/ha-anker-solix #194: [Question] Discharge priority does nothing.. Bug or am I holding it wrong?

orangecoding, 2025-01-31:
### System Health details

Hey,

Out of curiosity, I was clicking the button. My solarpanels still produce around 150W atm. Solarbank is at 100% and on Bypass. I was expecting to see in the Anker app that the Battery begins to discharge, but it was still on Bypass. I waited for 5 mins but it had no effect. Solarbank E1600 (V1) is on latest firmware

### Integration version

see above

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

see above

### Reproduction steps

see above

### Debug logs

```text
none
```

orangecoding, 2025-01-31:
I just tried it again. Solarpanels only deliver 77 Watts anymore, still it does not work. The mode stays on "Bypass". I assume I do not have to do anything else in Home assistant rather then setting the discharge prio button?

orangecoding, 2025-01-31:
Mhm. Interestingly, I checked again and it SEEMS to work, but it is extremly slow. If I set it in Home assistant, it takes around 10 mins to be reflected on the app. And if I set it back, it takes even longer to switch back on the solarpanels

thomluther, 2025-01-31:
> Mhm. Interestingly, I checked again and it SEEMS to work, but it is extremly slow. If I set it in Home assistant, it takes around 10 mins to be reflected on the app. And if I set it back, it takes even longer to switch back on the solarpanels

Yeah, as I said, no idea how Anker implemented it, but like other changes in the schedule, it should be directly applied by the solarbank, e.g. changing the output power.
However, if you verify only the values in the App/Integration for SB1, it may take up to 1 minute to see refreshed data.
If you have a smart plug on the SB output, you could see changes quicker.

So for the new switch it could have been implemented slowly. Typically also the auto discharge rule needs at least 10 min to switch from discharge to PV generation whenever it was kicked in shortly.
So often 10 minute PV loss and discharge if there is a short cloud and output preset was e.g. 250 W.

I assume they did not improve the toggle behavior, but only go into the same code functions when the switch is enabled (like  automatically triggered, but never automatically toggeled back to PV, unless the switch is disabled again...

thomluther, 2025-01-31:
It still questionable, whether the switch works at all while PV > 100 W
All auto trigger require PV < 100W.
I assume Anker made the implementation as simple as possible, so probably the switch just enforces/overrules the automatic rule condition (while the switch is enabled).
That means when switch enabled, it will delay 10 minutes and PV must be < 100 W to toggle to discharge mode.
When disabling the switch it might be the same: It no longer overrules the auto rule condition and the solarbank may toggle back to PV production after some delay (min 10-20 minutes)...

closed 2025-02-11 by thomluther

orangecoding, 2025-02-23:
Oh god damn it, sorry, I did not see you asked an additional question. Everything works now. Sorry again ❤