thomluther/ha-anker-solix #195: Question "set power"

heiseb02, 2025-02-02:
### System Health details

System: Solarbank 2 with Smartmeter

### Integration version

2.4.1

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Does anyone else have the same problem with setting the output power in the manual mode?

Each set with the slider is confirmed by Homeassistant with "The action number/set_value could not be executed. list index out of range".
I have been trying since yesterday, and I mean yesterday, to be all right.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Logger: homeassistant.components.websocket_api.http.connection
Quelle: components/websocket_api/commands.py:245
Integration: Home Assistant WebSocket API (Dokumentation, Probleme)
Erstmals aufgetreten: 11:57:50 (15 Vorkommnisse)
Zuletzt protokolliert: 13:37:08

[140123106639104] Unexpected exception
[140123264027808] Unexpected exception
[140122592404832] Unexpected exception
[140122747781472] Unexpected exception
[140123177316080] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1006, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1078, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/anker_solix/number.py", line 329, in async_set_native_value
    resp = await self.coordinator.client.api.set_sb2_home_load(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/config/custom_components/anker_solix/solixapi/schedule.py", line 1334, in set_sb2_home_load
    weekdays = weekdays | set(rate_plan[index].get("week") or [])
                              ~~~~~~~~~^^^^^^^
IndexError: list index out of range
```

DHLF, 2025-02-02:
Just a thought..

**You MUST keep holding the mouse key while selecting the desired power level!!**
Each time you lift your mouse button the value will be sent via the api to the device.
If you send new settings without a pause of at least half a minute the error log will fill. The newer version also reminds you with a nice little window to keep a minimum time limit between each new setting.

I saw an additional strange behaviour while using the api with ha:
This year the data fetching time was around five+ minutes -.- so you may have to wait more patiently until you may see a change.

The error _list index out of range_ seems to be a *bad* or misstyped/copied char the config but i assume you did NOT touch it?

You may also try to get near your solarbank 2(with phone) and check if the device is continuesly connected to your wlan.
There are sometimes issues with interference from other networks but the errors will NOT be shown in HA, only in the phone app, like repeating losing connection/reconnection..

Also the behaviour of the solarbank is wiered, if you set power output manually to 0.. nothing happens if solar input is lower than ~40W(above 40W it will charge the battery). Set it to 10W and surprise.. now at least those little power is put into the net.
Whenever the device loses wlan connection (or the router is rebooting).. the smartmeter is SHOWN but gets in fault mode error 35 -.-
and STAYS this way. I have to remove the smartmeter using the app and reinstall it again to get it work -.- every single time -.-

thomluther, 2025-02-03:
> File "/config/custom_components/anker_solix/solixapi/schedule.py", line 1334, in set_sb2_home_load
>     weekdays = weekdays | set(rate_plan[index].get("week") or [])
>                               ~~~~~~~~~^^^^^^^
> IndexError: list index out of range

Hi @heiseb02 , this is a weird problem. The code is getting the index number from the Api response of the schedule plan.
Typically that should be sequential and match the number of elements in the plan.
If there is an index out of range error, it means there exists a weird schedule in your system, where the index numbering may have a gap.
The Api code typically adjust all subsequent index numbers if you delete a slot with the Api.
Can you provide an Api export to see how the actual schedule looks like?

thomluther, 2025-02-05:
Hi @heiseb02, what is the status of your problem?
Please provide an Api export with the integration export action, or close the issue if no longer valid.
Thanks

heiseb02, 2025-02-05:
[home-assistant_anker_solix_2025-02-05T17-51-47.717Z.log](https://github.com/user-attachments/files/18676548/home-assistant_anker_solix_2025-02-05T17-51-47.717Z.log)

heiseb02, 2025-02-05:
[home-assistant_anker_solix_2025-02-05T17-51-47.717Z.log](https://github.com/user-attachments/files/18676691/home-assistant_anker_solix_2025-02-05T17-51-47.717Z.log)

thomluther, 2025-02-05:
Thanks.
Your plan index starts with 1, but should start with 0
'custom_rate_plan': [{'index': 1, 'week': [0, 1, 2, 3, 4, 5, 6],
The current code does not consider this weird index numbering.
I need to check, if that may have occurred by the code when modifying schedule.

You can clear their error by deleting the custom rate plan completely. Then the Preset goes to default 200 W and then the next modification should create a new single interval that starts with 1.
If that does not work with the integration, you have to temporarily fix your plan index via the App (delete all definitions and create one new slot, that the integration can then modify for the output.

BTW, you have a lot of other errors in the log from other entitites/integrations...

thomluther, 2025-02-06:
Proper index handling and cleanup of 'wrong' index numbers will be provided with 2.5.0

thomluther, 2025-02-06:
The 2.5.0 beta is available in the main branch.
You can update via the Action install update and selecting the Anker Solix update entity. For the version you need to specify 'main' as tag for the main branch.
Note: After installation and restart, HA will continue to report a repair action since it believes the available release version is newer than the installed 'main' version. Simple ignore that until the 2.5.0 has been released.
If you have any findings with the beta, please let me know. I don't plan code changes anymore, only documentation updates prior release next week.

heiseb02, 2025-02-06:
thank you.

now it works fine, great job.