thomluther/ha-anker-solix #198: Support of Anker SOLIX Solarbank 2 E1600 AC

stelzerroland, 2025-02-05:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I recently purchased the Anker **SOLIX Solarbank 2 E1600 AC** and tried integrating it with Home Assistant using the current version of this integration. However, only two devices are detected: **System** and **Account**—the Solarbank itself is **not recognized**.

I followed several tutorials where other versions of the Solarbank (non-AC models) are automatically detected without issues. Despite following the same steps, my AC version doesn’t show up, which leads me to believe that it is **currently not supported**.

Additionally, while the **Anker app** allows for dynamic power regulation, it requires both the **Solarbank and my Shelly 3EM smart meter** to be on the **same Wi-Fi network**, which isn’t possible in my setup. This makes it difficult to efficiently manage energy flow.

### Describe the solution you'd like

I would love to see support for the **Anker SOLIX Solarbank 2 E1600 AC** added to the integration, with the following key features:
- **Device detection and monitoring**, including battery status, power input/output, etc.
- The ability to **dynamically adjust the AC power output** via Home Assistant, allowing me to automate energy management based on real-time data from my Shelly 3EM.

This would eliminate the need for both devices to be on the same Wi-Fi network and enable smarter energy control within Home Assistant.

### Describe alternatives you've considered

- **Using the Anker app**, but it requires both devices to be on the same Wi-Fi network, which isn’t feasible.
- **Network bridging** to force both devices onto the same network, but this adds unnecessary complexity and reliability risks.
- **Custom API development**, but this would require more technical effort and time than I can currently invest.

### Additional context

- I’m happy to provide **API responses, debug logs, or assist with testing** if needed.
- My **Shelly 3EM** is already integrated into Home Assistant, and having the Solarbank supported would allow for advanced energy automation.
- The AC version of the Solarbank has only been available for a few months, so this feature might benefit other users with similar setups.

Thank you for your hard work on this integration!

thomluther, 2025-02-05:
While I implemented now all new AC values and mode switches, it appears that an output power change in manual mode is not really applied by the AC system that I can access.
But that systems seems to be in a weird state anyway. You will have to test whether output preset changes will be applied by your system once the new release is available.
If that won't work, you will need to dig into the Api and figure out what 'extra' calls or different paramters might be required to change the output preset in an AC schedule plan.

stelzerroland, 2025-02-05:
Hi,
thank you so much for your quick and detailed response!

It’s great to hear that support for the AC version is already in progress and tracked under issue #175. And sorry, for not recognizing this before.

To give you more context about my setup:
- I have a Shelly 3EM to monitor the overall household consumption/energy surplus.
- I also have a Shelly 1PM connected to the Solarbank’s output, which provides real-time monitoring of the AC output.

Using these two data sources, I plan to calculate the optimal Solarbank output directly in Home Assistant. My idea is to:
- Periodically average the power values from both smart meters.
- Based on the calculated surplus or deficit, adjust the Solarbank’s output accordingly.
- Implement logic to avoid unnecessary rapid changes, considering the delay in the Solarbank’s data updates (due to the 5-minute cloud reporting interval).

I fully understand that the integration’s role is to act as an interface for output adjustments, while Home Assistant handles the real-time data processing and automation logic.

Regarding testing: I’d be happy to test the output preset changes with my Solarbank once the new release is available to see if the behavior differs on my system. If there’s anything specific I can test—especially around API behavior or output changes—just let me know. I’d also be happy to capture API calls from the Anker app if that helps identify differences in parameter handling.

Thanks again for your support, and I’m looking forward to the next release!

thomluther, 2025-02-05:
> Regarding testing: I’d be happy to test the output preset changes with my Solarbank once the new release is available to see if the behavior differs on my system. If there’s anything specific I can [...]

If you can capture Api calls from the App, it will be helpful to know which endpoints are used when doing usage mode changes once a use time plan has been defined.
So toggling modes from/to use time plan may be play a role.
Also when you are in manual mode, what Api endpoints are used when doing only output changes in the active slot. (while there is a use time plan defined)
I think you cannot debug the payload since that is ECDH encrypted, so the parameters will remain unknown. But knowing the used endpoints would already help. This could be explored further via Api calls then.
It will also be usefull when the endpoints get_device_attrs and set_device_attrs are used, because their purpose is unclear and I have not found yet which attributes can be queried...
See [Explore usage of new device_attrs endpoints](https://github.com/thomluther/anker-solix-api/issues/107) in the Api library

thomluther, 2025-02-06:
The 2.5.0 beta is available in the main branch.
You can update via the Action install update and selecting the Anker Solix update entity. For the version you need to specify 'main' as tag for the main branch.
Note: After installation and restart, HA will continue to report a repair action since it believes the available release version is newer than the installed 'main' version. Simple ignore that until the 2.5.0 has been released.
If you have any findings with the beta, please let me know. I don't plan code changes anymore, only documentation updates prior release next week.

thomluther, 2025-02-06:
Here is an example extract of the SB2 AC fields from the Api cache.
New are the Inverter fields for Power and Limit. I saw another system that reported 800 W limit, but this one reports 0 W limit.
I wonder whether inverter fields do NOT refer to the internal inverter (since that is available also in other SB2 models and not used there), but they refer to the 'external' inverter that is plugged into the AC socket. So that 'inverter' power would be additional 'Solar' power coming from an external PV system and accumulates to the PV Channel power into the Solar Power of the device....
It looks like PV1 + PV2 + Inverter power is the Solar Power...
The would also explain why inverter limit can be different, e.g. 0 W is not limiting external inverter, but anything else is limiting it?
```
================================================================================
System         : Hammeke Base  (Site ID: ###masked###)
Type ID        : 11 (Solarbank)          Device models  : A17C2,A17X7
Cloud-Updated  : 2025-02-06 15:22:06     Valid Data     : YES (Requeries: 0)
SOC total      :   10 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :   50 W                  Battery Pwr Tot:    0 W
Output Pwr Tot :   50 W                  Home Load Tot  :    0 W
Active Mode    : Smartmeter (1)          Heating        : OFF
Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
--------------------------------------------------------------------------------
Device [A17C2] : Solarbank 2 E1600 AC    Alias          : Solarbank 2 E1600 AC
Serialnumber   : ###masked###            Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : detection               Status code    : 0
State Of Charge:   10 %                  Min SOC        :   -- %
Battery Energy :  160 Wh                 Capacity       : 1600 Wh
Exp. Batteries :    0 Pcs                AC socket      :    0 W
Solar Power    :   50 W                  Output Power   :   50 W
Solar Ch_1     :   13 W                  Solar Ch_2     :   11 W
Other Input    :    0 W                  Heating Power  :    0 W
Inverter Power :   26 W                  Grid to Battery:    0 W
Inverter Limit :    0 W                  Low Limit      :    0 W
Battery charge :    0 W                 
Battery Power  :    0 W                  Device Preset  :    0 W
Home Demand    :  317 W                  SB Home Load   :    0 W  (-)
Smart Plugs    :  --- W                  Other (Plan)   :  317 W
Schedule  (Now): 15:25:18 UTC +0100      System Preset  :    0 W
--------------------------------------------------------------------------------
Device [A17X7] : Smart Meter             Alias          : Smart Meter
Serialnumber   : AUJS020E35200795        Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Grid Status    : ok                      Status code    : 0
Grid Import    :  267 W                  Grid Export    :    0 W
================================================================================
```

distanzcheck, 2025-02-07:
The output limit can be set to 600 and 800 watts.

distanzcheck, 2025-02-07:
This is the setting for the AC appliance output 600 or 800 watts. AC Output limit,

distanzcheck, 2025-02-07:
Is it already possible to test 2.5.0? And if so, what do I have to do?
(In German would be better because then without translator).

thomluther, 2025-02-07:
> About which action Update ? Where can I find it so that I can get the 2.5.0 ?

[image]

> There is also a so-called 0Watt switch from Anker see link https://www.ankersolix.com/de/products/a17y0

This has nothing to do with AC model, that 0 W switch is only for SB1 and it does not report any values

> Other Input could be AC Solarbank charge

That is a separate new field for AC model, shown as  Grid to Battery:    0 W

distanzcheck, 2025-02-07:
Are you on Discord ? We could write in German, which would be much easier for me. I think I've managed that now with the update. I would never have thought of that.

distanzcheck, 2025-02-07:
Unfortunately, the correct charge status of the batteries and how many additional batteries are connected is reset to <1600 Wh and A additional batteries to 0 after a short time. Only a restart of HA solves the problem.

thomluther, 2025-02-07:
Actually that was an issue only at the beginning of SB2. You can invite my account lollygames2@gmail.com to your system, then I can maybe check it later.

thomluther, 2025-02-07:
You got an IP ban. Probably switched too often between accounts and devices, causing more than 10 login errors due to deleted account token...
You can try reconnect you DSL router to get a new IP, then it works again

thomluther, 2025-02-08:
Here is another example, showing also heating flag and power (first time I saw this used)
```
================================================================================
System         : Hammeke Base  (Site ID: ###masked###)
Type ID        : 11 (Solarbank)          Device models  : A17C2,A17X7
Cloud-Updated  : 2025-02-08 10:59:01     Valid Data     : YES (Requeries: 0)
SOC total      :   21 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :  586 W                  Battery Pwr Tot:  481 W
Output Pwr Tot :    0 W                  Home Load Tot  :    0 W
Active Mode    : Smartmeter (1)          Heating        : ON
Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
--------------------------------------------------------------------------------
Device [A17C2] : Solarbank 2 E1600 AC    Alias          : Solarbank 2 E1600 AC
Serialnumber   : ###masked###            Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : online                  Status code    : 1
Charge Status  : detection               Status code    : 0
State Of Charge:   21 %                  Min SOC        :   -- %
Battery Energy :  672 Wh                 Capacity       : 3200 Wh
Exp. Batteries :    1 Pcs                AC socket      :    0 W
Solar Power    :  586 W                  Output Power   :    0 W
Solar Ch_1     :   37 W                  Solar Ch_2     :  189 W
Other Input    :    0 W                  Heating Power  :  105 W
Inverter Power :  360 W                  Grid to Battery:    0 W
Inverter Limit :  800 W                  Low Limit      :    0 W
Battery charge :  481 W                 
Battery Power  :  481 W                  Device Preset  :    0 W
Home Demand    :    0 W                  SB Home Load   :    0 W  
Smart Plugs    :  --- W                  Other (Plan)   :    0 W
Schedule  (Now): 10:58:59 UTC +0100      System Preset  :    0 W
================================================================================
```
So PV1 + PV2 + external inverter power => Solar Power
That means now when the heating power field is used, the Solar Power is really every input power.
Substracting the Heating power from Solar power then gives also the correct Charge power

stelzerroland, 2025-02-09:
First of all, huge thanks for your amazing work on this integration! 🙌

I’ve been using the integration successfully, but I ran into an issue:

- At first, **everything worked fine**, including reading values and setting number.solarbank_2_e1600_ac_system_einspeisevorgabe.
- Then, I got the error:
- "Retries exceeded: (100053) Anker Api Error: Please try again in 24 hours."
- After waiting a day, I can now read all values again, but I **can no longer set number.solarbank_2_e1600_ac_system_einspeisevorgabe**.
- There is no error message, but the new value does not get applied to the Solarbank (seonsor.solarbank_2_e1600_ac_einspeisevorgabe stays unchanged in Home Assistant).

Do you have any idea what might be causing this? 

Thanks again for your great work, and I really appreciate any help! 😊

thomluther, 2025-02-09:
> Do you have any idea what might be causing this?

No, unfortunately not. I saw the same on the other AC that I can access and have no idea why the system does not apply the change.
When you open the App afterwards, it just says the previous configuration was not completed and you have to confirm the mode change again.
So I assume either a paramter is missing (but I don't know which that should be). There is also a command number which always was 17. Maybe the AC needs a different number, but I'm not going to test that on other systems.

So that is something you have to find out. Eventually there is another Api call required on top which I dont know (and which might be the reason why the App tells the configuration was not completed....
You would have to sniff the network of the App to see which Api endpoints are called when you do such changes. I have no capability or knowledge to test/debug this.

Once someone can tell me what to do via the Api, I can implement so. But so long, I guess manual preset with the AC is not working...

stelzerroland, 2025-02-11:
> So that is something you have to find out. Eventually there is another Api call required on top which I dont know (and which might be the reason why the App tells the configuration was not [...]

Which tool would you recommend to sniff the network traffic for the api endpoints? (I'm using the Anker App on Android)

thomluther, 2025-02-15:
> Which tool would you recommend to sniff the network traffic for the api endpoints? (I'm using the Anker App on Android)

Here is a description how you can likely sniff TLS traffic.
https://github.com/krikk/hoymiles-ms-a2-to-mqtt#how-to-sniff-the-app-traffic
But still this won't give you the readable Payload since that is likely EDHC encrypted after TLS decryption. But at least it should reveal the used endpoints and their order for an individual App action, like changing the output preset in manual mode...

pfeifferch, 2025-02-22:
> You got an IP ban. Probably switched too often between accounts and devices, causing more than 10 login errors due to deleted account token... You can try reconnect you DSL router to get a new IP, then it works again

Good morning. 
Unfortunately, I had the same problem a few days ago. A new IP helped. 

Unfortunately, in the meantime (about 3½ days) I can no longer log into the integration. Unfortunately, a new IP doesn't help here anymore. Unfortunately, the message is displayed too briefly and I cannot copy it. 
(Even a completely new user is blocked, which is why I assume that some Home Assistant ID is blocked)
From the message I could see that the API limit was exceeded. Unfortunately, the logs do not contain the complete API error message . 
Are there any solutions for me?

Protokollmeldungen trage ich nach sowie die komplette Fehlermeldung wenn ich es schaffe die zu kopieren.

udosw, 2025-03-14:
Hi, I recently got an Anker Solix Solabank 2 E1600 AC. I've set it up with the Anker Cloud an connected my Shelly EM3. Installed your HA integration and see 4 devices:
- 3em (Shelly)
- E1600AC (the SB2)
- System B3 (my home)
- Udo (DE) (my account)

That all went very smooth w/o any problems. The SB2 works very well with the Shelly EM3 an feeds exactly the same amount of energy currently used from the battery (of cause as long as this is below 800 W) "Nulleinspeisung".

I'd like to contribute, got the Python solix api working (though I have next to no knowledge of Python at all) and can provide system exports. Shall I upload them in this Issue or in https://github.com/thomluther/anker-solix-api/issues/115 ?

Also a place to exchange thoughts and info in German would be nice, is there any such place?

Thanks for the great work,
Udo

thomluther, 2025-03-15:
> I'd like to contribute, got the Python solix api working (though I have next to no knowledge of Python at all) and can provide system exports. Shall I upload them in this Issue or in [thomluther/anker-solix-api#115](https://github.com/thomluther/anker-solix-api/issues/115) ?

Hi @udosw 
Thanks for your offer. At this point, further exports don't help to solve issues with the AC model.
They are either related to Cloud, the firmware or the missing Api calls/parameters that are unknown. It can be also a mixture, but my impression is that most is related to firmware/cloud, since the AC does no longer send all fields to the cloud after the Admin account becomes inactive in the App.

So systems owners really need to explore various endpoints to understand whether there are new/unknown capabilities to support more than possible today. The endpoints are all documented in the apitypes.py and observations should be discussed and documented in the https://github.com/thomluther/anker-solix-api repo

For German discussion, you can use the Photovoltaik Forum thread.
https://www.photovoltaikforum.com/thread/223908-anker-solix-e1600-home-assistant-integration/

thomluther, 2025-04-10:
Full support delivered with 2.6.0.

closed 2025-04-10 by thomluther

distanzcheck, 2025-04-10:
What does full support mean? Does it all work, including the number of additional batteries?

thomluther, 2025-04-10:
> What does full support mean? Does it all work, including the number of additional batteries?

That means that all what might be possible via the Api works is integrated.
The battery count and other values in the cloud is not an integration issue, this is an AC firmware issue.