thomluther/ha-anker-solix #200: Two Anker integrations in Homeassistant - not working properly

RamsukKockoi, 2025-02-10:
### System Health details

System is healthy

### Integration version

beta 2.5.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

With to active integrations one of both regularly loosing connction and becomes unavailable. Here and there this produces unplausible values.

Sorry for beeing so special ;-)

### Reproduction steps

1. Wait for anavailability

### Debug logs

```text
Maybe not needed
```

### Diagnostics dump

[marxxx_2025-02-10_0650.zip](https://github.com/user-attachments/files/18728747/marxxx_2025-02-10_0650.zip)

[marxxx_2025-02-10_0655.zip](https://github.com/user-attachments/files/18728759/marxxx_2025-02-10_0655.zip)

thomluther, 2025-02-10:
Hi
This is a cloud issue, the more connections or Systems need to be monitored, the more queries run against the cloud and eventually in parallel. So some requests get an error code without explanation.
Nothing I can fix in the integration, since multiple configuration sessions cannot be serialized.

Your only chance is to exclude all Energy type categories in the configuration entry, since that will run many queries against the same endpoint.

thomluther, 2025-02-10:
> Is there a way, that the integration sees two systems in one account? Because that did not work for me ...

Sure, you can create multiple systems with your owner account and share any of those systems with other accounts.
So any account can see multiple systems, either as owner or as member.

RamsukKockoi, 2025-02-10:
Hey, this is what i tried first, but it did not seem, that the integration saw the second system. I will give it a try in the afternoon or tomorrow.

thomluther, 2025-02-10:
The integration will find all systems that you have under the account, but it may be that one has only member access rights, depending which account you used to create it.

closed 2025-02-10 by RamsukKockoi