thomluther/ha-anker-solix #203: Battery Pack BP 1600 vanishes after a couple of minutes in Dashboard and in the APP

tmb80c, 2025-02-11:
### System Health details

none

### Integration version

2.5.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

My system is a SB AC with 1 Battery Pack. After installing / activating the add-on (tried 2.4, 2.4.1, and today 2.5.0) with the admin account the HA dashboard shows everything correct including 1 Battery Pack. A couple of minutes later the Battery Pack counter is set to 0 and this is then also the case in the Anker App (the BP disappears from the App picture). When I disable the HA Integration and log-in again with the admin credentials in the Anker App then the Battery Pack is coming back in the App.

### Reproduction steps

1. Install the add-on
2. Activate with admin account
3. wait a couple of minutes
4. look at Solarbank AC dashboard and Anker App

### Debug logs

```text
none
```

thomluther, 2025-02-11:
@tmb80c , when you confirmed, that even the App picture is wrong when using the member account, then you have an argument to address this AC issue to Anker!

tmb80c, 2025-02-11:
thanks @thomluther

I sent two bug reports to Anker:

1) the annoying reconfiguration message when switching back to the admin user in the app

2) a screen shot from the family account after the symbol of the battery pack disappeared

distanzcheck, 2025-02-11:
> I sent two bug reports to Anker:
> 
>     1. the annoying reconfiguration message when switching back to the admin user in the app
> 
>     2. a screen shot from the family account after the symbol of the battery pack disappeared

I have a similar if not identical problem. My questions:
1. with which account am I logged into the HA Intertion?
2. with which account in the app? The main account or a family member.
3. with which account do the batteries disappear in the app, can you please upload a screenshot ?

When I log into the app with my main account, I can always see everything, including the battery charge status. In the family account, I can't access the battery data at all. I also have the other problem with the constant error messages when I log back into the app with my main account.

Translated with www.DeepL.com/Translator (free version)

tmb80c, 2025-02-11:
> I have a similar if not identical problem. My questions:
>
> 1. with which account am I logged into the HA Intertion?
> 2. with which account in the app? The main account or a family member.
> 3. with [...]

We have the same issue (your screen shots are identical to mine, except that you have more extra batteries). Please also send a bug report to Anker.

distanzcheck, 2025-02-11:
yes I have also reset it three times without success I opened the ticker with Anke and just wanted an answer

thomluther, 2025-02-12:
I see the same problems with another AC that I have admin access to.
- Battery packs disappear for member accounts if nobody uses main account in the App for a while
- Weird incompletion messages when logging back in as main account in the app

I also see problems changing the output preset in manual mode via the Api. That is nothing you can report to Anker, but it seems to be a general issue with the AC mode. Either related to the bugs above, or different because Api parameters or commands are missing for this change in the AC model...

tmb80c, 2025-02-12:
> I also see problems changing the output preset in manual mode via the Api. That is nothing you can report to Anker, but actually is does not seem to be accepted in general by the AC model. Either related to the bugs above, or different because Api parameters or commands are missing for this change in the AC model...

Indeed, I can move the slider, it stays at its position but the value next to the slider doesn't change nor in the app and therefore also not in the Anker AC. Wouldn't be surprised if there is a bug in the auhorization backend. Nothing we can do about unless the developer find it accidentally. 

Another difference between PRO and AC is that we are no longer seeing in the weekly production stats the production for each MPPT per day, at least in the latest APP 3.4.1. I wrote to ANKER and they confirmed that this is by design since they believe it is only necessary in a scenario where a user might have 4 solar panels. With HA we can workaround.

The AC is relatively new but strange that there are so many bugs/drawbacks although it should be mainly similar to the SB2 PRO. I start to regret having switched from PRO to AC (I need the AC charge to harvest some KW/h from the on the roof solar system). Let's give them some time to improve.

And finally not limited to the AC. It is disappointing that no support of the Shelly Smart Plugs has been added together with the Shelly EM Pro.

thomluther, 2025-02-12:
> Another difference between PRO and AC is that we are no longer seeing in the weekly production stats the production for each MPPT per day, at least in the latest APP 3.4.1. I wrote to ANKER and they [...]

Yes, the stats are still available for PV1 and PV2 with numbers from the cloud.
I haven't found whether there are new stats for the inverter input, since that is a new PV channel for AC and I don't know if collected separately in the cloud. The data is not provided on either PV3 or 4...since the inverter power is reported separately...

> And finally not limited to the AC. It is disappointing that no support of the Shelly Smart Plugs has been added together with the Shelly EM Pro.

I doubt this will ever come. Why should Anker support them if they have their own smart plugs...

tmb80c, 2025-02-12:
> I doubt this will ever come. Why should Anker support them if they have their own smart plugs which they want to sell...

They also have their own meter and finally provided the Shelly EM.

distanzcheck, 2025-02-12:
> > I doubt this will ever come. Why should Anker support them if they have their own smart plugs which they want to sell...
> 
> They also have their own meter and finally provided the Shelly EM.

For me, the Shelly Pro EM3 runs quite well together with the SB AC. Integrating it was also no problem.

tmb80c, 2025-02-13:
Anker support is still analyzing the two reported errors. Today I had to provide the APP and Device protocol from the APP. and therefore had to switch back to the admin user in the app. As always the annoying configuration message. Both Batteries showed up.

When I switched back to the second user in the APP I also could see both batteries. This time I waited 2-3 minutes before I re-activated the integration in HA. The Battery Pack disappeared regardless of a second / admin user logged-in from another device. So this bug is independent from Thom's work.

distanzcheck, 2025-02-13:
I also sent Anker more data, screenshots and logs today and uploaded them. Let's see what happens.

thomluther, 2025-02-15:
> I haven't found whether there are new stats for the inverter input, since that is a new PV channel for AC and I don't know if collected separately in the cloud. The data is not provided on either PV3 or 4...since the inverter power is reported separately...

I have found the new stats for energy from AC imput / inverter. It is like another PV channel query, I guest this will also be added to the App.
I will add that in the next release an extra energy entity for the AC.

tmb80c, 2025-02-17:
Recognized yesterday and today that my AC switches in the morning to "protective charge" which should be the state from 10%-12% SOC. When 12% is reached it starts to deliver to the home and charges the battery until  it reaches 100% SOC. The state doesn't toggle. The state changes when the battery is empty (SOC 10%). If I recall correctly there are more granular states with the SB Pro.

Is this a bug or an API limitation? Shall I record a separate issue?

thomluther, 2025-02-18:
Yes. The SB state ist just an abstrakt number you can find in the state attribute. You can extract the attribute via a template helper to get it as sensor and have a history of the number changes. That would have to be monitored and compared to Soc, output power and battery power. If the number does not change, it may be the 'translation' of all those conditions that I have to change in the library.
But I would need to know criteria to be used when a translated state should be changed.

thomluther, 2025-02-18:
@tmb80c please open another issue to track state description issues.
Basically when the state description is wrong, you need to get a system info response to correlate all values that translate into the description.
Protection charge is when code is 0 for detection and battery power >0 and to_home_load < set_load_power.

tmb80c, 2025-02-22:
In the meantime the Anker App updated to version 3.5.0 (iOS). The three described issues (reconfigured, battery disappears, only half power values in backup charge) still remain.

Last Monday the Anker support wrote to me, that they escalated to the next technical level and that I should hear back within 3 working days. Answer ist still pending. I will let them know that 3.5.0 doesn't fix any of the issues.

thomluther, 2025-02-23:
> In the meantime the Anker App updated to version 3.5.0 (iOS). The three described issues (reconfigured, battery disappears, only half power values in backup charge) still remain.
>
> Last Monday the [...]

I did not expect that an App update solves the problems, since the wrong data come from the cloud server and the AC firmware is responsible to provide valid data to the cloud after main account is no longer connected.

Regarding the 'wrong charging state description in the Integration', please open a new issue for it with a system info response at that time when you see the wrong state (or even better is a full system export to have the complete data set).

noseshimself, 2025-02-23:
I do not want to sound like the party pooper but the API seems to have become a moving target and Anker is either pulling data out of thin air or is mocking the users.

I'm not missing batteries (like in this report) or screws or some Watts anymore, entire components are going missing.

 I guess the bug fixes to all open bugs touching API issues should be put on hold until they get their act together.

tmb80c, 2025-02-26:
Today I received a message from the Anker support team. The issue with the disappearing battery expansion pack shall be fixed with the next app update (I'm currently on 2.5.0).

thomluther, 2025-02-27:
> Today I received a message from the Anker support team. The issue with the disappearing battery expansion pack shall be fixed with the next app update (I'm currently on 2.5.0).

This sounds odd to me. The App is just displaying what is received from the cloud. It does the same like the Integration.
To me this looks more like a problem of the device, which stopps to send updates of some values to the cloud at some point in time.
It is not only the battery number, it is also the AC Inverter limit and other unique values of the AC devices, which may not be visible in the App as member and therefore not being noticed by users.
But the Api will see it and so you will loose those values in the Integration.
Hopefully they fix the AC devices to send all data updates to the cloud in their regular 5 minute update interval.

tmb80c, 2025-02-27:
I have the same doubt, just repeating what they wrote. I guess they don’t want to overwhelm the standard user with the details of their cloud platform.

distanzcheck, 2025-02-27:
> Today I received a message from the Anker support team. The issue with the disappearing battery expansion pack shall be fixed with the next app update (I'm currently on 2.5.0).

I already have the app version 3.5.0 and the problem persists, I can't imagine that an update of the app will solve this problem.

noseshimself, 2025-02-27:
> This sounds odd to me

It is odd.

Yes, they incremented the major release number but the application itself has not changed. 

The problems I reported by mail (getting the "System"s but no "Device"s anymore inside the HA integration) have not changed either so I have the strong suspicion that they created their own API and keeping the rest of the world in the dark.

Even stranger: The E1600 Pro needed a firmware update to work with the new app version and even the batteries got a firmware update while the E1600 AC got neither. I can move a battery from Pro to AC but if it is upgraded the AC is ignoring it if moved back. *facepalm*

On the plus side: "Eigenverbrauch" with an E1600 AC is working now even with no panels connected; I could connect three full AC systems to my L1/L2/L3 if I installed three smart meters (more *facepalm*ing) which should be supported as it connecting batteries to pull cheap power from the grid is permitted now.

It all feels like dealing with the USS Ankerprise merrily sailing towards an iceberg and the captain is having a party.

noseshimself, 2025-02-27:
I have to correct myself. All the expected data about devices is received by the integration, it's just not showing up in the UI.

I'm going to send a part of the debug log by mail as it is of course showing sensitive data in the responses...

Dulcow, 2025-03-05:
> Today I received a message from the Anker support team. The issue with the disappearing battery expansion pack shall be fixed with the next app update (I'm currently on 2.5.0).

Did they say when they would release it? I'm having the same issue with the disappearing additional batteries as well.

tmb80c, 2025-03-05:
> > Today I received a message from the Anker support team. The issue with the disappearing battery expansion pack shall be fixed with the next app update (I'm currently on 3.5.0 - fixed typo).
> 
> Did they say when they would release it? I'm having the same issue with the disappearing additional batteries as well.

unfortunately not.

yesterday I read in another forum that an OTA update for the firmware 1.0.3.4 is rolling out . The user reported that a charging issue with the BP has been fixed. My SB hasn't received this firmware yet, I'm still on 1.0.3.2.

thomluther, 2025-03-05:
> unfortunately not.
>
> yesterday I read in another forum that an OTA update for the firmware 1.0.3.4 is rolling out . The user reported that a charging issue with the BP has been fixed. My SB hasn't [...]

It it more likely that an firmware update will fix this than an App update, because the values in the cloud is definitely 0 and that is provided by the device updates to the cloud. If that is fixed in the App only, then the App will start to ignore this Api field value and just masking the real issue that the AC has not only for the expansion battery field, but also many other fields in the Cloud not updated by the AC anymore after the App with admin user becomes inactive.

distanzcheck, 2025-03-06:
Got this info from Anker support today.
How do you do? 

Thank you for your patience.

We will release a new firmware within a week. Please check if the problem still exists after the firmware update.

distanzcheck, 2025-03-07:
Hallo Thomas, sind diese Unterschiede deiner Integration geschuldet ?

Der Screen Shot Automatisierung zeigt die Auswahlmöglichkeit bei einer Automatisierung an.

siehe Anhang.

 

Lg Dirk

thomluther, 2025-03-08:
> Hallo Thomas, sind diese Unterschiede deiner Integration geschuldet ? Der Screen Shot Automatisierung zeigt die Auswahlmöglichkeit bei einer Automatisierung an.

Which differences? Which attachment?

distanzcheck, 2025-03-09:
[image]
[image]

thomluther, 2025-03-09:
It may take up to 6 minutes until you see the result if the control entity change in other entities.
Also, the AC is still extremely buggy and may not apply a schedule change at all, like it does not apply a change of the manual preset power.
The sent object is correct, I have no idea what is wrong with the AC firmware since all other SB 2 models apply the schedule changes.

thomluther, 2025-03-09:
For the Screenshot, it seems that does not correctly use translations, but that is a core action for selection changes, not part of the integration itself which translates the options in the selection entity. 
For an action you always have to apply the original option keys, not the translated one presented by UI.
You can see original option keys in the developer tools for the select entity attributes.

distanzcheck, 2025-03-10:
I have just received the update to 1.0.3.4 and installed it. Let's see what happens.

distanzcheck, 2025-03-10:
So sieht es nach 9 Minuten nach dem Neustart aus.
[image]

distanzcheck, 2025-03-10:
And so after 11 minutes

[image]

tmb80c, 2025-03-10:
What ever has been fixed with version 1.0.3.4 it obviously does not resolve our bug. I haven't received the OTA update yet.

noseshimself, 2025-03-10:
What has been fixed is charging all batteries if you have an even number > 2 in your stack.

distanzcheck, 2025-03-10:
Mail in response to the question from Anker whether the problem with 1.0.3.4 has now been resolved

Hello,
The problem persists.
When I am logged in with the original account on smartphone A and logged in with the family account on smartphone B, the following happens.

A is logged in, B shows all batteries correctly in the graphic.
A logs out, B displays all batteries correctly for approx. 5-10 minutes. After 5-10 minutes, only the head battery is visible on B and no more additional batteries.

noseshimself, 2025-03-10:
The same is happening to the integration. The number of batteries is only correct for up to 10 minutes after logging in with the application... But both, Android app and integration, need access to the owner account.

thomluther, 2025-03-10:
> But both, Android app and integration, need access to the owner account.

Not sure if that makes a difference for the integration since through the cloud api there will never be a trigger for the device to push more frequent and real data updates to the cloud. They simply stay in their regular update after App owner became inactive.
And that does not push all fields for the AC device

noseshimself, 2025-03-12:
> > But both, Android app and integration, need access to the owner account.
> 
> Not sure if that makes a difference for the integration since through the cloud api there will never be a trigger for the device to push more frequent and real data updates to the cloud.

You can't change settings with an invited "family member", neither using the app nor the integration.

Right now I've set the E1600 Pro to scheduled discharging but I'm already preparing hourly changes based on remaining charge and weather forecast. It seems to be working if I'm logged in using the owner account.

And the app is a bit stupid; only the owner account can get at the stats of single batteries in a stack.

distanzcheck, 2025-04-23:
I got an e-mail from Anker support today that this problem will be fixed with the next firmware update.

thomluther, 2025-04-23:
> I got an e-mail from Anker support today that this problem will be fixed with the next firmware update.

So lets hope this will be really fixed and also the other fields that no longer being updated.

tmb80c, 2025-04-24:
The support team wrote to me today that the firmware which addresses this issue shall be released at the end of April.Gesendet mit der GMX Mail AppAm 23.04.25 um 22:44 schrieb distanzcheck
                
            
            
                
                    Von: "distanzcheck" ***@***.***>Datum: 23. April 2025An: "thomluther/ha-anker-solix" ***@***.***>Cc: "tmb80c" ***@***.***>,"Mention" ***@***.***>Betreff: Re: [thomluther/ha-anker-solix] SB2 AC Battery Pack BP 1600 vanishes after a couple of minutes in Dashboard and in the APP (Issue #203)
                
                
distanzcheck left a comment (thomluther/ha-anker-solix#203)

I got an e-mail from Anker support today that this problem will be fixed with the next firmware update.

distanzcheck, 2025-05-09:
So, I was offered the 1.0.3.8 update today, and so far all batteries are displayed. Only had a problem with the Shelly PRO 3EM at first, because the solar bank turned feed-in into consumption and vice versa. I had to remove the Shelly from the app and add it again, but after deleting and re-adding it 3 times it was correct again.
@thomluther have a look in your app with my login.

thomluther, 2025-05-09:
@distanzcheck , if you have the owner account in HA, please provide an systems export zip. I need to check whether the other owner values are also fixed. Will be useful to export when there is PV, but also another one when there is only discharge.
Especially the charging status code will hopefully not remain 0 all times...
But I guess the homepage data I can also extract with the sharing account. What was your system name? I don't know which one belongs to you and whether I can get the SW version as member (don't think so).

thomluther, 2025-05-09:
@distanzcheck 
Here is an actual monitor output, maybe it's your system:
```text
================================================================================
System (+00:00): Home  (Site ID: <masked>)
Type ID        : 11 (Solarbank)          Device models  : SHEMP3,A17C2
Energy Time    : 2025-05-09 17:44:08     Last Check     : 2025-05-09 17:49:51
Cloud-Updated  : 2025-05-09 17:46:09     Valid Data     : YES (Requeries: 0)
SOC total      :   92 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :    0 W                  Battery Pwr Tot:  150 W
Output Pwr Tot :    0 W                  Home Load Tot  :    0 W
Active Mode    : Smartmeter (1)          Heating        : OFF
Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
--------------------------------------------------------------------------------
Device [A17C2] : Solarbank 2 E1600 AC    Alias          : AC
Serialnumber   : AP## masked ##04        Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : Online                  Status code    : 1
Charge Status  : Detection               Status code    : 0
State Of Charge:   92 %                  Min SOC        :   -- %
Battery Energy : 8832 Wh                 Capacity       : 9600 Wh
Exp. Batteries :    5 Pcs                AC socket      :    0 W
Solar Power    :    0 W                  Output Power   :    0 W
Solar Ch_1     :    0 W                  Solar Ch_2     :    0 W
Other Input    :    0 W                  Heating Power  :    0 W
Inverter Power :    0 W                  Grid to Battery:    0 W
Inverter Limit :    0 W                  Low Limit      :    0 W
Battery charge :  150 W                 
Battery Power  :  150 W                  Device Preset  :    0 W
Home Demand    :    0 W                  SB Home Load   :    0 W  
Smart Plugs    :  --- W                  Other (Plan)   :    0 W
Schedule  (Now): 17:49:51 UTC +0200      System Preset  :    0 W
--------------------------------------------------------------------------------
Device [SHEMP3]: Shelly Pro 3EM          Alias          : Shelly Pro Home
Serialnumber   : 13# masked #84          Admin          : NO
Wifi SSID      :                        
Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
Cloud Status   : Online                  Status code    : 1
Grid Status    : Ok                      Status code    : 0
Grid Import    :    0 W                  Grid Export    :   23 W
================================================================================
```

While the batteries count is now correct, other value combinations still makes no sense...
Charging is 150W, but where does it come from? No Grid to battery and no PV input, so power from nowhere???

distanzcheck, 2025-05-09:
> [@distanzcheck](https://github.com/distanzcheck) Here is an actual monitor output, maybe it's your system:
> 
> ```
> ================================================================================
> System (+00:00): Home  (Site ID: <masked>)
> Type ID        : 11 (Solarbank)          Device models  : SHEMP3,A17C2
> Energy Time    : 2025-05-09 17:44:08     Last Check     : 2025-05-09 17:49:51
> Cloud-Updated  : 2025-05-09 17:46:09     Valid Data     : YES (Requeries: 0)
> SOC total      :   92 %                  Dischrg Pwr Tot:    0 W
> Solar  Pwr Tot :    0 W                  Battery Pwr Tot:  150 W
> Output Pwr Tot :    0 W                  Home Load Tot  :    0 W
> Active Mode    : Smartmeter (1)          Heating        : OFF
> Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
> --------------------------------------------------------------------------------
> Device [A17C2] : Solarbank 2 E1600 AC    Alias          : AC
> Serialnumber   : AP## masked ##04        Admin          : NO
> Wifi SSID      :                        
> Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
> SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
> Cloud Status   : Online                  Status code    : 1
> Charge Status  : Detection               Status code    : 0
> State Of Charge:   92 %                  Min SOC        :   -- %
> Battery Energy : 8832 Wh                 Capacity       : 9600 Wh
> Exp. Batteries :    5 Pcs                AC socket      :    0 W
> Solar Power    :    0 W                  Output Power   :    0 W
> Solar Ch_1     :    0 W                  Solar Ch_2     :    0 W
> Other Input    :    0 W                  Heating Power  :    0 W
> Inverter Power :    0 W                  Grid to Battery:    0 W
> Inverter Limit :    0 W                  Low Limit      :    0 W
> Battery charge :  150 W                 
> Battery Power  :  150 W                  Device Preset  :    0 W
> Home Demand    :    0 W                  SB Home Load   :    0 W  
> Smart Plugs    :  --- W                  Other (Plan)   :    0 W
> Schedule  (Now): 17:49:51 UTC +0200      System Preset  :    0 W
> --------------------------------------------------------------------------------
> Device [SHEMP3]: Shelly Pro 3EM          Alias          : Shelly Pro Home
> Serialnumber   : 13# masked #84          Admin          : NO
> Wifi SSID      :                        
> Wifi state     : Unknown                 Signal         :  --- % (--- dBm)
> SW Version     : Unknown (Unknown)       Auto-Upgrade   : Unknown (OTA Unknown)
> Cloud Status   : Online                  Status code    : 1
> Grid Status    : Ok                      Status code    : 0
> Grid Import    :    0 W                  Grid Export    :   23 W
> ================================================================================
> ```
> 
> While the batteries count is now correct, other value combinations still makes no sense... Charging is 150W, but where does it come from? No Grid to battery and no PV input, so power from nowhere??? I don't think Anker will get those values right for AC any time soon....

How can I generate the output you have shown me?

thomluther, 2025-05-09:
But they only fixed the battery count.
Why is Grid to battery 0W when it is charging from grid?
Why is charging status always 0 even if it is charging which is 3 normaly?
And by the way, also in the App there is no power flow into the battery while charging from grid....something that could also be reported to Anker.
Maybe the did it better for Sb3...we will see...

distanzcheck, 2025-05-10:
For me, the feed-in is displayed in the app in the family account as well as in the main account, see video.

https://github.com/user-attachments/assets/f132da43-d35f-478b-a513-6652a2b0e7e2

distanzcheck, 2025-05-10:
So ist mein Solar mit externen WR angeschlossen.

[image]

distanzcheck, 2025-05-10:
Aktuell eindeutig bei mir geladen.

[image]

thomluther, 2025-05-10:
> For me, the feed-in is displayed in the app in the family account as well as in the main account, see video.

Feed into the house is dispayed, yes. That is working for all SB2 models. But AC charge into battery is not displayed anywhere in the home screen. There is no flow into the battery, nor is there a power value.
Only when you open the solarbank details you can see it is charging and you see the AC charge power. But it is very difficult to understand, whether this charge is from PV or from AC, or whether it is combined...

thomluther, 2025-05-10:
> Aktuell wird bei mir geladen.

Laut den Werten im Screenshot wird Entladen mit 4 W

> How do I make a system export zip?

[Export systems action](https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action)

thomluther, 2025-05-10:
> How can I generate the output you have shown me?

That is from the monitor.py tool in the [Api library](https://github.com/thomluther/anker-solix-api). You need a python runtime and install the library to use it. You cannot use it in parallel to the HA integration since it uses a different client connection and will kick out HA client if you use the same account for login.

distanzcheck, 2025-05-10:
But then I wouldn't be able to see any more, as you would with my login data, right?

thomluther, 2025-05-10:
> But then I wouldn't be able to see any more, as you would with my login data, right?

It depends which account you use for login. Owner will make more queries, so you see more info. But the monitor ist more for monitoring, api testing and debugging...

thomluther, 2025-05-15:
Since the battery count is fixed with AC firmware 1.0.3.8, I'm closing this issue.
Other AC related field value issues can be discussed and tracked in issue https://github.com/thomluther/ha-anker-solix/issues/211

closed 2025-05-15 by thomluther