thomluther/ha-anker-solix #207: Debugging Request: Anker Solix

noseshimself, 2025-02-20:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I'm completely frustrated and confused by the data provided by the Anker Solix integration 8-).

The system should know (and the mobile app shows) four additional batteries but most of the time:

[image]

Right now I'm happily charging the batteries:

[image]

But if I am pulling power from the grid... Where is ist?

[image]

"Akkuleistung" shows the 1200 W being pulled in but I would expect it to show up as "Netzaufladung" (which it sometimes did in the past). "Eingang" is following with a lag of at least one hour and a granularity of hours... And shouldn't "Einspeisevorgabe" be considered a "Control" instead of a "Sensor"? Or how am I ever getting Energy out of it again?

### Describe the solution you'd like

To me this is looking like a mix of insufficient documentation by the vendor and  incorrect data being retrieved (but how does the mobile app get the correct values?).

Somewhere inside the code there should be a lot more plausibility checks (which means that the semantics of the data points have to be known and implicitly used for this purpose) and as long as things don't add up debug data should be returned to the maintainers. The data collected this way should be used for a qualified bug report to Anker or necessary modifications to the integration. I'm also ready to completely hand over control of the device

### Describe alternatives you've considered

Burning the device.

### Additional context

I'm really believing that the firmware is equally unstable. Example:

[image]

I checked the temperatures (there is a sensor right besides it) and it was never below 14° during the week (but I could see the temperature rising at 16:40 so this was not just incorrect data.

thomluther, 2025-02-20:
The AC still has couple of firmware Bugs that should all be reported to Anker, but you cannot use any data of the Api since that is not official and not supported.
Reporting issues that you see in the Integration are also visible in the App typically when using a shared account. Eg the additional batteries will disappear from the homepage picture after a while when Main Account is no longer connected via App.
You can always pull the system info data via the Solix Action in HA. This is described in the Repo documents.
Why is the heating power or AC charging power not continously reported? Probably for same reason as Akku pack count being lost...
The integration can only report what the cloud is responding...
The interpretation of various fields is up on us once their plausibility is understood.

thomluther, 2025-02-20:
> ### Checklist
> 
>     * [x]  I have filled out the template to the best of my ability.[x]  This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).[x]  This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).
> 
> 
> ### Is your feature request related to a problem? Please describe.
> 
> I'm completely frustrated and confused by the data provided by the Anker Solix integration 8-).
> 
> The system should know (and the mobile app shows) four additional batteries but most of the time:
> 
> [image]
> 
> And shouldn't "Einspeisevorgabe" be considered a "Control" instead of a "Sensor"? Or how am I ever getting Energy out of it again?
This sensor is showing what is the active setting according to the device settings and schedule. The System Einspeisevorgabe is the control entity to modify the output in the current slot of the active schedule, but this is only considered in Manual usage mode.

> To me this is looking like a mix of insufficient documentation by the vendor and incorrect data being retrieved (but how does the mobile app get the correct values?).
The App also gets data from the MQTT server which is not supported by the integration (no documentation for it as you say).
So the integration can only get data from the Api cloud server and only if someone finds the proper endpoints, queries and parameters.
And nobody has knowledge how and when the response fields are supposed to be used by the devices.
There are LOTS  of inconsistencies for various responses, so you can never say whether it's a bug or not or something for future usage...

Dulcow, 2025-03-05:
Hi there,

I just received my Solarbank 2 AC, updated to the last firmware available (v1.0.3.2) and anker-solix 2.5.0 integration. I cannot find the fields with kWh for the in/out energy from/to battery. Because of this, I cannot integrate it into HA Energy Dashboard.

Any idea what's missing here? I can open another bug but it looks like @noseshimself does not have this issue. It's odd :-/

[image]

thomluther, 2025-03-05:
> Nevermind, I realised that some stats were disabled by default. With "Solarbank Energy Stats" enabled, it works now.

I definitely do NOT recommend to use those energy stats for dashboard integration for various reasons.
Due to new endpoint limitations enforced by the Anker cloud, it may also be that you get request errors since only 10 requests per endpoint per minute per IP address are allowed. So if you watch charts in the App in parallel, this reduces the energy requests that the integration will run and end in errors.

[Here](https://github.com/thomluther/ha-anker-solix/discussions/16) is described how you can integrate the solarbank into the dashboard.

closed 2025-03-10 by thomluther

noseshimself, 2025-03-10:
You might revisit this thing with every firmware update.

Remember: I erased my setup and started a new one -- and did not find the devices belonging to sites. Only after adding some more topics (but not all) (and don't ask me which) the SBs themselves were created again and all values of the sites became available. And this was not depending on the model, it affected AC and Pro just the same.

I don't have a test instance right now and I'm a bit afraid of starting with a fresh setup right now until all serious bugs have left the firmware building but if you can't replicate the problem with an E1600-not-AC I'll try setting up a docker container.

thomluther, 2025-03-10:
This was likely due to Api errors you got in the backend. Don't know if you verified that in your HA logs.
Verson 2.5.5 now has request throttling to prevent load issues or too many Api requests for a specific endpoint.
It also tries to request once if the Api returns a busy error.

noseshimself, 2025-03-12:
Are you ever retrying to enumerate a full system inventory on a regular schedule? At what point would you notice a new smart meter or intelligent plug showing up after it has been added? I just checked the data showing up when all battery packs are announced and what is gone when the number of packs is 0...

thomluther, 2025-03-13:
Not automatically, you have to reload the configuration entry to restart a new setup of entities.
If Api no longer provides data for a created entity, it becomes unavailable.
If an entity was created and a reload does not find the required data anymore to fill the cache, it is reported as no longer provided by the integration. Then you can cleanup it manually, or force a new registration of all entities by confirming a reconfiguration of your account credentials