thomluther/ha-anker-solix #209: Improve toleration of multiple hub entries

thomluther, 2025-02-23:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Users reported issues once more than 1 hub entry is being used.
Standard Api queries may return an error and entities become unvailable for some time until next refresh interval will receive valid respnses again without Api errors.

### Describe the solution you'd like

Avoid unavailable entities when more than 1 Anker Solix configuration entry is being used.

### Describe alternatives you've considered

Add all systems and devices under a single Anker Solix account, which will require only one configuration entry and all queries are being serialized.

### Additional context

See discussion Adding a second Hub in HA results in frequent disconnections to Anker API #129

thomluther, 2025-02-23:
The root cause of this issue is that queries of multiple accounts cannot be serialized. Each account has its own request delays and refresh intervals, that may also have different duration depending on the account type, number of systems and devices and number of queries to be run for each refresh type.

Actually the allowed requests per minute are unknown, but Anker seems to limit them and record the IPs with exceeding requests, so temporary IP bans may be possible like for too many failed authentication requests (about 10 per day).

Individual Api endpoints are limited to about 25 queries per minute, but this is mostly critical for the energy analytics endpoint since multiple queries with various periods and energy types are required to fetch all daily energy statistics.

Recommendations to avoid these Api request issues:

1. Use just a single account in the HA integration. Every System can be shared with that account, so you can get at least monitoring access for the HA integration. All queries per account are serialized with the configured requests delays and interval.
2. Exclude energy statistic categories from the configuration options. Especially if you see errors upon energy statistic endpoints. A single device typically does not cause issues here, but if you have 3 or more Solarbank devices in your account, you can exceed this 25 query limit per minute already with a single account. Each SB2 typically queries 4 energy types for today for previous day initially. Previous day is queried only once and when new day starts to limit the requests. But still 8 energy queries per SB2 when configuration is loaded are being performed.
3. Increase the request delays or update interval may help.

thomluther, 2025-02-23:
When multiple accounts are driving Api queries in parallel from same IP address, there may be request errors from the cloud server.
This cannot be avoided or serialized for multiple configuration entries.

Following may help to better tolerate multiple configuration entries:

1. Consider this specific 'busy' error Code 21105 from Api as temporary error and attempt 1 query retry after random delay of 2-5 seconds. See [https://github.com/thomluther/anker-solix-api/issues/168](https://github.com/thomluther/anker-solix-api/issues/168)
2. Investigate how multiple data coordinators for Anker Solix configuration entries can be staggered to avoid the drive the system refresh or the device details refresh at same minute. Especially the device details refresh cycle or startup may call multiple requests against the energy statistics endpoint and result in errors. Eventually each coordinator needs to get a busy flag for the device details refresh run and others should be delayed for at least 1 minute.
3. An Api export action should generally block and delay normal queries for the account since Api cache is not usable at that time due to randomized sensitive data, and because export queries should have priority to ensure complete export data. This will also delay actions/changes via the control entities or automations.

thomluther, 2025-02-24:
What do you mean with removing the serialization? Serialization of the Api requests is essential to avoid request errors.
Once you start configuring another account you can have more parallel requests which cannot be centrally serialized and therefore may cause request errors.
Serialization does not reduce the number of requests, but stretches them over longer time frame...

pfeifferch, 2025-02-24:
I found the error. Too many API queries because I accidentally deleted the filters for "excluded entities"

thomluther, 2025-03-02:
I think with a central throttle capability, most of the seen Api error issues can be addressed.
The export systems action typically does not interfere with the Api cache, since the cache is just borrowed for very brief time to generate the randomized cache data as files in the export. When the cache is not usable, any coordinator update will be skipped already. However, if there are other control actions that would rely on cache data (such as schedule control entity changes), they may still use invalid data from cache. A new helper function to delay changes until cache becomes valid again would have to be implemented to avoid this less likely data mismatch.

With implemented 'Busy Error' retry and a central endpoint throttle, there may still be following errors logged, but the retry typically should not run into another error in most of the cases, although this may happen for similar retry delays or multiple parallel accounts configured.
```
2025-03-02 21:32:02.934 ERROR (MainThread) [custom_components.anker_solix] Api Busy Error: (21105) Anker Api Error: Failed to request.
2025-03-02 21:32:02.935 ERROR (MainThread) [custom_components.anker_solix] Response Text: {"code":21105,"msg":"Failed to request.","trace_id":"6a8ca1d958acb42c20c3e05fa38d7685"}
2025-03-02 21:32:02.935 WARNING (MainThread) [custom_components.anker_solix] Server busy, retrying request after delay of 2 seconds.
2025-03-02 21:32:06.699 WARNING (MainThread) [custom_components.anker_solix] Throttling next request for 58.2 seconds to maintain request limit of 10 for endpoint power_service/v1/site/energy_analysis
2025-03-02 21:33:06.703 ERROR (MainThread) [custom_components.anker_solix] Api Busy Error: (21105) Anker Api Error: Failed to request.
2025-03-02 21:33:06.703 ERROR (MainThread) [custom_components.anker_solix] Response Text: {"code":21105,"msg":"Failed to request.","trace_id":"dd67da08efa0241db9af85f9a99c6bc1"}
2025-03-02 21:33:06.703 WARNING (MainThread) [custom_components.anker_solix] Server busy, retrying request after delay of 5 seconds.
2025-03-02 21:33:14.520 WARNING (MainThread) [custom_components.anker_solix] Throttling next request for 55.2 seconds to maintain request limit of 10 for endpoint power_service/v1/site/energy_analysis
2025-03-02 21:34:11.351 ERROR (MainThread) [custom_components.anker_solix] Api Busy Error: (21105) Anker Api Error: Failed to request.
2025-03-02 21:34:11.351 ERROR (MainThread) [custom_components.anker_solix] Response Text: {"code":21105,"msg":"Failed to request.","trace_id":"ba3e23d88517a7a126342396535e2533"}
2025-03-02 21:34:11.351 WARNING (MainThread) [custom_components.anker_solix] Server busy, retrying request after delay of 3 seconds.
```

This should avoid that entities become unvailable until next refresh cycle for most incidents.

**HOWEVER:**
The disadvantage of the throttle is the huge delay of several minutes potentially. To avoid an initial energy data pull will delay the (re)load of the configuration entry upon restarts or major option changes, the first energy data pull will be deferred to the next refresh cycle (typically after 60 seconds).
This has the consequence, that all energy related entities will have delayed existance. They will not show up until the first energy data pull is completed including throttle loops. So it could last **several minutes** after the configuration is (re)leaded or changed until energy entities are added and usable.

thomluther, 2025-03-08:
A Throttle was added to a single client session. This avoids that a single client will run into the error, but a client cannot know, which requests have been run recently by other clients or the App. So the error can still occur, especially for energy stat queries.
The endpoint limit can be adjusted in the configuration options. 0 will disable the throttling, but this is not recommended since it will increase risk to see the too many requests error and entities become temporarily unavailable.

To avoid startup delays due to 1 or more minute throttle loops, the initial energy data poll will be deferred to the next update cycle.
That causes that the energy entities will now be created with a delay after 1 or more minutes, depending how many throttle loops are required on next data update cycle.

Furthermore will the data coordinator now stagger data polls of multiple configuration entries:
- update cycle may be delayed by 10 seconds to allow another cycle to finish
- device details refresh delayed by 2 or more minutes to allow another details refresh to finish with 1-2 throttle loops.

thomluther, 2025-03-10:
Delivered with 2.5.5