thomluther/ha-anker-solix #211: State remains in protective charge

tmb80c, 2025-02-25:
### System Health details

none

### Integration version

2.5.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

SB only toggles between two states (sleep during the night) and protective charge between 10%- 100%.

Protective charge should be only when all power is used for battery charge independent if power to home is greater 0. This is usually between 10% -11% SOC and if battery temperature is low.

[image]

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
none
```

tmb80c, 2025-02-25:
> thomluther
>  left a comment
> (thomluther/ha-anker-solix#211)
> @tmb80c
> Please send me a system export in that condition so I can use it to debug and test required changes in the code.
> Export systems action

I executed the action with cache enabled.

sensor.system_obererft_ac_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: 8740c5ad-1106-4fdc-a6c5-50fa8bc088f5
      site_name: Obererft AC
      site_img: ""
      device_type_list:
        - 3
      ms_type: 1
      power_site_type: 11
      is_allow_delete: true
      support_device_models:
        - A17C2
        - A17X7
        - A17X8
        - SHEM3
        - SHEMP3
      current_site_device_models:
        - A17C2
    site_type: solarbank
    site_admin: true
    data_valid: true
    requeries: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "101.84"
        unit: kWh
      - type: "2"
        total: "101.54"
        unit: kg
      - type: "3"
        total: "32.32"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C2
          device_sn: APCT780E39400278
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/11/11/iot-admin/NycsbknWijoZcv1b/output.lin.png
          battery_power: "62"
          bind_site_status: ""
          charging_power: "724"
          power_unit: W
          charging_status: "0"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "824"
          output_power: "100"
          create_time: 1735293158
          set_load_power: ""
          sub_package_num: 0
          output_cutoff_data: 0
          is_display: true
          bat_charge_power: "724"
          alias_name: Solarbank 2 E1600 AC
          parallel_home_load: "100"
          current_home_load: "100"
      total_charging_power: "724"
      power_unit: W
      charging_status: "0"
      total_battery_power: "0.62"
      updated_time: "2025-02-25 11:37:06"
      total_photovoltaic_power: "824"
      total_output_power: "100.00"
      display_set_power: false
      battery_discharge_power: "0"
      ac_power: "0"
      to_home_load: "0"
      is_display_data: true
      solar_power_1: "281"
      solar_power_2: "249"
      solar_power_3: "0"
      solar_power_4: "0"
      other_input_power: "0"
      micro_inverter_power: "294"
      micro_inverter_power_limit: "0"
      micro_inverter_low_power_limit: "0"
      grid_to_battery_power: "0"
      pei_heating_power: "0"
      backup_info:
        start_time: 0
        end_time: 0
        full_time: 0
    retain_load: 100W
    scene_mode: 3
    home_load_power: "100"
    updated_time: 01-01-0001 00:00:00
    power_site_type: 11
    site_id: 8740c5ad-1106-4fdc-a6c5-50fa8bc088f5
    powerpanel_list: []
    grid_info:
      grid_list: []
      photovoltaic_to_grid_power: "0"
      grid_to_home_power: "0"
      grid_status: 0
      has_history_data: true
    is_downgrade: false
    error_code: 0
    smart_plug_info: null
    feature_switch:
      0w_feed: true
      exceed_power: false
      heating: false
      manual_backup_mode: true
      micro_inverter_output_exceed: false
      micro_inverter_power_exceed: false
      offgrid_with_micro_inverter_alert: false
      peak_valley_mode: true
      shelly_meter: true
      smart_plug: false
    other_loads_power: "100"
    priority_discharge_switch: 0
    display_priority_discharge_tips: 0
    priority_discharge_upgrade_devices: ""
    style_id: 0
    is_show_priority_discharge: 0
    site_details:
      price: 0.4
      site_co2: 0
      site_price_unit: €
      price_type: fixed
      current_mode: 3
    energy_details:
      today:
        date: "2025-02-25"
        battery_discharge: "0.00"
        ac_socket: "0.00"
        battery_to_home: "0.00"
        grid_to_battery: "0.00"
        home_usage: "0.40"
        grid_to_home: "0.00"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.91"
        solar_production_pv2: "0.61"
        solar_production: "2.19"
        battery_charge: "1.79"
        solar_to_grid: "0.00"
        battery_percentage: "0.82"
        solar_percentage: "0.18"
        other_percentage: "0"
      last_period:
        date: "2025-02-24"
        battery_discharge: "1.45"
        ac_socket: "0.00"
        battery_to_home: "1.45"
        grid_to_battery: "0.20"
        home_usage: "2.03"
        grid_to_home: "0.00"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.14"
        solar_production_pv2: "0.12"
        solar_production: "0.78"
        battery_charge: "0.40"
        solar_to_grid: "0.00"
        battery_percentage: "0.51"
        solar_percentage: "0.49"
        other_percentage: "0"

thomluther, 2025-02-25:
You can simply copy the returned file name including path and append that to the browser URL of your home assistant.

Export systems action

Starting with version 2.1.2, a new action was added to simplify an anonymized export of known Api information available for the configured account. The Api responses will be saved in JSON files and the folder will be zipped in your Home Assistant /configuration folder under www/community/anker_solix/exports. The action response field export_filename will provide the zipped filename url path as response to the action. This allows easy download from your HA instance through your browser when navigating to that url path. Optionally you can download the zip file via Add Ons that provide file system access.

Notes:

- This action will execute a couple of Api queries and run about 10 or more seconds. The UI button will only show green once the action is finished
- An error will be raised if the action is triggered while there is still a previous action active
- There may be logged warnings and errors for queries that are not allowed or possible for the existing account. The resulting log notifications for the anker_solix integration can be cleared afterwards
- The url path that is returned in the response needs to be added to your HA server hostname url for direct download of the zipped file (the www filesystem folder is accessible as /local in the url navigation path as given in the response).

tmb80c, 2025-02-25:
Adding to the url of my HA is what I thought is the solution but it didn’t work.

In the meantime I browsed through the HA filesystem in the virtual machine with a terminal session and found the file. I also installed the downloader add-on but this seems to be a tool for downloading a file into HA and not from HA to the Macbook.

I give it a another try when I have time again.

tmb80c, 2025-02-26:
The url path that is returned in the response needs to be added to your HA server hostname url for direct download of the zipped file (the www filesystem folder is accessible as /local in the url navigation path as given in the response).
I tried this multiple times with Safari and Firefox
 
http://xxx:8123/local/community/anker_solix/exports/thoxxx_2025-02-25_1516.zip

and always receive Error 404 file not found although I can see the file in /config/www/community/anker_solix/exports 
when I browse to this location in a HA terminal window.

No idea how I can provide you with this report.

> Am 25.02.2025 um 15:11 schrieb Thomas Luther ***@***.***>:
> 
> The url path that is returned in the response needs to be added to your HA server hostname url for direct download of the zipped file (the www filesystem folder is accessible as /local in the url navigation path as given in the response).

tmb80c, 2025-02-26:
finally I found a way to download the file: [thoxxx.zip](https://github.com/user-attachments/files/18987606/thoxxx.zip)

thomluther, 2025-02-26:
> and always receive Error 404 file not found although I can see the file in /config/www/community/anker_solix/exports when I browse to this location in a HA terminal window.

Could be that you have to restart Home Assistant if the www folder was just created by the export because it was not mapped to /local

tmb80c, 2025-02-26:
yes, the file can be downloaded after a restart under development tools, may be this can be added to the description.

thanks

thomluther, 2025-02-26:
Thanks for confirming, will add this to the description.

I looked at your export. Did you pull it when you had the wrong state?
When I parse it with the solarbank monitor, I don't see a wrong state description, only detection as expected:
```
================================================================================
System         : Obererft AC  (Site ID: fa2de73d-2fa8-5de0-842c-caa81f548e3b)
Type ID        : 11 (Solarbank)          Device models  : A17C2
Cloud-Updated  : 2025-02-25 15:14:07     Valid Data     : YES (Requeries: 0)
SOC total      :  100 %                  Dischrg Pwr Tot:    0 W
Solar  Pwr Tot :  187 W                  Battery Pwr Tot:    0 W
Output Pwr Tot :  187 W                  Home Load Tot  :    0 W
Active Mode    : Manual (3)              Heating        : OFF
Offgrid Alert  : OFF                     Inv. Pwr Exceed: OFF
--------------------------------------------------------------------------------
Device [A17C2] : Solarbank 2 E1600 AC    Alias          : Solarbank 2 E1600 AC
Serialnumber   : MSE36FN164DHMDB0        Admin          : YES
Wifi SSID      : wifi-network-1         
Wifi state     : Online                  Signal         :  100 % (-47 dBm)
SW Version     : v1.0.3.2 (Latest)       Auto-Upgrade   : Enabled (OTA v1.0.3.2)
 -Component    : A17C2_esp32 (Latest)     -Version      : v0.1.7.7
 -Component    : A17C2_mcu (Latest)       -Version      : v1.0.3.2
 -Component    : A17C2_100Ah (Latest)     -Version      : v0.2.2.9
Cloud Status   : online                  Status code    : 1
Charge Status  : detection               Status code    : 0
State Of Charge:  100 %                  Min SOC        :   10 %
Battery Energy : 1600 Wh                 Capacity       : 1600 Wh
Exp. Batteries :    0 Pcs                AC socket      :    0 W
Solar Power    :  187 W                  Output Power   :  187 W
Solar Ch_1     :   52 W                  Solar Ch_2     :   40 W
Other Input    :    0 W                  Heating Power  :    0 W
Inverter Power :   95 W                  Grid to Battery:    0 W
Inverter Limit :    0 W                  Low Limit      :    0 W
Battery charge :    0 W                 
Battery Power  :    0 W                  Device Preset  :  100 W
Home Demand    :  187 W                  SB Home Load   :    0 W  (-)
Smart Plugs    :  --- W                  Other (Plan)   :  187 W
Schedule  (Now): 23:42:14 UTC +0100      System Preset  :  100 W
Usage Mode: Manual (3)       Def. Preset:  200 W
ID Start End   Output Weekdays                          <== custom_rate_plan
 0 00:00 09:00  200 W Sun,Mon,Tue,Wed,Thu,Fri,Sat
 0 09:00 18:00  100 W Sun,Mon,Tue,Wed,Thu,Fri,Sat
 0 18:00 24:00  200 W Sun,Mon,Tue,Wed,Thu,Fri,Sat
Backup Start         Backup End           Switch: OFF   <== manual_backup
2025-02-25 11:46     2025-02-25 14:46    
================================================================================

```

thomluther, 2025-02-26:
Nevermind, I think I could recreate the problem, it shows after one refresh with your mocked data from the previous system info. I will debug it to find the root cause.

tmb80c, 2025-02-27:
Happy to provide you with more reports at different states when needed.

tmb80c, 2025-02-27:
And keep in mind that the report might not show the BP1600 (and associated power values) correctly although it is connected and working.

thomluther, 2025-02-27:
OK, yes please send me another export when you see the charging state issue again, to make sure to have a complete data set I can use for proper debugging. I'm not sure if my mocked data are really good for the debug.

tmb80c, 2025-02-27:
Here is another report. 

[thoxxx_2025-02-27_0913.zip](https://github.com/user-attachments/files/19005086/thoxxx_2025-02-27_0913.zip)

thomluther, 2025-02-27:
Did you pull it while showing wrong state? I can see only detection, which seems to be OK for the given value constellation and schedule.

thomluther, 2025-02-27:
When I look at your previous system info, there is following constellation that is odd and leads to protective charge state evaluation:
```
          parallel_home_load: "100"
          current_home_load: "100"
      total_charging_power: "724"
      charging_status: "0"
      total_photovoltaic_power: "824"
      total_output_power: "100.00"
      to_home_load: "0"
      solar_power_1: "281"
      solar_power_2: "249"
      solar_power_3: "0"
      solar_power_4: "0"
      micro_inverter_power: "294"
```

All values indicate 100 W DC output power, but to_home_load is 0 W, which is being used as SB2 value to verify how much is actually sent to the house.
If power sent to the house is < than the house demand / Preset value and there is charge power, than this is considered as protective charge because the device does not fulfill the demand/preset.

So the question is why `to_home_load: "0"` in such cases?
You may verify that by pulling a system info action when you see the state again...
Eventually that is the same 0 value issue as for some other fields that the AC devices still has....

thomluther, 2025-02-27:
Well, that may also mean that even the charging_status field is not updated but stays 0?
0 is typically used as code for the detection mode.
However, that code is typically just used when using an automatic mode, smart meter etc.
So a code status of 0 may be wrong in general with the manual mode setting of your AC.
That means not only the to_home_load field is not reliable, but also the charging_code field is not reliable.
That would explain why you so only those 2 states, because those are the only onces that are 'translated' from a 0 charging status code

thomluther, 2025-03-02:
FYI, I was looking at other examples of AC output and it seems the AC has still a general issue with updating some values in the cloud.
For example following screenshot of an AC while discharging:

[image]

When no owner account is using the App, those fields will go to 0. That is definitely a firmware bug for the AC model, as for other SB2 models this works correctly :

- AC Home must typically show a value when battery is discharging, but it is always 0 => This should also be visible in the App as member account, since that is the value sent to the home. This should be addressed as ticket if provable in screen shots.
- The mode code is always 0, which can only be translated to detection or protective_charge. 
- The battery packs is 0 => Is visible by disappearing batteries on home screen of shared member in App => Address to Anker support
- The inverter limit goes 0 => Not visible on App home screen of shared member
- Heating power goes 0 => Not visible on App home screen of shared member
- Grid Charge power goes 0 => Actually I never saw something different. How does that show in the App Home screen as owner and as member? => Address via Anker Support if there is a difference as member.

tmb80c, 2025-03-03:
> * AC Home must typically show a value when battery is discharging, but it is always 0 => This should also be visible in the App as member account, since that is the value sent to the home. This should be addressed as ticket if provable in screen shots.

In HA this is shown 0 when using the member account. In the APP (3.5.0, member account) I can see the value sent to home (once the SOC is around 11% after a cold night).

> * The battery packs is 0 => Is visible by disappearing batteries on home screen of shared member in App => Address to Anker support, see [Battery Pack BP 1600 vanishes after a couple of minutes in Dashboard and in the APP #203](https://github.com/thomluther/ha-anker-solix/issues/203)

in discussion with support waiting for the fix

> * Grid Charge power goes 0 => Actually I never saw something different. How does that show in the App Home screen as owner and as member? => Address via Anker Support if there is a difference as member.

Correct this field is disappearing in the app when using the member account (ticket sent to support waiting for the fix).

thomluther, 2025-03-03:
> In HA this is shown 0 when using the member account. In the APP (3.5.0, member account) I can see the value sent to home (once the SOC is around 11% after a cold night).

You should always see that value whenever the battery is discharging, because that all must go to the house at least, plus PV that is bypassed.
Also PV that is not charged need to go to the house. So this value should always be correct, even if not using the owner account in the App for a couple of minutes. I assume it goes to 0 as well once the other values go to 0 (e.g. once the extra batteries disappear).
The member account should not have capabilities to get data from the MQTT server to my knowledge, so it has only the data from the cloud api.

thomluther, 2025-09-21:
The charging status is now assumed based on value constellations. This is all what still can be done for the AC model while the value reporting is not fixed in the cloud. Closing this issue now.

closed 2025-09-21 by thomluther