thomluther/ha-anker-solix #212: Feed-in specification

Koxxi89, 2025-02-26:
### System Health details

## System Information

version | core-2025.2.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.73-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4993
Installed Version | 2.0.5
Stage | running
Available Repositories | 1558
Downloaded Repositories | 44

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.1
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 9.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.18.0), Terminal & SSH (9.16.0), Spotify Connect (0.14.0), MQTT Explorer (browser-1.0.1), File editor (5.8.0), Mosquitto broker (6.5.0), Zigbee2MQTT (2.1.1-1), Let's Encrypt (5.4.0), Node-RED (19.0.2), Advanced SSH & Web Terminal (20.0.1), ESPHome Device Builder (2025.2.1), Matter Server (7.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 20
views | 14
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s18 (enabled, default, auto), docker0 (disabled), hassio (disabled), vethab02e46 (disabled), veth43c8fa0 (disabled), veth159fd91 (disabled), veth2e8f132 (disabled), veth620c0e2 (disabled), veth1a7bcc7 (disabled), veth976f31c (disabled), veth7505ee3 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s18 (192.168.178.65/24), docker0 (172.30.232.1/23), hassio (172.30.32.1/23), vethab02e46 (), veth43c8fa0 (), veth159fd91 (), veth2e8f132 (), veth620c0e2 (), veth1a7bcc7 (), veth976f31c (), veth7505ee3 ()
ipv6_addresses | lo (::1/128), enp0s18 (2003:c8:5f31:ac00:53a7:76a0:5450:4dbf/64, fd96:81b8:6140:0:1ec0:1962:cd7:a0/64, fe80::535c:f882:6354:9fd3/64), docker0 (fe80::42:89ff:fe0e:5f57/64), hassio (fe80::42:d5ff:feae:1bee/64), vethab02e46 (fe80::1883:67ff:fe58:4fc4/64), veth43c8fa0 (fe80::4457:a2ff:fea0:d9fb/64), veth159fd91 (fe80::7029:1cff:fe63:8f11/64), veth2e8f132 (fe80::ac4f:81ff:fe10:fecd/64), veth620c0e2 (fe80::705f:6fff:feba:d05b/64), veth1a7bcc7 (fe80::1cc1:5fff:fe54:1a18/64), veth976f31c (fe80::3cd6:e7ff:feff:eba8/64), veth7505ee3 (fe80::444a:e2ff:fefc:4230/64)
announce_addresses | 192.168.178.65, 2003:c8:5f31:ac00:53a7:76a0:5450:4dbf, fd96:81b8:6140:0:1ec0:1962:cd7:a0, fe80::535c:f882:6354:9fd3

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15. Februar 2025 um 18:13
-- | --
current_recorder_run | 26. Februar 2025 um 13:55
estimated_db_size | 751.98 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Integration version

Integration Anker 2.5.0

Anker Solix Firmware 1.0.3.2

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

As of today, I can no longer change the feed-in specification. It still worked yesterday. Setting possible via Anker app. Any ideas?

Everything else works.

Anker Solix E1600 AC

### Reproduction steps

1.try to change Feed in
2.
3.
...

### Debug logs

```text
[custom_components.anker_solix]
number.solarbank_2_e1600_ac_system einspe sevorgabe change to 90 will be applied
2025-02-26 15:03:42.403 DEBUG (MainThread)
[custom_components.anker_solix] Schedule to apply: {'mode_type': 3, 'custom_rate_plan': K'index': 0, 'week':
10, 1, 2, 3, 4, 5, 6], ranges: K'start_time: 00:00,
'end_time': '24:00', 'power': 90HY, 'blend_plan': None,
'use_time': None, 'manual_backup': {'ranges':
['start_time': 1740506940, 'end_time: 17405177404,
'switch': False}, 'default_home_load': 200, 'max_load':
800, 'min_load': 0, 'step': 10}
2025-02-26 15:03:42.403 DEBUG (MainThread)
[custom_components.anker_solix] Request Url: POST
https://ankerpower-api-eu.anker.com/power_service/v1/ site/set_site_device_param
2025-02-26 15:03:42.403 DEBUG (MainThread)
[custom_components.anker_solix] Request Headers:
{'Content-Type': 'application/ison', 'Model-Type':
'DESKTOP', 'App-Name': 'anker_power', 'Os-Type':
'android', 'Country': 'DE', 'Timezone': 'GMT+01:00', 'x-auth-token':
'27###masked###1828###masked###ce1e###masked## #45', 'gtoken': '13###masked###249e###masked###4f'}
2025-02-26 15:03:42.403 DEBUG (MainThread)
[custom_components.anker_solix] Request Body:
{'site_id': '4334d5e7-fa42-4271-9ac0-c105b635f6b2',
'param_type': '6', 'cmd': 17, 'param_data': '{"mode_type" 3, "custom_rate_plan": ["index": 0, "week": [O, 1, 2, 3, 4, 5, 6], "ranges": ["start_time": "00:00", "end_time":
"24:00", "power": 90}]}], "blend_plan": null, "use_time": null, "manual_backup": {"ranges": ["start_time":
1740506940, "end_time": 1740517740}], "switch": false},
"default_home_load": 200, "max_load": 800, "min_load":
0, "step": 10}'}
2025-02-26 15:03:42.568 DEBUG (MainThread)
[custom_components.anker_solix] Kev*** request post https://ankerpower-api-eu.anker.com/power_service/v1/
```

thomluther, 2025-02-26:
Hi
This is a known issue, see https://github.com/thomluther/ha-anker-solix/issues/198#issuecomment-2646624057
I have no idea what is causing it.
It could be a firmware problem.
And the fact that it worked before but no longer is also pointing to this.

At least it does not appear to be a generic Api or integration issue, since the format of the schedule that must be passed for this control did not change.
It is simply ignored by the device or unknown reasons...

Koxxi89, 2025-02-26:
It's really very strange. All other data is imported correctly by Homeassistant and is displayed... If I manually change the feed in preset, it does not work via Ha. It works directly via the Anker app. There is also no error in ha. Strange

thomluther, 2025-02-26:
Did you recently create a use time plan?
If so, can you delete it and try if that makes a difference?
Or maybe delete a manual backup setting that is active?
Those new plans have been added to the schedule object in the Api, maybe the firmware does not fully support it yet when changed via the Api?

Koxxi89, 2025-02-26:
I have run a dynamic load adjustment via automation. This ran yesterday without any problems and this morning it stopped working.

I have also deactivated this automation and tried manual changes. Without success. 

Enclosed is my "main" automation. There is another one that switches to manual at 20% charge and smart meter goes on at 90%.

alias: Dynamische Solarbank Steuerung mit Durchschnitts-Solarleistung
description: Passt die Einspeisung basierend auf Solarleistung und Akkustand an.
triggers:
  - entity_id: sensor.solarbank_2_e1600_ac_solarleistung
    trigger: state
  - entity_id: sensor.solarbank_2_e1600_ac_ladestand
    trigger: state
  - entity_id: sensor.smart_meter_netzbezug
    trigger: state
conditions:
  - condition: or
    conditions:
      - condition: numeric_state
        entity_id: sensor.solarbank_2_e1600_ac_solarleistung
        above: 0
      - condition: numeric_state
        entity_id: sensor.solarbank_2_e1600_ac_ladestand
        above: 20
actions:
  - alias: Durchschnitt berechnen
    variables:
      avg_solar_power: "{{ states('sensor.durchschnittliche_solarleistung') | float(0) | int }}"
  - alias: Variablen setzen
    variables:
      batterieladung: "{{ states('sensor.solarbank_2_e1600_ac_ladestand') | int }}"
      netzbezug: "{{ states('sensor.smart_meter_netzbezug') | int }}"
      hausverbrauch: "{{ states('sensor.hausnetzverbrauch') | int }}"
      max_einspeisung: "{{ netzbezug if netzbezug | int > 0 else 0 }}"
      einspeisung: |-
        {% if avg_solar_power == 0 and batterieladung <= 20 %}
          0
        {% elif batterieladung >= 90 %}
          {{ hausverbrauch }}
        {% elif batterieladung < 20 %}
          0
        {% elif batterieladung < 90 %}
          {% set val = (avg_solar_power * 0.5) | int %}
          {{ val if val <= max_einspeisung else max_einspeisung }}
        {% else %}
          {% set val = (avg_solar_power * 0.9) | int %}
          {{ val if val <= max_einspeisung else max_einspeisung }}
        {% endif %}
  - alias: Einspeisevorgabe setzen
    target:
      entity_id: number.solarbank_2_e1600_ac_system_einspeisevorgabe
    data:
      value: "{{ einspeisung }}"
    action: number.set_value
mode: restart

thomluther, 2025-02-26:
This has nothing to do with the automation.

I just noticed in your debug output that you have no use time plan, only an old backup plan which is no longer active.
Did you recently ran this manual backup and is the issue since that backup was defined/active?
Unfortunately I don't think you can remove the old backup plan from the object. You can only deactivate the option switch, but that probably does not clear the plan in the schedule object...
```
'param_data': '
  {
    "mode_type" 3,
    "custom_rate_plan": 
      ["index": 0, "week": [O, 1, 2, 3, 4, 5, 6], 
        "ranges": ["start_time": "00:00", "end_time": "24:00", "power": 90
             }]}],     => Those brackets have wrong order!?!?
    "blend_plan": null, 
    "use_time": null, 
    "manual_backup": 
      {
        "ranges": ["start_time": 1740506940, "end_time": 1740517740}],      => Those brackets have wrong order!?!?
        "switch": false
       },
    "default_home_load": 200, 
    "max_load": 800, 
    "min_load": 0, 
    "step": 10
  }'
```

While formatting the output, I think the passed format is weird. Since that is passed as a string it could be wrong. Need to verify that in the code...
But I also noticed the O instead of 0 in the week day list which is weird!?! Copy and paste error from the debug log?
Can you provide the debug file when you process a change to verify the real data?

Koxxi89, 2025-02-26:
What is happening here? Login data is correct, I can also see the network consumption etc.[image]

thomluther, 2025-02-26:
> What is happening here? Login data is correct, I can also see the network consumption etc.[image]
> 
> And idea where i can find it?

Parallel access with same account by multiple devices, they kick out eachother
When you do that 10 times, you get an IP ban and have to wait 24h or reconnect DSL router to hopefullly get a new IP address.

Koxxi89, 2025-02-26:
So I work with the user-defined mode, the time 0 o'clock to 24 was activated yesterday, but the control worked perfectly between homeassistant and Anker.
I therefore think that this is not the problem.

I have restarted the router and also got a new ip. The problem persists.

Koxxi89, 2025-02-26:
[image]

Koxxi89, 2025-02-26:
[image]

Maybe something with this?

thomluther, 2025-02-26:
These are all different things that you mix up here.

1. MWR Limit goes to 0
That is likely a firmware problem. The same happens with the Expansion packs, that go to 0 if no Anker App is connected with the main account for few minutes. This is NOT an Integration problem

2. Issue with output preset not applied
I need the debug log when you apply such a change. Activate debug mode in the Integration, make a change and disable debug mode, then you will automatically have the download file when you do it in the browser

3. The new warning
I don't know if that is from a recent core code change or not. Its only a warning for some core function that integration is using that will no longer work end of the year. Nothing to worry about now.
Why your via device is not found is strange, because that may mean that you changed you account alias? You probably should delete the configuration entry and create it again to cleanup the device registry...

4. The automatic kick out from the account login has nothing to do with a new IP
When different clients try to login with same account, the last one kicks out the first one. The App does not automatically authenticate again, but the HA integration automatically retries authentication on next refresh cycle (or first new Api request which reports authentication token has been kicked off. You don't have to get a new IP if you had no IP ban. You will see in the error message that login is blocked for 24 hours, then you can recovery it with obtaining new IP.

thomluther, 2025-02-27:
Well
In this output the format is correct when reformatting the output:
```
2025-02-27 13:40:37.418 DEBUG (MainThread) [custom_components.anker_solix] Request Body: 
{
  'site_id': '4334d5e7-fa42-4271-9ac0-c105b635f6b2',
  'param_type': '6',
  'cmd': 17,
  'param_data':'
    {
      "mode_type": 3,
      "custom_rate_plan":
        [
	  {
	    "index": 0,
	    "week": [0, 1, 2, 3, 4, 5, 6],
	    "ranges": 
	      [
	        {
		  "start_time": "00:00",
		  "end_time": "24:00",
		  "power": 320
		}
	      ]
	  }
	],
      "blend_plan": null,
      "use_time": [],
      "manual_backup":
        {
	  "ranges": [{"start_time": 1704136980, "end_time": 1704147780}],
	  "switch": false
	},
      "default_home_load": 200,
      "max_load": 800,
      "min_load": 0,
      "step": 10
    }'
}
```

It is also accepted by the Api, but not applied by the AC device.
So still at the same point, I have no clue why the AC is not applying the new schedule object...

Koxxi89, 2025-03-01:
Maybe it has something to do with that? I have the problem
Still can't solve solix resetter, reinstalled etc... unfortunately no longer works[image]

thomluther, 2025-03-02:
I don't know what you mean.
You should disable the Integration with the Api switch and then do the firmware upgrade via the App using your owner account.
After the firmware update you can re-enbable the Api switch and test if that new firmware made anything better.

noseshimself, 2025-03-05:
Things just got better; Anker is obviously working on the back end.

I'm now randomly losing the extension battery packs, too, but the dependent parameters (e. g. current battery charge and "Ladestand") are changing accordingly. The debug log shows the data for the additional batteries to be missing. So whatever the integration is doing is based on the API data -- which does not have anything to do with the data provided by the mobile application (which is now showing updates every 5 seconds...)

thomluther, 2025-03-10:
@noseshimself , @Koxxi89 
Hi, regarding the original issue that AC manual plan changes are not being applied...
I think this may be related to the fact, that the [set_device_parm method](https://github.com/thomluther/anker-solix-api/blob/0a17050a185359fc8c31e7c6a7ba4f0a56f8c12c/api/schedule.py#L217) may need to use another command integer than the default.
So far, for SB1 and SB2 plan changes, the default of 17 was good. Maybe for AC this must be something different to have the cloud applying the plan properly on the device?!?

Someone with the AC device would need to verify that via the Api library by testing the set_device_parm method with different command integers while providing a change in the custom_rate_plan object of the parm_data object.
A subsequent call with the get_device_parm method should show whether the custom_rate_plan change was really applied in the cloud, then it will probably be applied by the device too.

For SB2 models, you have to use paramType="6"
Optionally you can call the endpoint directly as done in the [set_device_parm method Line 251](https://github.com/thomluther/anker-solix-api/blob/0a17050a185359fc8c31e7c6a7ba4f0a56f8c12c/api/schedule.py#L251)

achimmayer, 2025-04-08:
Hello Thomas, first of all many thanks for the great integration. Are there any new findings regarding the error? I have the same problem with Solarbank AC, that the change of the feed-in value via Homeassistant is not accepted.

thomluther, 2025-04-09:
By circumstances, I fortunately found the trick today after I got some time with admin access to an AC System.
The problem is not with the Api or cloud, but with another dirty implementation of the set_device_parm Endpoint for AC systems. So far, the whole schedule could be applied with the modified fields and plan. For AC this is different.
Whenever the schedule contains a plan that is unique to AC, only the AC plans will be applied and the others in the schedule are ignored. Therefore no changes were applied to the common SB2 plans. To modify them for AC systems, the AC plan must be removed from the applied schedule!?! It seems just the the cloud applies the common plan changes.
I modified my helper methods already for this new requirement and now everything seems to be applied as intended.
Will be available soon with the support for usage time plan modifications.

achimmayer, 2025-04-09:
That sounds great. As soon as the 2.6.0 is available, I will test it and give feedback. Thank you

thomluther, 2025-04-10:
Fix Delivered with 2.6.0

closed 2025-04-10 by thomluther