thomluther/ha-anker-solix #215: Add daily energy stats for SB2 AC external inverter input

thomluther, 2025-02-28:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The SB2 AC model supports 'PV' input via an external balcony power system aka micro inverter.
While the individual PV channel data cannot be accessed yet via the Anker App, the data is being tracked in the cloud and can be pulled per 'PV channel'.

### Describe the solution you'd like

The micro inverter channel should be added as additional PV channel to track its daily energy.

### Describe alternatives you've considered

Use the micro inverter power sensor and create a Rieman Integral helper to calculate power and a utility meter over the integral helper to get daily energy.

### Additional context

This may add more queries to the energy endpoints and trigger the Too many Requests limit sooner if no throttling is being applied.

noseshimself, 2025-03-05:
I don't want to spoil your party but it seems that the AC is adding all sources (grid power, solar panels and upstream inverter) into one value. I tried that already with a UPS on the bottom right socket, a grid connection and a panel input. The panel is shown separately but all input power was added up and there was no way to split grid power and inverter power without additional power meter.

thomluther, 2025-03-06:
I don't really care about the value validity of the statistics, for the SB1 they are partially garbage and therefore I don't use them.
This issue is just to add the energy from the additional (AC unique) external inverter input energy that counts as extra Solar Channel.
This seems to be combined with the PV channel Energy to the Solar Energy.
If grid charge energy is also added to Solar Energy, then it is as it is.
But there is also a grid to battery energy field, not sure if that is used by the AC however.

thomluther, 2025-03-10:
Delivered with 2.5.5

closed 2025-03-10 by thomluther