thomluther/ha-anker-solix #218: Charge throttling mode?

orangecoding, 2025-03-06:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hey.

I'm not sure if this is possible (probably not), but if the Battery (in my case the E1600) is almost full (~95%) it is going into some kind of throttle mode, charging only with ~80Watts. If I have the battery in charge mode (no power export), then a lot of guice is wasted. 

Is it possible to check if the battery is in throttle mode and set the value accordingly as a sensor?

### Describe the solution you'd like

Just a question if it is possible

### Describe alternatives you've considered

none

### Additional context

none

thomluther, 2025-03-06:
Hi @orangecoding 
That what you are asking for is not the job of an integration, therefore you have HA.
You can build automations of any kind an set a binary helper as needed, or do actions as you like based on various sensor conditions.

orangecoding, 2025-03-06:
I got that, but my question basically was, is there a way to figure out when it is going into throttle mode.

thomluther, 2025-03-06:
> I got that, but my question basically was, is there a way to figure out when it is going into throttle mode.

The device itself has no metric or Api field which indicates a throttle mode. You have to build these kind of throttle criteria on your own.
Why don't you generally allow output once 90% SOC are reached, then you don't have the issue at all.

thomluther, 2025-03-06:
And BTW, the throttle can start anywhere in the 90's, the SOC must not be that accurate, but the throttling may start on other conditions.
And furthermore, not allowing output is also not efficient if you have base consumption that can be bypassed.
Otherwise, the throttle can also occur when the battery charge is limited of any reason. Not only when close to full, but also when being empty, too cold etc...
If you allow a small bypass, anything that cannot be charged will be bypassed without PV throttle, and you can directly use it without battery storage loss of ~20% Energy.

orangecoding, 2025-03-07:
Ok. Thank you :)

closed 2025-03-07 by orangecoding