thomluther/ha-anker-solix #219: Add synthetic sensors to

noseshimself, 2025-03-07:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

The sensors provided by Anker are a bit frustrating and not that useful for the Energy panel (which is mostly dealing with amounts over time). Using (integral) helpers  for the energy dashboard is leading to problems when API access is turned off (e. g. while configuring parameters not available inside the integration (yet)).

There are also parameters which are missing but provided by the API calls (e. g. battery pack serial numbers).

### Describe the solution you'd like

Please add device and  battery serial numbers which are available via API; it might also be possible to receive individual battery pack charge levels via API (I found them on the AC but not on the Pro if/when batteries have been detected). 

The only reliable way to find out how much battery power is provided on the E1600 Pro is deriving the data from "Akkuleistung"; "AC Hausabgabe" and "AC Steckdose" do not deliver reliable data depending on the operating mode. It would be making things easier for users turning this into two positive numbers "power to battery" and "power from battery" depending on Akkuleistung being positive or negative.

### Describe alternatives you've considered

Helpers. But they are not that helpful.

### Additional context

 -

thomluther, 2025-03-07:
> The only reliable way to find out how much battery power is provided on the E1600 Pro is deriving the data from "Akkuleistung"; "AC Hausabgabe" and "AC Steckdose" do not deliver reliable data [...]

This is something you have to create on your own to consider any specifics that your devices may have. The integration can only rely on the values provided from the cloud, and even those are inconsistent to some degree, especially for battery charge/discharge.
This  had to be made consistent since the beginning to provide reliable fields for the integration, but actually that are just work arounds for Anker's ugly implementation of the Api. And this is getting more and more inconstent for various device types or new fields that they implemented over time.

If you need separate positive values for battery charge and discharge, please follow [this guide](https://github.com/thomluther/ha-anker-solix/discussions/16#discussioncomment-10285726).

In general I try to avoid adding virtual sensors as much as possible. This is not the job of the integration, since that has to provide the values as offered by the devices. One excample is the Energy capacity that is calculated by the number of batteries and the nominal capacity of the batteries. And this created already lot of issues in the community, since it is also wrong when battery pack number is wrong or capacity is wrongly assumed. Furthermore it is only a theoretical value of no practicable use...
Virtual sensors can be created by yourself, that is the job of HA and you, and you can customize them as needed and how they make any value for your setup and automations.

> Please add device and battery serial numbers which are available via API; it might also be possible to receive individual battery pack charge levels via API (I found them on the AC but not on the Pro if/when batteries have been detected).

@noseshimself If you can provide me example Api calls with parameters and responses showing the battery expansion serials or individual charge levels, I can easily add them to the integration. But I'm not aware how to get these data via the Cloud Api.

thomluther, 2025-03-07:
> The sensors provided by Anker are a bit frustrating and not that useful for the Energy panel (which is mostly dealing with amounts over time). Using (integral) helpers for the energy dashboard is [...]

What you need for the HA energy dashboard is a total_increasing energy sensor.
The Anker Cloud provides this only for the total generated PV Energy. There is nothing for the total battery charge or discharge energy.
All other energies can be queried just in INTERVALS that you have to specify. The integration does that only for 'today' to have a daily energy sensor, and for 'yesterday' to provide that as attribute to the daily sensor to see the full previous period energy.
The cloud does not provide anything that can be directly provided by the integration for use in the energy dashboard.
Furthermore do the energy sensors require LOTS of Api queries for various types and various intervals and you quickly exceed the Api endpoint request limit, that has been further reduced to 10 per minute as of Feb. 2025.
If no total increasing energy or yield is provided by the Api, the integration cannot provide that either.

thomluther, 2025-03-07:
> > [@noseshimself](https://github.com/noseshimself) If you can provide me example Api calls with parameters and responses showing the battery expansion serials or individual charge levels, I can easily add them to the integration. But I'm not aware how to get these data via the Cloud Api.
> 
> Not anymore -- my batteries are completely absent and seem to enjoy their holidays. And I do not have high hopes to see them again right now. My daily training consists of looking at charge levels of batteries using the mobile app, powering down the E1600 AC, removing all charged packs from the stack and putting E1600 AC of those that are not getting charged. So the bugs seem to get worse as soon as you have more than one additional BP1600.

Which kind of endpoint/query did you use to get expansion serials or charge levels?
The expansion charge issues are supposed to be fixed in firmware *.3.4 what I have read. You may try to ask Anker support to get the latest firmware if not available yet.

noseshimself, 2025-03-07:
> Which kind of endpoint/query did you use to get expansion serials or charge levels? 

I remember seeing the "internal" serial numbers (06... and 07...) in the device stats in January when I took a look at the integration's debug output.

>The expansion charge issues are supposed to be fixed in firmware *.3.4 what I have read. You may try to ask Anker support to get the latest firmware if not available yet.

I'm one problem ahead. Independent of the number of batteries announced by the device only two of the batteries are getting charged. All Anker is willing to offer is replacing my device as they can't see any good reason for this behavior. We agreed on waiting for a firmware update that will also update the batteries to 0..2.3.1 (the AC is currently only permitting 0.2.2.9 and replacing everything if things are not changing).

thomluther, 2025-03-08:
> I'm one problem ahead. Independent of the number of batteries announced by the device only two of the batteries are getting charged. All Anker is willing to offer is replacing my device as they [...]

I have read that 1.0.3.4 of the main device is about to fix the charging issues for 3 or more expansions. 
https://www.photovoltaikforum.com/thread/237945-solix-solarbank-2-ac/?postID=4203225#post4203225

> I remember seeing the "internal" serial numbers (06... and 07...) in the device stats in January when I took a look at the integration's debug output.

I don't think so, then they should be contained in any export. But all SB2 exports I have seen do not show anything of the expansion batteries. They are not listed as individual devices.

noseshimself, 2025-03-09:
> I have read that 1.0.3.4 of the main device is about to fix the charging issues for 3 or more expansions. https://www.photovoltaikforum.com/thread/237945-solix-solarbank-2-ac/?postID=4203225#post4203225

I made a grave mistake today; I connected all BP1600 to my E1600 Pro because I wanted to see if they would get charged there.

Yes. They do. After getting upgraded to 2.3.1, that is. Guess what's not working with E1600 AC? BP1600 with Firmware above 2.2.9.

But this is certainly fixing my charging problem, too.

thomluther, 2025-03-10:
> Guess what's not working with E1600 AC? BP1600 with Firmware above 2.2.9.

Maybe you have to upgrade the AC first to 1.0.3.4 before this accepts battery packs above 2.2.9?

noseshimself, 2025-03-10:
> > Guess what's not working with E1600 AC? BP1600 with Firmware above 2.2.9.
> 
> Maybe you have to upgrade the AC first to 1.0.3.4 before this accepts battery packs above 2.2.9?

Yes. That occurred to me, too. After auto-upgrading the battery packs. 8-). I already put a sign on the devices warning of moving batteries around. I great at locking the barn after the horses are gone.

Let's close this topic before I'm making more mistakes.

noseshimself, 2025-03-10:
Ummmmm. No,

[image]

no attributes there.

thomluther, 2025-03-10:
> no attributes there.

Which attributes are you looking for?

noseshimself, 2025-03-10:
> > no attributes there.
> 
> Which attributes are you looking for?

Any attributes at all. As you said in https://github.com/thomluther/ha-anker-solix/issues/219#issuecomment-2705927953 there should be some hanging below "SB Solarleistung" but there are none.

There is really not much to be found in the attributes of E1600 AC devices and components (at least not on my system).

thomluther, 2025-03-11:
> > > no attributes there.
> > 
> > 
> > Which attributes are you looking for?
> 
> Any attributes at all. As you said in [#219 (comment)](https://github.com/thomluther/ha-anker-solix/issues/219#issuecomment-2705927953) there should be some hanging below "SB Solarleistung" but there are none.
> 
> There is really not much to be found in the attributes of E1600 AC devices and components (at least not on my system).

No, what I wrote in that comment is how you can get the device attributes of any entity. The device attributes are NOT shown in the developer panel, you can see some of the device attribute values in the UI when you look at the Integration device.
An you can use ANY entity of the device to get the device attributes. Optionally you would have to use the device ID, but that is like a hash without context, so easier will be to use one of the entities in your jinja template.

I was referring to the developper panel for entity attributes (not device attributes).
Especially for the select entities, since they will show the possible raw options as you have to use them in your action or automation YAML code. The UI may translate those raw options to something different, but the translation does not work in YAML.

[image]

noseshimself, 2025-03-12:
> No, what I wrote in that comment is how you can get the **device** attributes of any entity. The device attributes are NOT shown in the developer panel

*facepalm*

How stupid is *that*?  Sorry, I did not know that; I expected to see anything worth to be seen there.

thomluther, 2025-03-12:
That is the way how HA is built. An entity has a state and attributes and can belong to a device, which also has its own attributes.
Device attributes however cannot be seen in the developer panels directly...
You need to use jinja templates to grab those attributes from the core state machine