thomluther/ha-anker-solix #221: Losing network connection using the integration after Update to V1.0.5.32

RamsukKockoi, 2025-03-10:
### System Health details

## System Information

version | core-2025.3.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1575
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21. April 2025 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | eu-central-1-18.ui.nabu.casa
certificate_status | ready
instance_id | 862965b5ef2343c99194cf582313bfc6
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 457.7 GB
disk_used | 11.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0), Grafana (10.4.2), Samba share (12.5.0), InfluxDB (5.0.2), Mosquitto broker (6.5.0), Terminal & SSH (9.16.0), SQLite Web (4.3.1), File editor (5.8.0), Glances (0.21.1), Plex Media Server (3.6.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 10
-- | --
resources | 9
views | 38
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), vethbe7fd50 (disabled), vetha833f60 (disabled), vethaba3ed4 (disabled), veth837ddb2 (disabled), veth2972432 (disabled), vethfe3d94e (disabled), veth58ea774 (disabled), veth2a5193b (disabled), veth0f76c4d (disabled), veth482c24a (disabled), veth2726e43 (disabled), veth72c104f (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.178.150/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), vethbe7fd50 (), vetha833f60 (), vethaba3ed4 (), veth837ddb2 (), veth2972432 (), vethfe3d94e (), veth58ea774 (), veth2a5193b (), veth0f76c4d (), veth482c24a (), veth2726e43 (), veth72c104f ()
ipv6_addresses | lo (::1/128), end0 (2a02:8108:9400:9000:b48:c57a:e463:82aa/64, fe80::c7cd:7aff:d62a:190c/64), hassio (fe80::42:f5ff:fe9e:7273/64), docker0 (fe80::42:eaff:fe55:ba05/64), vethbe7fd50 (fe80::8ff:63ff:fe27:956c/64), vetha833f60 (fe80::2cfa:2eff:fec2:ecdc/64), vethaba3ed4 (fe80::f8c5:b9ff:fe2b:8856/64), veth837ddb2 (fe80::986c:a7ff:fe72:8cac/64), veth2972432 (fe80::74c3:19ff:fe00:ba19/64), vethfe3d94e (fe80::f0b9:76ff:fe65:613a/64), veth58ea774 (fe80::e012:b2ff:fe61:1906/64), veth2a5193b (fe80::c3:49ff:fe65:dd98/64), veth0f76c4d (fe80::b4e7:49ff:fe54:bcec/64), veth482c24a (fe80::60f5:a7ff:fe01:f205/64), veth2726e43 (fe80::940e:eeff:fe02:8bc2/64), veth72c104f (fe80::2cc0:9eff:fe2b:58cb/64)
announce_addresses | 192.168.178.150, 2a02:8108:9400:9000:b48:c57a:e463:82aa, fe80::c7cd:7aff:d62a:190c

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 27. Februar 2025 um 12:46
-- | --
current_recorder_run | 9. März 2025 um 12:43
estimated_db_size | 2126.20 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Integration version

2.5.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The SB2 Pro loses network and "freezes" if the integration is running and collecting data. In some cases it loses network immediately, in some cases it needs 1-2 hours to lose it and freeze.
Only a restart fixes this issue. This has not been with the previous version.

Special for me: 2 Energysystems under one logon, quite a little more api requests.

The app shows "no network" and SB2 is grey. And the app displays, that the "energy configuration" is not complete.

Is this issue known?

### Reproduction steps

1. Reboot SB2
2. Wait a little (while it works)
3. start the HA integration and it will lose network connection
...

### Debug logs

```text
Will be there if needed
```

### Diagnostics dump

Will be there if needed

thomluther, 2025-03-10:
Hi
Actually I have no clue what is issue could be.
Since the integration never communicated directly with the devices of your system (only to the cloud server), I have no idea what may cause the devices to hang.

However, Anker reduced the endpoint limit. Check the updated documentation in the README for the coming release 2.5.5.
You can install that release already from the `main` branch and see if things get better.
[Here are instructions for pre-release and main branch installations of your HACS integrations](https://github.com/thomluther/ha-anker-solix/discussions/222)

RamsukKockoi, 2025-03-10:
Hey there, as it now ran for over 12 hours without API, it is now stuck again. So I thnik we now can say, it is not the API / integration.

closed 2025-03-10 by RamsukKockoi