thomluther/ha-anker-solix #225: Concerning sensors for Energy Dashboard

Mahn249, 2025-03-12:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hi!
I Added this really good Add-on, but as a programming noob i cannot correct things.
I noticed the following:
The entity sensor.solarbank_e1600_solarpower has the device_class “power”.
This means that Homeassistant cannot integrate it into the official energy overview.
For this, the sensor would have to have the device_class “energy”. 

This applies to all sensors, both those for the solar panels (solar power) and the sensors for the battery (discharging/charging power)

Like seen on this article:
https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities

Can you change that?

### Describe the solution you'd like

For linking the entities from ythis API, the sensors have to have the device_class “energy”. 

This applies to all sensors, both those for the solar panels (solar power) and the sensors for the battery (discharging/charging power)

Like seen on this article:
https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities

### Describe alternatives you've considered

didn't look for a workaround.

### Additional context

[image]

Should look like this:
https://www.home-assistant.io/home-energy-management/

thomluther, 2025-03-12:
Hi @Mahn249 
I think you got that wrong.
Power Sensors are Power class and Energy Sensor are Energy class. Energy is power over time.
You will have to create Rieman Intergral helper over the power sensors to get energy which you can add to your energy dashboards and cards. Please review the documentation and the discussion threads, especially [this one](https://github.com/thomluther/ha-anker-solix/discussions/16#discussioncomment-10285726).

Mahn249, 2025-03-12:
Oh i'm really sorry. I didn't search for this doc...

closed 2025-03-12 by Mahn249