thomluther/ha-anker-solix #230: Add API calls per minute (limits) as sensor to the Diagnostics section

noseshimself, 2025-03-18:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I'm always frustrated when I try to find out what happened by reading logs. 

### Describe the solution you'd like

I would like to see update rate limits and current update rates in the  diagnostic parts of my UI. Is it possible to get current rates, limits, the time of the last refresh and the last API error (as a text sensor) into the Diagnostic section of the site/device? 

### Describe alternatives you've considered

Constant monitoring of the logs and using script for grabbing the data there, turning it into sensors. This sounds counter intuitive to me. I'm suffering from something like this to get my current IPv6 address already.

### Additional context

 .

thomluther, 2025-03-18:
BTW, the made calls per minute and hour you can find under your account device.
As for the limits per endpoint....I have no clue what that is. It is different per endpoint, and the integration never knows how many other calls have been done in last minute under your IP address.

noseshimself, 2025-03-19:
I understood the commit-message of the latest version as "the integration might have to (dynamically) adapt the number of API calls due to communication/HTTP errors returned in normal operation" (and I can easily see that because the amount of data is changing depending on batteries being detected or not).

I already prepared moving every site into its own Anker account (the next E1600 AC already arrived; I'm waiting for the batteries to show up) and another is on the way to another site I wanted to include, too.  I would like to know if anything is happening to the API communication without screening logs.

This extends to the API error results; I would like to know if something was a general problem that needs reporting, a transient error (I've already seen 503 results from their load balancer/proxy) or exceeding the number of API calls per minute.

thomluther, 2025-03-19:
> I understood the commit-message of the latest version as "the integration might have to (dynamically) adapt the number of API calls due to communication/HTTP errors returned in normal operation" [...]

That is a misunderstanding. The only change I did was that the configured throttle of 10 requests per minute per endpoint is not applied per default. Only for endpoints that run into the error. But it is not clear which endpoints are limited by what number.

And again, the throttle counts only for its own client. Every configuration entry has its own client that runs in parallel, plus you have the App. All may drive requests to the same endpoint under the same IP and the cloud will only see the combined requests.

> I already prepared moving every site into its own Anker account (the next E1600 AC already arrived; I'm waiting for the batteries to show up) and another is on the way to another site I wanted to [...]

I would NOT recommend that, since that causes much more traffic than with a single account.
There is no serialisation or combined throttling, you will probably see all kind of Api errors the more you configure and the more Energy data you want!!!!
There is NOTHING I can do to change that.

> This extends to the API error results; I would like to know if something was a general problem that needs reporting, a transient error (I've already seen 503 results from their load balancer/proxy) or exceeding the number of API calls per minute.

I don't know.

noseshimself, 2025-03-19:
And just now Anker support has hinted at pondering whether MQTT might provide a better API for access to the kind of data energy storage and production products are generating.

Let's close this request as "not going to be implemented."

thomluther, 2025-03-19:
I leave it open to track implementation of diagnostic sensors for last data refresh and details refresh timestamps.
That might be useful for someone interested in those background pulls. However, they will not give any clue how old the presented data in the cloud actually is...therefore the Cloud data timestamp was implemented...

thomluther, 2025-04-10:
Delivered with 2.6.0

closed 2025-04-10 by thomluther