thomluther/ha-anker-solix #232: Unknown SB2 charging status when battery is full

thomluther, 2025-03-18:
Hi all,
everytime when my Solarbank 2 pro is full the operating status is set to "unknown". The energy shows 1,600 Wh, the charge status is 100%.
Why isn't the operating status set to 'full' instead of 'unknown'.
Is this a bug or is my configuration wrong.
Current installed version is 2.5.7

Thanks
Regards
Tom

_Originally posted by @HA-Tom in https://github.com/thomluther/ha-anker-solix/discussions/231_

thomluther, 2025-03-18:
It seems code 5 is used by SB2 once the battery is fully charged.
New charging description to be added for fully_charged with code 5

thomluther, 2025-03-18:
@HA-Tom , can you please upload a systems export when the battery is full?
Then I can verify the code changes accordingly.

HA-Tom, 2025-03-18:
Okay, that should be possible tomorrow, with full sunshine :-)

HA-Tom, 2025-03-18:
How and where can I create this systems export?

thomluther, 2025-03-19:
> How and where can I create this systems export?

https://github.com/thomluther/ha-anker-solix/blob/main/INFO.md#export-systems-action

HA-Tom, 2025-03-19:
Do I have to start something? I don't have the folder 'exports' under 'www/community/anker_solix/' There is only a folder with images.. neither do I have the folder /config
do i have to create them manually?

HA-Tom, 2025-03-19:
I created them manually now. I'll restart the Api.

HA-Tom, 2025-03-19:
The folder is empty. :-(

thomluther, 2025-03-19:
The /config folder is where you have your configuration.yaml.
On my instance it is /homeassistant/configuration.yaml
So /config could be /homeassistant

HA-Tom, 2025-03-19:
hmmmm okay, my configuation.yaml is also where yours is:

[image]

but there is no /config

HA-Tom, 2025-03-19:
[image]

thomluther, 2025-03-19:
But there is the www.
You don't need a /config, that is /homeassistant on yours

HA-Tom, 2025-03-19:
Yeeh I know there is www and under 'www' is 'community'. Under 'community is 'anker_solix' and under 'anker_solix' is 'exports' which I created manually. But there is no export_file which I can send you. The SB2 still is complete fully charged.

[image]

thomluther, 2025-03-19:
When you run the action, you may have to wait 1-3 minutes until throttle loops have been done. You will get an action response once finished and this will show you the created zip file. That must be located somewhere....

HA-Tom, 2025-03-19:
That is my entire question: which action should I run? Where can I start the mentioned action, I don't see a button or anything similiar to start any action.

HA-Tom, 2025-03-19:
Sorry Thomas, for my stupid questions. I'm still quite a newbie in HA and some things are still new for me.

HA-Tom, 2025-03-19:
I restarted the Api, I restarted complete HA but no zip-file has been created.

thomluther, 2025-03-19:
That is a HA Action in developper Tools

[image]

HA-Tom, 2025-03-20:
Ohhh okay, haven't used that before. Thanks a lot.
I'll start the action as soon as the SB 2 in fully charged. should I put the created file here in our discussion?

thomluther, 2025-03-20:
Yes, you can upload it here as reference for the status code 5 state.

HA-Tom, 2025-03-20:
It worked!

[image]

[Thomas Jaeger_2025-03-20_1356.zip](https://github.com/user-attachments/files/19367473/Thomas.Jaeger_2025-03-20_1356.zip)

HA-Tom, 2025-03-20:
100% fully charged at 2 pm, 1.600Wh...  status 'unknown'

HA-Tom, 2025-03-20:
Sorry Fully charged at 1:51 pm

HA-Tom, 2025-03-21:
Could the zip file help you find the code you need?

thomluther, 2025-03-21:
Hi, yes, I verified that my fix works. Will be in the next release

closed 2025-04-10 by thomluther