thomluther/ha-anker-solix #234: changed “Daily...” Sensors in the Anker “System”

Franzi24, 2025-03-27:
### System Health details

## System Information

version | core-2025.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.12.18-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1605
Downloaded Repositories | 27

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.0
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.3
agent_version | 1.7.2
docker_version | 28.0.1
disk_total | 30.8 GB
disk_used | 27.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.5.0), Advanced SSH & Web Terminal (20.0.2), Whisper (2.3.1), File editor (5.8.0), Duck DNS (1.18.0), NGINX Home Assistant SSL proxy (3.11.1), openWakeWord (1.10.0), Piper (1.5.2), Z-Wave JS (0.9.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 20
views | 15
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp0s3 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth46c042e (disabled), veth2e38ae6 (disabled), vethb04a3a0 (disabled), vethf48a5ef (disabled), veth105b118 (disabled), vethe79aa48 (disabled), vethcc37435 (disabled), veth5c78cf2 (disabled), veth3620904 (disabled), vethfafa79a (disabled), vethafb9925 (disabled), veth54d7f58 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp0s3 (192.168.178.56/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth46c042e (), veth2e38ae6 (), vethb04a3a0 (), vethf48a5ef (), veth105b118 (), vethe79aa48 (), vethcc37435 (), veth5c78cf2 (), veth3620904 (), vethfafa79a (), vethafb9925 (), veth54d7f58 ()
ipv6_addresses | lo (::1/128), enp0s3 (fd76:bc6b:2b72:0:c233:2f07:3e39:f6dc/64, 2a03:fc82:31f:5600:4374:4a6:1726:54bf/64, fe80::9191:febb:20de:3975/64), hassio (fe80::345a:8ff:fe8d:9eb0/64), docker0 (fe80::b0cf:56ff:fe35:fb15/64), veth46c042e (fe80::1410:63ff:fe80:d48c/64), veth2e38ae6 (fe80::3095:65ff:fe8e:dc55/64), vethb04a3a0 (fe80::306a:ecff:fe7b:4fcb/64), vethf48a5ef (fe80::480:e0ff:fed3:7fb1/64), veth105b118 (fe80::86:45ff:fe5e:3c5/64), vethe79aa48 (fe80::d82f:c1ff:feb2:d20a/64), vethcc37435 (fe80::3c3b:7cff:fe79:c35/64), veth5c78cf2 (fe80::34b4:bff:fe70:3611/64), veth3620904 (fe80::f422:c9ff:fe74:b786/64), vethfafa79a (fe80::8820:96ff:fe33:d687/64), vethafb9925 (fe80::7422:b6ff:fe8e:6ee8/64), veth54d7f58 (fe80::acc4:9fff:fe25:9728/64)
announce_addresses | 192.168.178.56, fd76:bc6b:2b72:0:c233:2f07:3e39:f6dc, 2a03:fc82:31f:5600:4374:4a6:1726:54bf, fe80::9191:febb:20de:3975

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 17. März 2025 um 02:01
-- | --
current_recorder_run | 27. März 2025 um 03:00
estimated_db_size | 736.62 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Integration version

2.5.7

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

The values of the “Daily...” sensors in the Anker “System” were always created cumulatively over the day in the past, so that the first aggregated figures were already available during the day, which I was very happy to use in my monitoring. For a few days now, there has only been a final value at the end of the day/night for a certain time, otherwise the value is 0. The data now only helps you to compare the individual days as a history, but you no longer have an overall value per day (I sleep at night when the value is generated). My hope is that this is perhaps a bug and can be adjusted in the interface or do the daily figures only come from the API like this?
Many thanks for any feedback.

### Reproduction steps

1. open "Anker Solix" integration
2. open "devices" of the "hub"
3. open device "system"
4. open various daily sensors (e.g. “Daily PV1”, “Daily discharge” or “Daily solar yield” etc.)
5. value = 0
...

### Debug logs

```text
Probably not necessary in this case.
```

thomluther, 2025-03-27:
Hi

Can you please first use the Anker Solix Action to get the system information (use the include cache data option).
This should show the calculated offset in seconds between your HA instance and the Solarbank data in the cloud. Maybe that got wrong over time...

To fix it, you can try to reload your configuration entry and see if that resolves the issue?
The latest version tries to identify a time gap between the server reported data and the HA instance in order to know, when the query for the next day (=date) can be started. Maybe this offset calculation got wrong and therefore the date toggle for the daily query is for wrong date.
The offset is calculated from the Solarbank Cloud data timestamp. But if that is not up to date when the configuration is loaded by HA, the offset is calculated incorrect.

Franzi24, 2025-03-27:
Many, many thanks for the quick response. Does this mean that nothing has actually changed in the sensors and that they increase throughout the day as before? That would be good. I have now pulled the current system information from Anker. Could you take a look at it? I can't find anything that might be strange, or am I wrong?
`sensor.system_hausdach_ertrag_gesamt:
  system_info:
    type: system
    site_info:
      site_id: ....
      site_name: Hausdach
      site_img: ""
      device_type_list:
        - 3
      ms_type: 2
      power_site_type: 5
      is_allow_delete: true
      support_device_models:
        - A17C1
        - A17X7
        - SHEM3
        - SHEMP3
        - A17X8
        - A17C0
      current_site_device_models:
        - A17C1
    site_type: solarbank
    site_admin: false
    data_valid: true
    requeries: 0
    energy_offset_seconds: -431
    energy_offset_check: "2025-03-27 14:44:27"
    energy_offset_tz: 0
    home_info:
      home_name: Home
      home_img: ""
      charging_power: "0.00"
      power_unit: W
    solar_list: []
    pps_info:
      pps_list: []
      total_charging_power: "0.00"
      power_unit: W
      total_battery_power: "0.00"
      updated_time: ""
      pps_status: 0
    statistics:
      - type: "1"
        total: "214.06"
        unit: kWh
      - type: "2"
        total: "213.42"
        unit: kg
      - type: "3"
        total: "68.50"
        unit: €
    topology_type: "1"
    solarbank_info:
      solarbank_list:
        - device_pn: A17C1
          device_sn: ...
          device_img: >-
            https://public-aiot-fra-prod.s3.dualstack.eu-central-1.amazonaws.com/anker-power/public/product/2024/05/24/iot-admin/5iJoq1dk63i47HuR/picl_A17C1_normal%281%29.png
          battery_power: "52"
          bind_site_status: ""
          charging_power: "233"
          power_unit: W
          charging_status: "6"
          status: "1"
          wireless_type: "1"
          main_version: ""
          photovoltaic_power: "233"
          output_power: "0"
          create_time: 1739007704
[... 257 lines cut ...]
        battery_charge: "0.00"
        solar_to_grid: "0.00"
        battery_percentage: "0"
        solar_percentage: "0"
        other_percentage: "0"
      last_period:
        date: "2025-03-26"
        battery_discharge: "2.06"
        ac_socket: "0.00"
        battery_to_home: "2.00"
        grid_to_battery: "0.00"
        home_usage: "6.43"
        grid_to_home: "3.55"
        smartplugs_total: "0.00"
        solar_production_pv1: "0.55"
        solar_production_pv2: "0.49"
        solar_production_pv3: "0.53"
        solar_production_pv4: "0.00"
        solar_production: "1.57"
        battery_charge: "0.69"
        solar_to_grid: "0.01"
        battery_percentage: "0.44"
        solar_percentage: "0.56"
        other_percentage: "0"
`

thomluther, 2025-03-27:
Hm normal offset:
```yaml
    energy_offset_seconds: -431
    energy_offset_check: "2025-03-27 14:44:27"
```
And you can see that the data is pulled from today. Not sure why those fields are 0 for most of the daily energies.
Maybe something is broken with the cloud stats? 
I don't think this is something I can fix.

Franzi24, 2025-03-27:
What a pity. I took a second look. The sensors have changed since last week, Thursday, March 20. It happened from one day to the next.

[image]

I was already thinking whether it could be due to the time difference, which was 3 weeks ago in America and is coming this weekend here, but that doesn't match the date.

I have now pulled one of these exports. Was that an offer for you to have a look or can you tell me where I might start looking?

thomluther, 2025-03-28:
> What a pity. I took a second look. The sensors have changed since last week, Thursday, March 20. It happened from one day to the next.

Did you update to the new Integration version or was it just a change in the cloud?
The numbers from today should reflect the total numbers that you find in the App when watching the intraday charts. If they are 0 from the cloud, you should see that also in the App intraday chart totals.

You can upload the zip file and I can check the Api responses of the energy stats just to make sure what comes from the cloud.

Franzi24, 2025-03-28:
That's very kind of you, thank you very much. And I can't remember any changes I made or confirmed last week.
And I can still see the daily value in kWh increasing in the app, which is why I'm so surprised that HAOS only shows a value at night.
[image]

[lizxxx_2025-03-27_2102.zip](https://github.com/user-attachments/files/19502252/lizxxx_2025-03-27_2102.zip)

thomluther, 2025-03-28:
Your export is somehow corrupted. It seems 2 exports have been triggered at the same time. Don't know how that can happen, since I changed this action to run only once if triggered.
```
21:01:21,181 anker_solix_export INFO Exporting Anker Solix data for all account sites and devices of api nickname tre***
21:01:21,185 anker_solix_export INFO Using AnkerSolixApiExport Version: 2.6.0.0, Date: 2025-03-27, Export Services: set()
21:01:21,185 anker_solix_export INFO 
21:01:21,185 anker_solix_export INFO Querying common endpoint data...
21:01:21,185 anker_solix_export INFO Querying site information...
21:01:21,191 anker_solix_export INFO Using AnkerSolixApiExport Version: 2.6.0.0, Date: 2025-03-27, Export Services: set()
21:01:21,192 anker_solix_export INFO 
21:01:21,192 anker_solix_export INFO Querying common endpoint data...
```
Due to randomization of the IDs in the output, that data export does not provide a good picture.
I tried to reproduce, but was not able to trigger 2 exports together...weird.

Can you export again?

thomluther, 2025-03-28:
After fixing the output files, I could extract some data, so that does not seem to be the issue. But the export queries energies for the last 2 days, which is different to the energy poller which does that only for today, so basically a different query interval from the cloud.
```yaml
    "energy_details": {
      "last_period": {
        "date": "2025-03-26",
        "battery_discharge": "2.06",
        "ac_socket": "0.00",
        "battery_to_home": "2.64",
        "grid_to_battery": "0.00",
        "home_usage": "6.43",
        "grid_to_home": "9.27",
        "smartplugs_total": "0.00",
        "solar_production_pv1": "0.55",
        "solar_production": "1.57",
        "battery_charge": "1.50",
        "solar_to_grid": "0.02",
        "battery_percentage": "0.50",
        "solar_percentage": "0.50",
        "other_percentage": "0"
      },
      "today": {
        "date": "2025-03-27",
        "battery_discharge": "0.66",
        "ac_socket": "0.00",
        "battery_to_home": "2.64",
        "grid_to_battery": "0.00",
        "home_usage": "6.97",
        "grid_to_home": "9.27",
        "smartplugs_total": "0.00",
        "solar_production_pv1": "0.49",
        "solar_production": "1.44",
        "battery_charge": "1.50",
        "solar_to_grid": "0.02",
        "battery_percentage": "0.50",
        "solar_percentage": "0.50",
        "other_percentage": "0"
      }
```

Anyway, due to the fact that the export was running twice, I assume something is weird in your HA instance. If you have just one Anker configuration enrty, you should try to restart HA and see if the problem goes away.

Franzi24, 2025-03-28:
I have two hubs in the integration. In one hub there are 3 devices (SmartMeter, solar bank, system) and in the other hub there are 2 devices (solar bank, system). I have now made another export from each of them. 
I also restarted Home Assistant just now and then it also pulls numbers from the daily sensors, but then they remain at the same level and do not develop any further. I've noticed this phenomenon over the last few days. A restart gives me numbers but without it the sensors don't seem to update regularly.

[trexxx_2025-03-28_2059.zip](https://github.com/user-attachments/files/19511654/trexxx_2025-03-28_2059.zip)

[lizxxx_2025-03-28_2101.zip](https://github.com/user-attachments/files/19511656/lizxxx_2025-03-28_2101.zip)

thomluther, 2025-03-28:
Hm....don't know what is going on. Eventually the energy data is not polled again after initial poll?
When the data is stall again, you can try to enforce a device data refresh, which should also trigger the energy poll.
Can't day what is going on, you would have to enable debug mode of the integration and collect the logs after 15 minutes in debug.
In that time frame, there should have run energy refresh at least.

thomluther, 2025-03-29:
One more reason you need to check with more than 1 hub entry.
Do you have errors in the log for Too many Requests?
The Endpoint limit of 10 has to be distributed to all your hub entries. Otherwise there can be situations where the Endpoint limit of the cloud is exceeded.

thomluther, 2025-03-31:
Hi @Franzi24 , any finding for errors in your logs that would explain why the daily energy data is not being refreshed regularly.

Franzi24, 2025-04-01:
The problem is that I can't get two solar banks into one system, therefore the two hubs, and since it worked for a while, I didn't think it was because of the two hubs.
At the moment I don't see any error messages, but the daily sensors are still not updated every hour.
Does it generally help if I deactivate some sensors that I don't need so as not to load the interface with them?

thomluther, 2025-04-01:
> The problem is that I can't get two solar banks into one system, therefore the two hubs,

I know that this is not possible, but you can create multiple systems in the App with the same account. Therefore access both systems with the same account. This would require only one hub for HA as well, and also in the App you would not need to toggle accounts for access to the other system.

thomluther, 2025-04-02:
It seems there is an issue with the refresh details staggering when multiple hubs are used and that may be delayed continously, not doing any more energy refresh after initial data poll.....
I reopen this to track and fix this for the next release.

thomluther, 2025-04-10:
Fix Delivered with 2.6.0

closed 2025-04-10 by thomluther