thomluther/ha-anker-solix #235: Solarbank 2 E1600 AC + 1bank

renard9, 2025-03-27:
### System Health details

When you add a bank, you see only 1 bank 

[image]

HA can see the adding bank here

[image]

We have In that case 1600Wh *2 

### Integration version

[image]

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

the energy storage data is false

### Reproduction steps

1. install a solarbank 2 E1600 AC
2. add a new bank
3. wait the  bank charging 
...

### Debug logs

```text
no log
```

### Diagnostics dump

no dump

renard9, 2025-03-27:
During 1 polling (around 13 o'clock), we have the good value for the storage

thomluther, 2025-03-27:
Hi, This is an AC issue with firmware or cloud. Nothing with the library, it just reports the data it receives from the cloud.

thomluther, 2025-03-27:
Closing as duplicate of #203